espressif/openocd-esp32 #241: esp32-c3 cannot launch built-in JTAG

sxliujin, 2022-08-08:
### Development Kit

my PCB

### Module or chip used

ESP32-C3-MINI-U

### Debug Adapter

USB built-in JTAG

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Win 11

### Using an IDE ?

ESPRESSIF-IDE 2.5.0

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

40000 kHz

### ESP-IDF version

4.4.2

### Problem Description

1. new launch configuration
2. choose ESP-IDF GDB OpenOCD Debugging
3. debug

### Debug Logs

```shell
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Flashing C:/Espressif/frameworks/esp-idf-v4.4/workspace/BMS76952_4S/build/partition_table/partition-table.bin at 0x8000
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [0] Found 8 triggers
Error: Failed to get flash maps (4294967295)!
Warn : Failed to get flash mappings (-4)!
Error: Failed to get flash size!
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32c3.flash'!
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Error: Failed to get flash maps (4294967295)!
Warn : Failed to get flash mappings (-4)!
Error: Failed to get flash size!
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
shutdown command invoked
```

### Expected behavior

run JTAG debug

### Screenshots

use same project configuration, I can debug ESP32C3 chip without inner flash(which use external flash such as GD25Q32)。

sxliujin, 2022-08-08:
untick the mark “Enable memory protection”, but It didn't work.

erhankur, 2022-08-09:
In the latest idf 4.4.2 release we updated the OpenOCD version to `v0.11.0-esp32-20220411` Can you try with this version?

sxliujin, 2022-08-09:
I have downloaded openocd v0.11.0-esp32-20220706. but openocd command of ESP-IDF 4.4 PowerShell  still start from v0.11.0-esp32-20211220. if I delete v0.11.0-esp32-20211220, shell can not find openocd command. How can I change the version of openocd?

sxliujin, 2022-08-09:
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Flashing C:/Espressif/frameworks/esp-idf-v4.4/workspace/BMS76952/build/bootloader/bootloader.bin at 0x0
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: serial (84:F7:03:BC:EF:3C)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
Info : [esp32c3] Found 8 triggers
Error: Failed to get flash maps (4294967295)!
Warn : Failed to get flash mappings (-4)!
Error: Failed to get flash size!
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32c3.flash'!
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Error: Failed to get flash maps (4294967295)!
Warn : Failed to get flash mappings (-4)!
Error: Failed to get flash size!
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
shutdown command invoked

finally I try to slower the JTAG clock to 10MHz or lower,but it still didn't work.

erhankur, 2022-08-16:
So, you say with the 4MB internal flash you get this error but external flash is working? 

Any chance to upload a test application to reproduce the issue?   I will try with the ESP32-C3 DEVKITM-1 . https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html

18506711722, 2022-09-04:
The same problem occurs in esp32-c3fh4

erhankur, 2022-09-09:
I don't see an issue with the idf example applications using boards with 4MB internal flash. If you can send me a small application to reproduce the issue, I can try and comment.

18506711722, 2022-09-09:
与应用系统无关,与是否内置flash有关,我有两块板子 @erhankur

18506711722, 2022-09-09:
It has nothing to do with the application, it has something to do with the built-in FLASH. I have two teams, the external flash and the built-in flash  @erhankur

GiulioDallaVecchia, 2022-09-23:
I have the same issue with my custom board, but with the ESP32C3 DevKit, OpenOCD runs without any problems

erhankur, 2022-09-23:
@GiulioDallaVecchia In terms of Jtag and flash pin connections what is the differences with your custom board and Devkit? 
Did you run the same application on 2 boards? How can you help me to reproduce this issue?

GiulioDallaVecchia, 2022-09-23:
[image]

I'm running the same application. This is my openocd command:

openocd -s ${openocd_path}/share/openocd/scripts -f board/esp32c3-builtin.cfg -c "init; halt; esp appimage_offset 0x20000"

erhankur, 2022-09-23:
@calandoa can you also share your detailed OpenOCD logs. (with -d3 option)

GiulioDallaVecchia, 2022-09-23:
[log.txt](https://github.com/espressif/openocd-esp32/files/9633148/log.txt)

erhankur, 2022-09-23:
Can you try ? `-c "init; reset halt; esp appimage_offset 0x20000"`

GiulioDallaVecchia, 2022-09-23:
same result:

[log2.txt](https://github.com/espressif/openocd-esp32/files/9633538/log2.txt)

erhankur, 2022-09-23:
OK I will ask this internally. In the meantime you can try to disable flash and continue debugging without software breakpoints. ESP32-C3 8 hw bps so it should be enough.
```
openocd -c 'set ESP_FLASH_SIZE 0' -f board/esp32-wrover-kit-3.3v.cfg
```
But with this option you cannot load the application from the jtag.

GiulioDallaVecchia, 2022-09-23:
I've started openocd with -c 'set ESP_FLASH_SIZE 0' but when I try to debug I have the same problem (openocd log):

[log3.txt](https://github.com/espressif/openocd-esp32/files/9633950/log3.txt)

erhankur, 2022-09-23:
Should be set before scripts. 
```
openocd -c "set ESP_FLASH_SIZE 0"  -s ${openocd_path}/share/openocd/scripts -f board/esp32c3-builtin.cfg  -d3
```

GiulioDallaVecchia, 2022-09-23:
ok, at least I can debug... I hope to hear you soon with some news... thanks!

Zoptune, 2022-12-30:
Hello, is there any update ?
I have the same issue with a custom board based on an ESP32-C3FH4.
Thanks.

CharlesScoville, 2023-01-05:
Try adding -c "gdb_memory_map disable" to your OpenOCD command. Worked for me +1

araldit, 2023-01-16:
That did not do it for me either. Its really annoying to have to program through serial for afterwards to connect through JTAG for debugging. I really hope it will be sorted out some day soon.

erhankur, 2023-01-16:
@araldit in the master branch we had a new feature that enables stub flasher logs and prints them to the openocd screen. 

`-c "init; reset halt; esp stub_log on"`

Can you build from master? Or if you tell me your OS, I can send you the binary to collect stub logs.  For windows, I already shared to test another issue. https://github.com/espressif/openocd-esp32/issues/255#issuecomment-1384207506

araldit, 2023-01-17:
I just compiled and tested the master, as suggested.  But it didn't do anything different:

**openocd --version:**
Open On-Chip Debugger v0.11.0-esp32-20221026-85-g0718fffd (2023-01-17-09:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

**Eclipse GDB Hardware Debugging commands:**
init; reset halt; esp stub_log on
mon program_esp ${workspace_loc:station/build/station.bin} 0x10000 verify
flushregs
set remote hardware-watchpoint-limit 2

**Result:**
openocd -f ~/Development/openocd-esp32/share/openocd/scripts/board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026-85-g0718fffd (2023-01-17-09:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (DC:54:75:89:BC:C8)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32c3] Found 8 triggers
Info : [esp32c3] Halt cause (7) - (PMP Store access fault)
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32c3] Halt cause (7) - (PMP Store access fault)
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Halt cause (7) - (PMP Store access fault)
Info : [esp32c3] Halt cause (7) - (PMP Store access fault)
Error: Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32c3] Halt cause (7) - (PMP Store access fault)
Error: Failed to get flash size!
Info : [esp32c3] Halt cause (7) - (PMP Store access fault)
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected

Its a ESP32C3-mini im trying to program and debug.

erhankur, 2023-01-18:
So, logs are enabled. This is what I want. 
```
Info : [esp32c3] Stub logs enabled!
Info : [esp32c3] Found 8 triggers
````
 Now just run the telnet commands, please.  In the meantime, I will test your binary.

araldit, 2023-01-18:
The problem is that it after running the command, it returns to the command line. But if i do this:

**openocd -f ~/Development/openocd-esp32/share/openocd/scripts/board/esp32c3-builtin.cfg -c "init; reset halt;"**
Open On-Chip Debugger v0.11.0-esp32-20221026-85-g0718fffd (2023-01-18-09:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : esp_usb_jtag: serial (DC:54:75:89:BC:C8)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

**Now telnet**
telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
\> esp stub_log on
[esp32c3] Stub logs enabled!
[esp32c3] Found 8 triggers
Failed to get flash maps (-1)!
Failed to get flash mappings (-4)!
Failed to get flash size!
Using flash bank 'esp32c3.irom' size 0 KB
Failed to get flash maps (-1)!
Failed to get flash mappings (-4)!
Failed to get flash size!
Using flash bank 'esp32c3.drom' size 0 KB
Failed to get flash maps (-1)!
Failed to get flash mappings (-4)!
Failed to get flash size!
Failed to get flash size!
Failed to probe flash, size 0 KB
auto_probe failed
Failed to find bank 'esp32c3.flash'!

\> halt
\> flash probe 0 
Failed to get flash maps (-1)!
Failed to get flash mappings (-4)!
Failed to get flash size!
Failed to get flash size!
Failed to probe flash, size 0 KB

Thank you very much for looking into it!

araldit, 2023-01-18:
This is the output from when I flash using serial:

Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:54:75:89:bc:c8
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x000fdfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Erasing flash...

araldit, 2023-01-18:
Another funny observation. If the esp32c3 is in bootloader mode it seems JTAG programming is working. The firmware will not boot correctly though, but the openocd can erase, flash and verify successfully.

erhankur, 2023-01-18:
I tested your application. I can say, it is OK, not breaks the JTAG communication. 
Did you try with another usb cable or usb hub? Can you send the photo of your setup?

araldit, 2023-01-18:
[image]
The JTAG pins is directly connected to a connector. There are no other components in between.
The USB cable is attached to a 8 port USB hub - I will try to leave that one out.

developillo, 2023-01-23:
Hi, i don't know it that can help. i had the same problem and log. 
almost gave up, until i tried to change the state of GPIO8
if GPIO8 is low at power on, it will never connect with the same debug log of this topic (USB-JTAG embedded port).
if GPIO8 is hi it works.
GPIO 9 is hi and GPIO2 is float on my application...
we checked all espressif documentation but there's no reason to explain such behaviour

erhankur, 2023-01-23:
> Hi, i don't know it that can help. i had the same problem and log. almost gave up, until i tried to change the state of GPIO8 if GPIO8 is low at power on, it will never connect with the same debug [...]

@developillo this is an interesting finding.  Adding hw expert for the comment. @Spritetm

araldit, 2023-01-23:
This also worked for me! 

Thank you so much developillo!

developillo, 2023-01-23:
i hope espressif know reason for such behaviour.. 
we lost days of work on this problem here...

CharlesScoville, 2023-01-26:
> Hi, i don't know it that can help. i had the same problem and log. almost gave up, until i tried to change the state of GPIO8 if GPIO8 is low at power on, it will never connect with the same debug [...]

The information on this is in the ESP32-C3 datasheet in the section [2.4 Strapping Pins.](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf#b)

developillo, 2023-01-27:
> The information on this is in the ESP32-C3 datasheet in the section [2.4 Strapping Pins.](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf#b)

Hi, of course i saw such page. that's why, if you notice, i included the status of GPIO2-8-9 in my post.

CharlesScoville, 2023-01-28:
>Hi, of course i saw such page. that's why, if you notice, i included the status of GPIO2-8-9 in my post.
but still it doesn't explain why the usb-jtag is not working if GPIO8 is low at power-on.

The fact that you, yourself, called attention to GPIO2-8-9 in this problem's thread says all that needs to be said about this reply; you wouldn't have mentioned it if it wasn't on your mind as a suspect for problem. The fact that you are now suddenly dismissing it does not sound like my error. It sounds more like cognitive disconnect on your part.

>We're here to find out a problem not to just link a document and don't understand the root reason of the problem.

Sure! ... and OP's problem may or may not have anything to do with strapping of GPIO2-8-9. However, since *YOU* brought it up, I felt compelled to bring more to the table regarding the subject. Namely, the documentation discussing it in detail.

Specifically, according to the linked section of the data sheet, GPIO8 and GIPO9 cannot both be strapped to ground/low/0, or it will cause undefined behavior. Which sounds suspiciously like what's happening. And the fact that you and one other person had this same problem, and corrected it with respect to this very information, only further supports my assertion. 

Furthermore, I would like to add that there is honestly almost no compelling reason to even strap GPIO8 to ground in the first place. Doing so is asking for troubles with respect to the mentioned undefined behavior. Every devboard and application schematic has GPIO8 strapped to 3v3/high/1, why on earth would any rational developer change this?

So, I ask rhetorically, which one of us really doesn't understand the root reason of the problem?

>The problem is open since August 2022, be realistic, you have a problem.

I am not with espressif, and my devboard works fine. I have no problem, and it wouldn't be *my* problem either way.

CharlesScoville, 2023-01-28:
For any rational people looking for closure on the issue of how precisely to handle the pin strapping issue, I have a hot tip. . .

If you use 4k7-10K Ohm resistors to tie GPIO2 and GPIO8 to 3v3, you can then dedicate these pins to I2C master SCL and SDA functions without fear that there will be a lurking strapping/boot problem!

This is true because, by the specs, I2C slaves/subordinates are not allowed to pull either of these lines down without express permission from the I2C master/commander. Even more convenient is that you already have to tie up SDA and SCL anyway to make the I2C bus function at all. Two birds, one stone.

Having solved GPIO2 and GPIO8, it becomes trivial to deal with GPIO9; it's just the expected boot function.