espressif/openocd-esp32 #243: OpenOCD cannot get flash map when using mcuboot instead of esp-idf bootloader

jaybuckeye06, 2022-08-28:
### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Mac OS X

### Using an IDE ?

n/a

### OpenOCD command line

openocd -f board/esp32-wrover-kit-3.3v.cfg -c "init; halt; esp32 appimage_offset 0x10000"

### ESP-IDF version

v4.2

### Problem Description

- When using esp-idf image format, openocd gets flash map just fine.

[image]

- When using mcuboot, although the partition table keeps the same, openocd cannot find flash map.
[image]

_This is more of a feature request rather than an issue._

### Expected behavior

OpenOCD should be able to find the flash map no matter what bootloader is used.

gerekon, 2022-09-01:
Hi @jaybuckeye06 

OpenOCD retrieves flash map from app image on flash. As far as I understand mcuboot uses different header format for app images on flash as compared to IDF bootloader. So I do not think this is a bug.

erhankur, 2023-12-29:
Looks like in the near future, we do not have plans to support MCUboot image format. 

Also, AFAIK there is no table to get flash mapping from the image like we do for the IDF bootloader. It is written only in the linker.

closed 2023-12-29 by erhankur