espressif/openocd-esp32 #244: How do I build the stubs??

johnboiles, 2022-08-30:
### Development Kit

ESP32-S3R8

### Module or chip used

Raw chip

### Debug Adapter

Builtin

### OpenOCD version

e70f59eee

### Operating System

macOS

### Using an IDE ?

No

### OpenOCD command line

make

### JTAG Clock Speed

n/a

### ESP-IDF version

v4.4.2

### Problem Description

I'm trying to fix [an issue](https://github.com/espressif/openocd-esp32/blob/e70f59eee961319571e1f741cb467be68eb58ef1/contrib/loaders/flash/esp/stub_flasher.c#L684) with the stub where the stub does not work with flash sizes >16MB. But I'm really struggling to get the stubs built. Is there documentation on how to build the stubs?

Let's start simply with this question. What IDF version should be used when building the stubs in `contrib/loaders/flash/esp`. Looks like all the paths in `contrib/loaders/flash/esp/stub_common.mk` aren't consistent with paths that exist in IDF v4.4.2 so it must be some other version.

I'm happy to write/contribute docs here, assuming I can get this to build once!

### Expected behavior

There should be a documented way to build the stubs.

erhankur, 2022-08-31:
Hi @johnboiles , 

You can follow the below steps;

1 - cd openocd-esp32/contrib/loaders/flash/esp/esp32s3
2- run esp-idf/export.sh
3- make sure to checkout specific esp-idf commit written here. https://github.com/espressif/openocd-esp32/blob/master/contrib/loaders/flash/esp/esp32s3/stub_flasher_image.h#L6 
4- make clean and make

johnboiles, 2022-08-31:
Ah that's a helpful comment! Thanks, I'll try it!

johnboiles, 2022-09-01:
Sure. I'd like to document this in a README file possibly in `contrib/loaders/flash/esp` should I put that in my existing PR, or a new PR. That comment in [stub_flasher_image.h#L6](https://github.com/espressif/openocd-esp32/blob/master/contrib/loaders/flash/esp/esp32s3/stub_flasher_image.h#L6) is key to being able to build stubs. README might be a more discoverable way to point people to the right place.

Something potentially as simple as this could help folks like me find their way
```
To build these stubs you'll need to checkout a specific version of IDF. You can find that in the `stub_flasher_image.h` for the chip you're building for.
```

erhankur, 2022-09-01:
Thanks for the suggestion. Feel free to add README into your PR.

closed 2022-09-01 by erhankur