espressif/openocd-esp32 #246: JTAG Debug buffer have been optimized out

anvv0202, 2022-09-05:
### OpenOCD version

v0.11.0-esp32-20220411

### Operating System

esp32-wrover-kit-3.3v

### Using an IDE ?

VS Code

### OpenOCD command line

clear & start openocd -d3 -c \"set ESP_RTOS none\" -f board/esp32-wrover-kit-3.3v.cfg & exit

### ESP-IDF version

4.4

### Problem Description

1. clear & start openocd -d3 -c \"set ESP_RTOS none\" -f board/esp32-wrover-kit-3.3v.cfg & exit
2. launch.js
                {"text": "target remote 127.0.0.1:3333"},
                {"text": "set remote hardware-watchpoint-limit 2"},
                {"text": "monitor reset halt"},
                {"text": "flushregs"},
                {"text": "mon program_esp build/bootloader/bootloader.bin 0x1000 verify"},
                {"text": "mon program_esp build/partition_table/partition-table.bin 0x8000 verify"},
                {"text": "mon program_esp build/proj.bin 0x10000 verify"},
                {"text": "monitor reset halt"},
                {"text": "flushregs"}

### Expected behavior

the array is minimized

### Screenshots

[image]

gerekon, 2022-09-05:
Hi @anvv0202 

I do not think this is related to OpenOCD. Looks like GCC optimized that variable. I'd suggest to try building you project with `-O0 -g` GCC flags.

anvv0202, 2022-09-05:
where can i add `-O0 -g`?

igrr, 2022-09-05:
If you are building your project using ESP-IDF, you can enable CONFIG_COMPILER_OPTIMIZATION_NONE option: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-compiler-optimization

closed 2023-01-17 by erhankur