espressif/openocd-esp32 #247: Erase flash using openocd-esp32

Wasabinary, 2022-09-09:
### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Windows

### Using an IDE ?

Eclipse

### OpenOCD command line

openocd -f interface/jlink.cfg -f target/esp32s2.cfg -c "..."

### ESP-IDF version

v4.4

### Problem Description

I have been trying to erase the whoel flash of my esp32s2 using a J-link and openocd-esp32. I was looking for the right command to use on the Internet, and found a few people trying to do the same thing as I do using *openocd -f interface/jlink.cfg -f target/esp32s2.cfg -c " flash erase_address ..."* or *openocd -f interface/jlink.cfg -f target/esp32s2.cfg -c " flash erase_sector ..."*

When I try using one of these 2 commands, it seems like they are not recognized by openocd-esp32 v0.11.0-esp32-20211220. So my question is simple: which command should i use to erase the flash of my esp32 using a J-link and openocd ?

### Expected behavior

The command "flash erase_sector" or "flash erase_address" is recognized by openocd-esp32

### Screenshots

[image]

erhankur, 2022-09-09:
You can try `flash erase_sector 0 1 last` to erase whole flash.

```
> flash erase_sector 0 1 last
[esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
PROF: Erased 16773120 bytes in 54276.2 ms
erased sectors 1 through 4095 on flash bank 0 in 54.276268s
```

Wasabinary, 2022-09-09:
Thanks for your answer @erhankur 
Unfortunately, i get the following output: *Error: invalid subcommand "erase_sector 0 1 last"*
It really seems like *erase_sector* is not a valid openocd-esp32 command:
[image]

Wasabinary, 2022-09-09:
I was missing that part, now the logs look like the chip is erased. Thank you very much @erhankur ! Closing the issue

Wasabinary, 2022-09-13:
Hi @erhankur,

I have just have another question about the erasing procedure.

I successfully erased my ESP32-S2 chip with the command you sent above. Basically, I flashed a firmware using openocd-esp32, and then erase it straight after with `-c "flash init; init; flash erase_sector 0 1 last"`.
However, If I power my board and try to erase it just after, I get an error:
[image]
I was expecting the command to work even though I don't flash something just before. Why do I get the "Error: Target not halted" and "Error: auto_probe failed" messages ?

erhankur, 2022-09-13:
Before any flash operation OpenOCD tries to get flash parameters by calling probe function. This function runs on the halted target. 

I am not sure why it fails in the consecutive erase. You can  add `halt` command just before `flash erase`  

```
-c "flash init; init; halt; flash erase_sector 0 1 last" 
```

Wasabinary, 2022-09-13:
Ok thank you, it partially resolved this issue, as I get "Error: Invalid stub" for some of the chips I want to erase... So I still have an issue somewhere I guess

Wasabinary, 2022-09-13:
Here's the log:
[debug.txt](https://github.com/espressif/openocd-esp32/files/9558455/debug.txt)
You can find the "Invalid stub!" error multiple time here

erhankur, 2022-09-13:
Can  I ask you to update your OpenOCD (master or release) to the latest version and check that the issue is present there? 

If still exists, please send the new log.txt and any file that you modified such as `esp32s2.cfg`

Wasabinary, 2022-09-14:
Well, with the latest openocd-esp32, I have a new error. Nothing to do with "Invalid stub", but this: "Expression: (first <= last) && (last < bank->num_sectors)"

[debug.txt](https://github.com/espressif/openocd-esp32/files/9563595/debug.txt)

Probably something to do with flash bank sector number issue ?

erhankur, 2022-09-15:
Please try `-c "flash init; init; reset halt; flash erase_sector 0 1 last"`

Wasabinary, 2022-09-15:
Thank you, it seems like it works normally now. Thank you for your help

closed 2022-09-15 by Wasabinary