espressif/openocd-esp32 #250: OpenOCD issue: Error: invalid command name "espusbjtag"

h2oFanti, 2022-09-28:
### Development Kit

ESP32-S2-WROOM (Brand: Gravitech - Cucumber Model M)

### Module or chip used

ESP32 S2 WROOM

### Debug Adapter

STLink/V2

### OpenOCD version

v0.11.0-esp32-20220706

### Operating System

Windows 10 Pro 21H2 19044.2006

### Using an IDE ?

VSCode ESP-IDF Terminal

### OpenOCD command line

openocd -f interface/stlink.cfg -f board/esp32s2-bridge.cfg

### JTAG Clock Speed

adapter speed 40000

### ESP-IDF version

v4.4

### Problem Description

1. Open idf terminal
2. Run openocd -f interface/stlink.cfg -f board/esp32s2-bridge.cfg 

### Debug Logs

```shell
>openocd -f interface/stlink.cfg -f board/esp32s2-bridge.cfg -d
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 6 options.c:63 configuration_output_handler(): debug_level: 3
User : 4 13 options.c:63 configuration_output_handler(): 
Debug: 5 21 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 72 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 84 options.c:246 add_default_dirs(): exepath=C:/Users/Annie/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin
Debug: 8 108 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 9 114 configuration.c:44 add_script_search_dir(): adding C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts
Debug: 10 128 configuration.c:44 add_script_search_dir(): adding C:/Users/Annie/AppData/Roaming/OpenOCD
Debug: 11 137 configuration.c:44 add_script_search_dir(): adding C:/Users/Annie/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin/../share/openocd/site
Debug: 12 151 configuration.c:44 add_script_search_dir(): adding C:/Users/Annie/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin/../share/openocd/scripts
Debug: 13 162 command.c:166 script_debug(): command - ocd_find interface/stlink.cfg
Debug: 14 169 configuration.c:99 find_file(): found C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/interface/stlink.cfg
Debug: 15 182 command.c:166 script_debug(): command - adapter driver hla
Debug: 16 188 command.c:166 script_debug(): command - hla_layout stlink
Debug: 17 192 hla_interface.c:241 hl_interface_handle_layout_command(): hl_interface_handle_layout_command
Debug: 18 201 command.c:166 script_debug(): command - hla_device_desc ST-LINK
Debug: 19 205 hla_interface.c:228 hl_interface_handle_device_desc_command(): hl_interface_handle_device_desc_command
Debug: 20 213 command.c:166 script_debug(): command - hla_vid_pid 0x0483 0x3744 0x0483 0x3748 0x0483 0x374b 0x0483 0x374d 0x0483 0x374e 0x0483 0x374f 0x0483 0x3752 0x0483 0x3753 0x0483 0x3754
Debug: 21 268 command.c:166 script_debug(): command - ocd_find board/esp32s2-bridge.cfg
Debug: 22 274 configuration.c:99 find_file(): found C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/board/esp32s2-bridge.cfg
Debug: 23 288 command.c:166 script_debug(): command - ocd_find interface/esp_usb_bridge.cfg
Debug: 24 306 configuration.c:99 find_file(): found C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/interface/esp_usb_bridge.cfg
Debug: 25 319 command.c:166 script_debug(): command - adapter driver esp_usb_jtag
Warn : 26 324 adapter.c:364 handle_adapter_driver_command(): Interface already configured, ignoring
User : 27 331 command.c:619 command_run_line(): C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/interface/esp_usb_bridge.cfg:8: Error: invalid command name "espusbjtag"
in procedure 'script'
at file "embedded:startup.tcl", line 26
at file "C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/board/esp32s2-bridge.cfg", line 11
at file "C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/interface/esp_usb_bridge.cfg", line 8
```

### Expected behavior

GDB console output/debugging session to open

erhankur, 2022-09-28:
To connect ESP32-S2 from jtag using stlink, you can try 
```
openocd -f interface/stlink.cfg -f target/esp32s2.cfg
```

h2oFanti, 2022-09-28:
> To connect ESP32-S2 from jtag using stlink, you can try
> 
> ```
> openocd -f interface/stlink.cfg -f target/esp32s2.cfg
> ```

I hadn't tried target/esp32s2.cfg, but no dice.

"Debug adapter doesn't support 'jtag' transport" ?

Maybe I have the wrong driver? I've looked over some past issues, I've tried a handful of drivers. I'm currently on WinUSB (v2.2.0.0) for stlink, and FTDIBUS (v2.12.36.4) for COM, according to zadig.  

`openocd -f interface/stlink.cfg -f target/esp32s2.cfg -d
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 5 options.c:63 configuration_output_handler(): debug_level: 3
User : 4 12 options.c:63 configuration_output_handler():
Debug: 5 19 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 29 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 38 options.c:246 add_default_dirs(): exepath=C:/Users/Annie/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin
Debug: 8 97 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 9 102 configuration.c:44 add_script_search_dir(): adding C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts
Debug: 10 125 configuration.c:44 add_script_search_dir(): adding C:/Users/Annie/AppData/Roaming/OpenOCD
Debug: 11 143 configuration.c:44 add_script_search_dir(): adding C:/Users/Annie/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin/../share/openocd/site
Debug: 12 155 configuration.c:44 add_script_search_dir(): adding C:/Users/Annie/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin/../share/openocd/scripts
Debug: 13 169 command.c:166 script_debug(): command - ocd_find interface/stlink.cfg
Debug: 14 175 configuration.c:99 find_file(): found C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/interface/stlink.cfg
Debug: 15 188 command.c:166 script_debug(): command - adapter driver hla
Debug: 16 194 command.c:166 script_debug(): command - hla_layout stlink
Debug: 17 199 hla_interface.c:241 hl_interface_handle_layout_command(): hl_interface_handle_layout_command
Debug: 18 204 command.c:166 script_debug(): command - hla_device_desc ST-LINK
Debug: 19 211 hla_interface.c:228 hl_interface_handle_device_desc_command(): hl_interface_handle_device_desc_command
Debug: 20 219 command.c:166 script_debug(): command - hla_vid_pid 0x0483 0x3744 0x0483 0x3748 0x0483 0x374b 0x0483 0x374d 0x0483 0x374e 0x0483 0x374f 0x0483 0x3752 0x0483 0x3753 0x0483 0x3754
Debug: 21 232 command.c:166 script_debug(): command - ocd_find target/esp32s2.cfg
Debug: 22 237 configuration.c:99 find_file(): found C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/target/esp32s2.cfg
Debug: 23 249 command.c:166 script_debug(): command - transport select jtag
Error: 24 253 transport.c:318 jim_transport_select(): Debug adapter doesn't support 'jtag' transport
User : 25 261 command.c:619 command_run_line(): C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/target/esp32s2.cfg:4: Error: 
in procedure 'script'
at file "embedded:startup.tcl", line 26
at file "C:\Users\Annie\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/target/esp32s2.cfg", line 4

h2oFanti, 2022-09-29:
I can only assume that JTAG with this adapter for this chipset isn't supported - quoting the docs:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html#jtag-debugging-configure-and-connect-jtag-interface

> JTAG adapters that are hardcoded to a specific product line, e.g. ST-LINK debugging adapters for STM32 families, will not work.

Though I would like some confirmation from either the esp32 community or ST.

erhankur, 2022-09-29:
Sorry I wasn't aware of the stlink limitation. One of the user wrote a comment about this. https://github.com/espressif/openocd-esp32/issues/145#issuecomment-791395399

h2oFanti, 2022-09-30:
> Sorry I wasn't aware of the stlink limitation. One of the user wrote a comment about this. [#145 (comment)](https://github.com/espressif/openocd-esp32/issues/145#issuecomment-791395399)

Thanks @erhankur 

Wow that's frustrating. Guess I'll just... buy new hardware. _Thanks ST!_ Why would they do that?

closed 2022-09-30 by h2oFanti