espressif/openocd-esp32 #251: Problem with JTAG

sYCH3L, 2022-09-30:
### Development Kit

ESP32-S3-USB-OTG

### Module or chip used

ESP32-S3

### Debug Adapter

jlink

### OpenOCD version

v0.11.0-esp32-20220706-218-gbf177879

### Operating System

Linux

### Using an IDE ?

No

### OpenOCD command line

openocd -f esp32_jlink

### JTAG Clock Speed

4000

### ESP-IDF version

v5.1-dev-992-gaf28c1fa21

### Problem Description

Works weird without -d3 in openocd launch.

Stops at s_rtc function not app_main
Error with cpenable
Error with flash mappings.

[gdb_log_unverbose.txt](https://github.com/espressif/openocd-esp32/files/9683709/gdb_log_unverbose.txt)
[openocd_log_unverbose.txt](https://github.com/espressif/openocd-esp32/files/9683710/openocd_log_unverbose.txt)
[openocd_log_verbose.txt](https://github.com/espressif/openocd-esp32/files/9683712/openocd_log_verbose.txt)
[gdb_log_verbose.txt](https://github.com/espressif/openocd-esp32/files/9683713/gdb_log_verbose.txt)

openocd configuration file.
`source [find interface/jlink.cfg]

adapter speed 4000

set ESP32_RTOS FreeRTOS

source [find target/esp32s3.cfg]
`

### Debug Logs

```shell
Logs are attached.
```

### Expected behavior

Should run to app_main without errors in console.

sYCH3L, 2022-09-30:
JTAG is accessed from GPIO pins and according eFuses are burned DIS_USB_JTAG and USB_PHY_SEL

erhankur, 2023-05-04:
@sYCH3L sorry for the late response. We did many changes at the Xtensa related codes. Can you please try with the latest OpenOCD release?

erhankur, 2023-06-12:
Closed due to inactivity. Feel free to reopen if you still have a problem.

closed 2023-06-12 by erhankur