espressif/openocd-esp32 #252: Debug JTAG USB port of ESP32-WROVER-KIT gets lost after successful start of openocd

volkmarnissen, 2022-09-30:
### Development Kit

ESP32

### Module or chip used

ESP32-WROOM-KIT

### Debug Adapter

ESP32-WROVER-KIT

### OpenOCD version

v0.11.0-esp32-20220706

### Operating System

Linux

### Using an IDE ?

Yes, but it's reproducible with command line

### OpenOCD command line

openocd -f board/esp32-wrover-kit-3.3v.cfg

### JTAG Clock Speed

default

### ESP-IDF version

4.3.4

### Problem Description

- Connect ESP32-WROVER-KIT board to USB
- ls /dev/ttyUSB* will show two entries
- call openocd -f board/esp32-wrover-kit-3.3v.cfg
   It establishes the tcp port 3333 correctly
-  Press ^C to abort the session
- ls /dev/ttyUSB* will show one entry the /dev/ttyUSB0 (Connection to JTAG) is missing now.
This is the issue:
 A next call to ocddebug will fail, because the device is lost.

- Unplug the USB cable
- Plug it in

Now, you'll see two ports again.

### Debug Logs

```shell
$ ls /dev/ttyUS*
/dev/ttyUSB0  /dev/ttyUSB1
$ openocd -f board/esp32-wrover-kit-3.3v.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unexpected OCD_ID = ffffffff
Warn : target esp32.cpu0 examination failed
Error: Unexpected OCD_ID = ffffffff
Warn : target esp32.cpu1 examination failed
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
^Cshutdown command invoked

$ ls /dev/ttyUS*
/dev/ttyUSB1
```

### Expected behavior

A call to openocd should not stop the second USB port.

Luckily there is a workaround, but it's cumbersome.

erhankur, 2022-10-03:
I see same behaviour on linux. But this is not a problem for the jtag communication. Or in other words, after connection established with the OpenOCD, ttyUSB0 is not seen in the device list but I can still communicate via telnet e.g. 

Does your application, configuring the jtag pins as gpio?  Did you try with `hello_world` application?

volkmarnissen, 2022-10-03:
Hi,

after playing around with it, one time I was able to debug via JTAG, but I don't know, why it worked. I was not able to reproduce it afterwards. I'll try it with ESP-prog to see whether this is different.

Thank you,

Volkmar

erhankur, 2023-01-17:
@volkmarnissen I am closing this issue. Feel free to re-open after your test.

closed 2023-01-17 by erhankur