espressif/openocd-esp32 #253: Help with debugging

guirespi, 2022-10-07:
### Development Kit

ESP32-WROVER-KIT

### Module or chip used

ESP32-WROVER

### Debug Adapter

 FT2232 

### OpenOCD version

v0.11.0-esp32-20220706

### Operating System

Windows

### Using an IDE ?

Eclipse

### OpenOCD command line

Not sure

### JTAG Clock Speed

20000 Kh

### ESP-IDF version

v4.4.2

### Problem Description

Start debug locally with v0.11.0-esp32-20220706 configured in Eclipse IDE.

### Debug Logs

```shell
Debug: 7491 17942 command.c:166 script_debug(): command - echo Started by GNU MCU Eclipse
User : 7492 17948 command.c:696 handle_echo(): Started by GNU MCU Eclipse
Info : 7493 17948 server.c:303 add_service(): Listening on port 6666 for tcl connections
Info : 7494 17948 server.c:303 add_service(): Listening on port 4444 for telnet connections
Debug: 7495 17948 command.c:166 script_debug(): command - init
Debug: 7496 20480 server.c:609 sig_handler(): Terminating on Signal 2
Debug: 7497 20574 command.c:166 script_debug(): command - shutdown
User : 7498 20580 server.c:755 handle_shutdown_command(): shutdown command invoked
Debug: 7499 20580 esp_xtensa.c:133 esp_xtensa_target_deinit(): start
Debug: 7500 20580 xtensa.c:2476 xtensa_target_deinit(): start
Debug: 7501 20581 target.c:2239 target_free_all_working_areas_restore(): freeing all working areas
Debug: 7502 20581 target.c:1986 print_wa_layout():    0x40090000-0x40093fff (16384 bytes)
Debug: 7503 20581 esp_xtensa.c:133 esp_xtensa_target_deinit(): start
Debug: 7504 20581 xtensa.c:2476 xtensa_target_deinit(): start
Debug: 7505 20582 target.c:2239 target_free_all_working_areas_restore(): freeing all working areas
```

### Expected behavior

Start the debugging session.

### Screenshots

[image]
[image]

guirespi, 2022-10-07:
Also, I would like to share the error message from eclipse :
[image]

tasha-davis, 2023-06-22:
Were you able to fix this error? I am having the same problem now

erhankur, 2024-01-15:
Closed due to inactivity. If you still have an issue please re-open and share the verbose OpenOCD logs (-d3).

closed 2024-01-15 by erhankur