espressif/openocd-esp32 #254: ESP32 jtag with st-link not working

hannescam, 2022-10-16:
### Development Kit

Wemos D1 mini ESP32

### Module or chip used

ESP32-wroom

### Debug Adapter

st-link

### OpenOCD version

v0.11.0-esp32-20220706

### Operating System

Archlinux

### Using an IDE ?

No

### OpenOCD command line

bin/openocd -s share/openocd/scripts -f interface/stlink-dap.cfg -c "transport select dapdirect_jtag" -f share/openocd/scripts/board/esp32-wrover.cfg

### JTAG Clock Speed

default

### ESP-IDF version

PlatformIO in Arduino mode

### Problem Description

just run the command

### Debug Logs

```shell
[hannes@Hannes-GUSCHTAV openocd-esp32]$ bin/openocd -s share/openocd/scripts -f interface/stlink-dap.cfg -c "transport select dapdirect_jtag" -f share/openocd/scripts/board/esp32-wrover.cfg 
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
dapdirect_jtag
WARNING: boards/esp32-wrover.cfg is deprecated, and may be removed in a future release.
         If your board is ESP32-WROVER-KIT, use board/esp32-wrover-kit-1.8v.cfg instead.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : STLINK V2J39S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.136665
Error: init mode failed (unable to connect to the target)

[hannes@Hannes-GUSCHTAV openocd-esp32]$
```

### Expected behavior

run without error

### Screenshots

[image]

hannescam, 2022-10-16:
i use the st-link v2: 
[image]
on the newest firmware version
and i hope this is a issue that can be fixed

erhankur, 2023-06-12:
Closing since we have nothing to do for now.

closed 2023-06-12 by erhankur