espressif/openocd-esp32 #255: ESP32-S3 erroneous "Error: 330 126 libusb_helper.c:190 jtag_libusb_open(): libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED"

Panometric, 2022-10-25:
### Development Kit

NA, custom

### Module or chip used

ESP32-S3-WROOM-1

### Debug Adapter

Builtin

### OpenOCD version

v0.11.0-esp32-20220411 (2022-04-11-08:48)

### Operating System

Windows

### Using an IDE ?

no

### OpenOCD command line

openocd -d3 -f board/esp32-wrover-kit-3.3v.cfg 2>&1 | tee openocd.log

### JTAG Clock Speed

default 40000

### ESP-IDF version

4.4.2

### Problem Description

Erroneous error on startup: "Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED"

PS C:\DCI\spi_lcd\lcd> openocd  -f board/esp32s3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
**Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED**
Info : esp_usb_jtag: serial (F4:12:FA:C1:A1:64)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections

### Debug Logs

```shell
openocd : Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)
At line:1 char:1
+ openocd -d3 -f board/esp32s3-builtin.cfg 2>&1 | tee openocd.log
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (Open On-Chip De...22-04-11-08:48):String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
[... 502 lines cut ...]
		}
	
Debug: 418 185 command.c:166 script_debug(): command - arm semihosting enable
Debug: 419 187 esp_xtensa_semihosting.c:46 esp_xtensa_semihosting_setup(): [esp32s3.cpu1] semihosting enable=1
Debug: 420 201 command.c:166 script_debug(): command - arm semihosting_resexit enable
Debug: 421 203 command.c:166 script_debug(): command - arm semihosting_basedir .
Debug: 422 205 esp32s3.c:593 esp32s3_handle_target_event(): 21
Debug: 423 205 esp_xtensa_smp.c:609 esp_xtensa_smp_handle_target_event(): 21
Debug: 424 206 esp_xtensa.c:79 esp_xtensa_handle_target_event(): 21
Debug: 425 206 xtensa.c:2450 xtensa_handle_target_event(): 21
Debug: 426 206 xtensa.c:726 xtensa_smpbreak_write(): esp32s3.cpu1: write smpbreak set=0x30000 clear=0x600000
Debug: 427 206 command.c:166 script_debug(): command - flash init
Debug: 428 208 tcl.c:1386 handle_flash_init_command(): Initializing flash devices...
Debug: 429 208 command.c:166 script_debug(): command - nand init
Debug: 430 210 tcl.c:498 handle_nand_init_command(): Initializing NAND devices...
Debug: 431 210 command.c:166 script_debug(): command - pld init
Debug: 432 213 pld.c:205 handle_pld_init_command(): Initializing PLDs...
Debug: 433 213 command.c:166 script_debug(): command - tpiu init
Info : 434 215 gdb_server.c:3742 gdb_target_start(): starting gdb server for esp32s3.cpu0 on 3333
Info : 435 215 server.c:303 add_service(): Listening on port 3333 for gdb connections
```

### Expected behavior

Do not report an Error if connection is OK.

Panometric, 2022-11-13:
Can anyone help on this, it's so annoying.. I CAN debug, but the vscode extension pops up this error every session.

erhankur, 2022-11-14:
cc @kolipakakondal

kolipakakondal, 2022-11-14:
Hi @Panometric 

> libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED

This is ok for windows and the JTAG connection should work even with this.  Looking at the above console log messages I don't see any other error.

However, run https://visualgdb.com/UsbDriverTool/ instead of Zadig for installation of libusb and see if that helps to avoid the above error as well.

erhankur, 2023-01-16:
Hi @Panometric I have updated libusb version to 1.0.26. Looks like the error disappeared in my windows machine.  (Win10)

Can you also test and verify with the attached package? 

[openocd-esp32-win32-0.11.0-esp32-20221026-87-gd269916f9.zip](https://github.com/espressif/openocd-esp32/files/10427063/openocd-esp32-win32-0.11.0-esp32-20221026-87-gd269916f9.zip)

erhankur, 2023-01-16:
```
c:\esp_\openocd-esp32\bin>openocd.exe -s ..\share\openocd\scripts -f board/esp32s3-builtin.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026-87-gd269916f (2023-01-16-14:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:E1:19:00)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x4037A9E0, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] Target halted, PC=0x4037A9E0, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
```

Panometric, 2023-01-16:
@erhankur  Just updating the driver didn't do it, but the package did. Where/ when will this be officially released?  Thanks!

erhankur, 2023-01-16:
Glad it worked for you, too. The official release date is not decided yet. Until release, you can just replace openocd-esp32 folder under `.espressif\tools\openocd-esp32\<your version>\`  We have backward compatibility between idf versions. So
 there should be no new issue.

sirapol, 2023-06-27:
Hello I found same issues.

OS : Ubuntu
VScode
ESP-IDF : 5.2.0
Hardware : ESP32-2424S012 // ESP32-C3 with Round display

`/home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/openocd

Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

/home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/openocd

[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 1 options.c:52 configuration_output_handler(): debug_level: 3User : 4 1 options.c:52 configuration_output_handler(): 
Debug: 5 1 options.c:233 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 1 options.c:234 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 1 options.c:235 add_default_dirs(): exepath=/home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin
Debug: 8 1 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 1 configuration.c:33 add_script_search_dir(): adding /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
Debug: 10 1 configuration.c:33 add_script_search_dir(): adding /home/sirapol/.config/openocd
Debug: 11 1 configuration.c:33 add_script_search_dir(): adding /home/sirapol/.openocd
Debug: 12 1 configuration.c:33 add_script_search_dir(): adding /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/site
Debug: 13 1 configuration.c:33 add_script_search_dir(): adding /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts
Debug: 14 1 command.c:152 script_debug(): command - ocd_find board/esp32c3-builtin.cfg
Debug: 15 1 configuration.c:88 find_file(): found /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/board/esp32c3-builtin.cfg
Debug: 16 1 command.c:152 script_debug(): command - ocd_find interface/esp_usb_jtag.cfg
Debug: 17 1 configuration.c:88 find_file(): found /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/interface/esp_usb_jtag.cfg
Debug: 18 1 command.c:152 script_debug(): command - adapter driver esp_usb_jtag

[OpenOCD]
Info : 19 1 transport.c:107 allow_transports(): only one transport option; autoselecting 'jtag'
Debug: 20 1 command.c:152 script_debug(): command - espusbjtag vid_pid 0x303a 0x1001
Info : 21 1 esp_usb_jtag.c:894 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Debug: 22 1 command.c:152 script_debug(): command - espusbjtag caps_descriptor 0x2000
Info : 23 1 esp_usb_jtag.c:905 esp_usb_jtag_caps_descriptor(): esp_usb_jtag: capabilities descriptor set to 0x2000
Debug: 24 1 command.c:152 script_debug(): command - adapter speed 40000
Debug: 25 1 adapter.c:253 adapter_config_khz(): handle adapter khz
Debug: 26 1 adapter.c:217 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 27 1 adapter.c:217 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 28 1 command.c:152 script_debug(): command - ocd_find target/esp32c3.cfg
Debug: 29 1 configuration.c:88 find_file(): found /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp32c3.cfg
Debug: 30 1 command.c:152 script_debug(): command - transport select jtag
Warn : 31 1 transport.c:280 handle_transport_select(): Transport "jtag" was already selected
Debug: 32 1 command.c:152 script_debug(): command - ocd_find bitsbytes.tcl
Debug: 33 1 configuration.c:88 find_file(): found /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/bitsbytes.tcl
Debug: 34 2 command.c:152 script_debug(): command - ocd_find memory.tcl
Debug: 35 2 configuration.c:88 find_file(): found /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/memory.tcl
Debug: 36 2 command.c:152 script_debug(): command - ocd_find mmr_helpers.tcl
Debug: 37 2 configuration.c:88 find_file(): found /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/mmr_helpers.tcl
Debug: 38 2 command.c:152 script_debug(): command - ocd_find target/esp_common.cfg
Debug: 39 2 configuration.c:88 find_file(): found /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg
Debug: 40 2 command.c:152 script_debug(): command - ocd_find target/esp_version.cfg
Debug: 41 2 configuration.c:88 find_file(): found /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_version.cfg
Debug: 42 2 command.c:152 script_debug(): command - version
Debug: 43 2 command.c:152 script_debug(): command - add_help_text program_esp write an image to flash, address is only required for binary images. verify, reset, exit, compress, restore_clock and encrypt are optional
Debug: 44 2 command.c:152 script_debug(): command - add_usage_text program_esp <filename> [address] [verify] [reset] [exit] [compress] [no_clock_boost] [restore_clock] [encrypt]
[... cut ...]
User : 118 6 command.c:613 command_run_line(): /home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:4: Error: 
at file "/home/sirapol/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 4
Debug: 119 6 riscv.c:490 riscv_deinit_target(): riscv_deinit_target()
Debug: 120 6 target.c:2223 target_free_all_working_areas_restore(): freeing all working areas

[OpenOCD]
[Stopped] : OpenOCD Server
[Flash]
Can't perform JTag flash, because OpenOCD server is not running!`

Panometric, 2023-06-27:
@sirapol  > LIBUSB_ERROR_ACCESS is not the same error at all . Please delete to  prevent complicating the issue.

sirapol, 2023-10-15:
Thank you everyone