espressif/openocd-esp32 #256: Debug: Failed to write breakpoint instruction (-4)!

MarkStokes71, 2022-10-27:
### Development Kit

ESP-WROVER-KIT-VE equivalent

### Module or chip used

ESP-WROVER

### Debug Adapter

ESP-PROG

### OpenOCD version

 v0.10.0-esp32-20200709 (2020-07-09-08:54)

### Operating System

Win 11

### Using an IDE ?

VS Code and CLion

### OpenOCD command line

start "OpenOCD server status"  call openocd -f openocd_isk.cfg  -c "init; halt; esp appimage_offset 0x600000"

### JTAG Clock Speed

10000k

### ESP-IDF version

ESP-IDF v4.2.1

### Problem Description

We are using OpenOCD and want to get SW breakpoints working. I have tried suggestions online, but OpenOCD seems to revert to HW breakpoints every time. I got the address offset from the isk_partitions.csv from "ota_0". In the OpenOCD log, I'm getting "Failed to add SW breakpoint!" So, it's clearly trying to set the SW breakpoint. I have also tried using 0x600000 for the offset (the address of "ota_1"). Please let me know what I might be doing wrong. All referenced .cfg files are default (within the .espressif folder structure).  I have tried setting the adapter_khz to 10000.

### Debug Logs

```shell
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Configured 2 cores
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset.
Info : esp32: Core 0 was reset.
Info : Listening on port 3333 for gdb connections
Info : Target halted. CPU0: PC=0x402732CE (active)
Info : Target halted. CPU1: PC=0x00000000
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : Empty flash mapping!
Warn : Failed to get flash mappings (0)!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Using flash bank 'esp32.irom' size 0 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Error: Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Using flash bank 'esp32.irom' size 0 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : Empty flash mapping!
Warn : Failed to get flash mappings (0)!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Using flash bank 'esp32.drom' size 0 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Error: Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Using flash bank 'esp32.drom' size 0 KB
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : Empty flash mapping!
Warn : Failed to get flash mappings (0)!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.flash' size 16384 KB
Info : Using flash bank 'esp32.flash' size 16384 KB
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400
Error: Too large number of threads 2779096485!
Error: Too large number of threads 2779096485!
Error: cpu0: Failed to write breakpoint instruction (-4)!
Error: cpu0: Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Error: cpu0: Failed to write breakpoint instruction (-4)!
Error: cpu0: Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Info : dropped 'gdb' connection
```

### Expected behavior

I want SW breakpoints to work.

gerekon, 2022-10-28:
@MarkStokes71 
Hi.
It seems that the problem is because OpenOCD can not read flash mappings from app image on flash. So it can not detected ROM and RAM boundaries, so it falls back to HW breakpoints usage.
```
Warn : Empty flash mapping!
Warn : Failed to get flash mappings (0)!
```
You use `esp appimage_offset 0x600000`. It is correct command. Are you sure that you have valid app image located at that offset?

gerekon, 2022-10-28:
If app image offset is correct I'd suggest to try the latest OpenOCD release (https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20221026). You have two options to use it:
- Overwrite installed OpenOCD. I suppose that gor your system it is located in `.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709`
- Configure your IDE to use custom OpenOCD distro

MarkStokes71, 2022-10-28:
Thanks for the reply.  How do I find the flash mapping offset?  I understood it to be one of the two ota partitions.  I have tried both addresses and they both cause OpenOCD to give the error: 
Error: Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
I did try your updated OpenOCD download (and overwrote my local files).

erhankur, 2022-10-28:
@MarkStokes71 you can print the app offset like below. 

```c
const esp_partition_t *running = esp_ota_get_running_partition();
ESP_LOGI(TAG, "app has started from @%lx", running->address);
```
Can you also copy/paste your partition_table.csv?

MarkStokes71, 2022-10-29:
erhankur,
Thanks for the information.  I was able to verify the flash offset is 0x100000.  I am unable to paste the partition table due to IP concerns.  What in that table are you interested?  With 0x100000 in the offset, I still get the same error in the OpenOCD log.

erhankur, 2022-10-29:
I just wanted to see OTA related partitions size/addresses.
For me, it is strange because in both offsets there should be a valid application (if you did OTA once),  so OpenOCD should skip the mapping step successfully. 

Did you change the working area config settings in your TCL config file? 

This is the default.
```
configure_esp_workarea $_TARGETNAME_0 0x40090000 0x4000 0x3FFC0000 0x18000
```

Can you send the full OpenOCD log with -d3 option?

MarkStokes71, 2022-10-30:
Sure.  Here's the log.  I see the appimage_offset command, but I also see it says the flash is 0kb...  I also have included the .cfg file.
[openocd.cfg.txt](https://github.com/espressif/openocd-esp32/files/9896651/openocd.cfg.txt) <- Added the ".txt" for upload to this forum only).
[OpenOCDLog-d3.txt](https://github.com/espressif/openocd-esp32/files/9896653/OpenOCDLog-d3.txt)
Not sure about any changes to any .tcl files.  All of the files under .espressif are exactly what gerekon send in the .zip (which btw, doesn't show the newer version number).

gerekon, 2022-10-31:
> All of the files under .espressif are exactly what gerekon sent in the .zip (which btw, doesn't show the newer version number).

@MarkStokes71 What zip file do you mean?

gerekon, 2022-10-31:
I see that for some reason the first time OCD reads mappings it uses proper offset and therefore succeeds:
```
Debug: 911 2200 xtensa_algorithm.c:597 xtensa_run_do(): Set arg[0] = 5
Debug: 912 2200 xtensa_algorithm.c:609 xtensa_run_do(): Set arg[1] = 1048576
Debug: 913 2210 xtensa_algorithm.c:609 xtensa_run_do(): Set arg[2] = 0
Debug: 914 2210 xtensa_algorithm.c:108 xtensa_stub_load(): stub: base 0x0, start 
...
Info : 1179 2930 esp_xtensa.c:303 esp_xtensa_get_mappings(): Flash mapping 0: 0x100020 -> 0x3f400020, 610 KB
Info : 1180 2940 esp_xtensa.c:303 esp_xtensa_get_mappings(): Flash mapping 1: 0x1a0020 -> 0x400d0020, 1703 KB
Info : 1181 2940 esp_xtensa.c:780 esp_xtensa_probe(): Using flash bank 'esp32.irom' size 1704 KB
Debug: 1182 2940 esp_xtensa.c:797 esp_xtensa_probe(): allocated 426 sectors
```
The second time it uses default offset (`-1`). So it fails:
```
Debug: 1185 2950 xtensa_algorithm.c:597 xtensa_run_do(): Set arg[0] = 5
Debug: 1186 2950 xtensa_algorithm.c:609 xtensa_run_do(): Set arg[1] = -1
Debug: 1187 2950 xtensa_algorithm.c:609 xtensa_run_do(): Set arg[2] = 0
Debug: 1440 3650 xtensa.c:618 xtensa_core_status_check(): cpu1: DSR (00000000)
Warn : 1441 3650 esp_xtensa.c:297 esp_xtensa_get_mappings(): Empty flash mapping!
Warn : 1442 3650 esp_xtensa.c:735 esp_xtensa_probe(): Failed to get flash mappings (0)!
```
Also I see that sometimes flash size can not be detected and some times it can.
```
Debug: 1444 3660 xtensa_algorithm.c:597 xtensa_run_do(): Set arg[0] = 4
Debug: 1654 4259 esp_xtensa.c:252 esp_xtensa_get_size(): esp_xtensa_get_size size 0x1000000
Info : 1655 4259 esp_xtensa.c:780 esp_xtensa_probe(): Using flash bank 'esp32.drom' size 0 KB
Debug: 1656 4270 esp_xtensa.c:797 esp_xtensa_probe(): allocated 0 sectors
Debug: 1657 4274 esp_xtensa.c:723 esp_xtensa_probe(): Flash size = 0 KB @ 0x01000000 'esp32' - 'halted'
```
```
Debug: 2426 46847 xtensa_algorithm.c:597 xtensa_run_do(): Set arg[0] = 4
Debug: 2636 47699 esp_xtensa.c:252 esp_xtensa_get_size(): esp_xtensa_get_size size 0x1000000
Info : 2637 47703 esp_xtensa.c:778 esp_xtensa_probe(): Auto-detected flash bank 'esp32.flash' size 16384 KB
Info : 2638 47706 esp_xtensa.c:780 esp_xtensa_probe(): Using flash bank 'esp32.flash' size 16384 KB
Debug: 2639 47709 esp_xtensa.c:797 esp_xtensa_probe(): allocated 4096 sectors
Debug: 2640 47712 gdb_server.c:1013 gdb_new_connection(): New GDB Connection: 1, Target esp32, state: halted
```
@MarkStokes71 What flash chip do you use? BTW in your last log I do not see any error related to breakpoints.

MarkStokes71, 2022-11-03:
gerekon, 
Ya, the last log didn't include any actual debugging.  How do I determine which flash chip I'm using?  Here are some of the numbers from the module: ESP32-WROVER-E.  ICID:21098-ESPWROVERE, CMIT ID: 2020DP1674, XX0H28

gerekon, 2022-11-03:
> How do I determine which flash chip I'm using? Here are some of the numbers from the module: ESP32-WROVER-E

Ahh. Ok. I thought you used custom flash chip.

In your latest log I do not see any `Error: cpu0: Failed to add SW breakpoint!` messages. Did the problem disappear?

> Here's a new debug log w/ attempted adding of three and four breakpoints (Clion). Doesn't look like they worked at all (code didn't stop).

From the log I see that OpenOCD successfully set 2 HW and one SW breakpoint in flash.
```
Debug: 3109 103198 xtensa.c:1716 xtensa_breakpoint_add(): cpu0: placed HW breakpoint @ 0x401240cd
Debug: 3110 103198 xtensa.c:1716 xtensa_breakpoint_add(): cpu1: placed HW breakpoint @ 0x401240cd
Debug: 3114 103208 xtensa.c:1716 xtensa_breakpoint_add(): cpu0: placed HW breakpoint @ 0x401240f7
Debug: 3115 103208 xtensa.c:1716 xtensa_breakpoint_add(): cpu1: placed HW breakpoint @ 0x401240f7
Debug: 3435 104228 esp_xtensa.c:923 esp_xtensa_flash_breakpoint_add(): cpu0: Placed flash SW breakpoint at 0x4012411c, insn [00 40 00] 3 bytes
```
But none of the breaks is hit. Are you sure that program really runs to those locations/addresses? Could you verify this by adding, for example print statements just after the locations where you set breakpoints?

MarkStokes71, 2022-11-04:
Ok, it would seem it is only working when I use the -d3 option.  However, the lag introduced in the debug messages makes this pretty unusable.  I've attached four files.  The ones that say Success clearly show SW breakpoints being made and during debug, the breakpoints seem to work correctly.  When debugging with -d3 removed, the code flow seems to work correctly and it reacts much faster.  However, the log looks goofy with all the error messages about HW breakpoints not being hit.  That doesn't make any sense.  This can be seen in the log files with "MaybeWorking" in the name (these are made with -d3 removed from the command line).
BTW, the problem with the breakpoints not getting hit was because the image on the chip did not match the image in CLion.  This was my mistake earlier and is because I'm used to IAR which checks to ensure the image on the chip matches what is in the debugger.  Is there a way for the debugger to ensure they match?

So, I guess it's working.  I will say the debug log (without the -d option) should probably show when SW and HW breakpoints are being added and when they are being hit to help us understand what's happening.  Maybe this is a lower -d level that I don't know about.
[OpenOCDLog-d3_2c-MaybeWorking2.txt](https://github.com/espressif/openocd-esp32/files/9934552/OpenOCDLog-d3_2c-MaybeWorking2.txt)
[OpenOCDLog-d3_2d-Success.txt](https://github.com/espressif/openocd-esp32/files/9934553/OpenOCDLog-d3_2d-Success.txt)
[OpenOCDLog-d3_2a-Success.txt](https://github.com/espressif/openocd-esp32/files/9934554/OpenOCDLog-d3_2a-Success.txt)
[OpenOCDLog-d3_2b-MaybeWorking.txt](https://github.com/espressif/openocd-esp32/files/9934555/OpenOCDLog-d3_2b-MaybeWorking.txt)

closed 2023-10-09 by gerekon