espressif/openocd-esp32 #257: ESP32-S3 openocd debugging error on builtin JTAG - Error: libusb_bulk_read error: LIBUSB_ERROR_PIPE

yellobyte, 2022-11-12:
### Development Kit

esp32-s3-devkitc-1

### Module or chip used

ESP32-S3-WROOM-1

### Debug Adapter

ESP32-S3 builtin JTAG per USB-C

### OpenOCD version

v0.11.0-esp32-20220706 (2022-07-06-15:48)

### Operating System

Windows 10 22H2

### Using an IDE ?

VSCode/PlatformIO - framework = arduino

### OpenOCD command line

openocd.exe -f board\esp32s3-builtin.cfg

### JTAG Clock Speed

default (40000)

### ESP-IDF version

PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)

### Problem Description

Target is a dev board YD-ESP32-S3(Manufacturer: VCC-GND.COM). Schematic & picture of board is attached below.
OpenOCD detects the ESP32-S3 chip:
...
  Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
  Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
  Info : [esp32s3.cpu0] Debug controller was reset.
  Info : [esp32s3.cpu0] Core was reset.
  Info : [esp32s3.cpu1] Debug controller was reset.
  Info : [esp32s3.cpu1] Core was reset. 
and then always runs into the same errors:
  Error: libusb_bulk_read error: LIBUSB_ERROR_PIPE 
  Error: libusb_bulk_read error: LIBUSB_ERROR_IO
  ....etc.
- I tried different device drivers for ESP32-S3's builtin JTAG device: WinUSB (6.1.7600.16385) installed with Zadig 2.7 as well as Espressifs newer version 6.1.7600.16386 from 14.07.2021. Same errors with both drivers.
- File platformio.ini looks like:
    platform_packages = platformio/tool-openocd
    platform = espressif32
    board = esp32-s3-devkitc-1
    framework = arduino
    monitor_speed = 115200
    monitor_port = COM7
    monitor_filters = esp32_exception_decoder, time, colorize
    upload_port = COM7
    upload_speed = 921600
    debug_tool = custom
    debug_server = 
    debug_port = localhost:3333
    debug_init_break = break loop
    build_flags = 
    -DCORE_DEBUG_LEVEL=5
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue  

Have tried all sorts of things for 2 days but can't get a working solution. Do I have the config wrong or is that an openOCD problem ? Any help appreciated.
Thx Thomas

### Debug Logs

```shell
Processing debug (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (custom) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - tool-openocd @ 2.1100.211028 (11.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 61 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
C:\Users\tj\.platformio\packages\tool-openocd-esp32\bin\openocd.exe -f board\esp32s3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:F2:FD:28)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Error: libusb_bulk_read error: LIBUSB_ERROR_PIPE
Error: libusb_bulk_read error: LIBUSB_ERROR_IO
Error: libusb_bulk_read error: LIBUSB_ERROR_IO
Error: libusb_bulk_read error: LIBUSB_ERROR_IO
Error: libusb_bulk_read error: LIBUSB_ERROR_IO
Error: libusb_bulk_read error: LIBUSB_ERROR_IO
Error: libusb_bulk_read error: LIBUSB_ERROR_IO
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: missing data from bitq interface
Error: Failed to exec JTAG queue!
Error: Failed to read debug stubs address location (-104)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: missing data from bitq interface
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: missing data from bitq interface
Error: Failed to exec JTAG queue!
Error: Failed to read debug stubs address location (-104)!
Polling target esp32s3.cpu0 failed, trying to reexamine
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: esp_usb_jtag: device not found!
....
```

### Expected behavior

openOCD should not produce such errors

### Screenshots

[image]
[image]
[YD-ESP32-S3_Schematic.pdf](https://github.com/espressif/openocd-esp32/files/9995482/YD-ESP32-S3_Schematic.pdf)

yellobyte, 2022-11-12:
Oh yes, having "build_type = debug" in platformio.ini & therefore building in **debug** mode produces similar errors:
...
**Building in debug mode**
C:\Users\tj\.platformio\packages\tool-openocd-esp32\bin\openocd.exe -f board\esp32s3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001        
Info : esp_usb_jtag: capabilities descriptor set to 0x2000      
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:F2:FD:28)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Error: Unexpected OCD_ID = 00000000
Error: Unexpected OCD_ID = 00000000
Error: Unexpected OCD_ID = 00000000
Error: Unexpected OCD_ID = 00000000
Warn : target esp32s3.cpu1 examination failed
Error: Unexpected OCD_ID = 00000000
Error: Unexpected OCD_ID = 00000000
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Error: libusb_bulk_read error: LIBUSB_ERROR_PIPE
Error: libusb_bulk_read error: LIBUSB_ERROR_IO
....

erhankur, 2023-03-02:
@yellobyte can you please try the latest OpenOCD release? We have updated the libusb version and this update fixed similar issues.  

Also some users had a problem when they left strapping pins floated. You can read from here. It is C3 topic but can be an issue for S3, too. 
https://github.com/espressif/openocd-esp32/issues/241

erhankur, 2023-06-28:
Closed due to inactivity.

closed 2023-06-28 by erhankur