espressif/openocd-esp32 #258: Vendor IDs and *.cfg scripts ?

franz-ms-muc, 2022-11-22:
### Development Kit

ESP32-S3-DEVKITC

### Module or chip used

ESP32-S3 (several)

### Debug Adapter

none. Want to use USB-JTAG

### OpenOCD version

not matter

### Operating System

Windows

### Using an IDE ?

Espressif-IDE

### OpenOCD command line

not matter

### JTAG Clock Speed

not matter

### ESP-IDF version

esp-idf-v4.4.3

### Problem Description

there seems to be 2 different Vendor-IDs out for the USB-JTAG Adapter. 

0x1001 and 0x1002. 

while 0x1002 reffers to esp32s3-bridge.cfg
while 0x1001 reffers to esp32s3-builtin.cfg

can you Explain or show a source of Documentation ? 

### Expected behavior

have a good documentation.

franz-ms-muc, 2022-11-22:
ah, i see: 

esp32s3-bridge.cfg reffers to --> esp\_usb\_bridge.cfg  
esp32s3-builtin.cfg reffers to --> esp\_usb\_jtag.cfg

and 

in the diffs it is clearly visible, but still unclear when to use what: 

[image]

[image]

igrr, 2022-11-22:
esp32s3-builtin.cfg is for the built-in USB_SERIAL_JTAG peripheral of the ESP32-S3 chip.

"ESP USB Bridge" is an external JTAG adapter implemented in https://github.com/espressif/esp-usb-bridge.

The official list of config files is available here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/tips-and-quirks.html#configuration-of-openocd-for-specific-target. We haven't added esp-usb-bridge to that list since we don't yet have an official development board which implements that protocol.

franz-ms-muc, 2022-11-22:
@igrr Thanks for this Response. 

i did read about ESP USB Bridge but did no more have it in mind. 

BUT then you have a Bug in Espressif-IDE: 

there are only 2 Configurations available: 

first: 

[image]

second:

[image]

should i raise a new Issue in Espressif-IDE ?

franz-ms-muc, 2022-11-22:
is this the Place the Espressif-IDE Developers work on ? 

[https://github.com/espressif/idf-eclipse-plugin](https://github.com/espressif/idf-eclipse-plugin) ?

igrr, 2022-11-22:
Eclipse IDE gets the list from esp-config.json file from OpenOCD distribution. The missing config was added in https://github.com/espressif/openocd-esp32/commit/610883f6520edf1d67478342f29d75531d93372c. When IDF updates the to latest release of OpenOCD (https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20221026), you will see the esp32s3-builtin.cfg in the list.

closed 2023-01-17 by erhankur