espressif/openocd-esp32 #259: JTAG Programming Flash Speed Slow for N32R8V

JeremiahGillis, 2022-11-23:
### Development Kit

Custom

### Module or chip used

ESP32-S3-WROOM-2-N32R8V

### Debug Adapter

Built in JTAG over USB

### OpenOCD version

v0.11.0-esp32-20220706 

### Operating System

Windows 10 x64

### Using an IDE ?

VS Code v1.73.1

### OpenOCD command line

ESP-IDF Flash device button via Espressif IDF Extension

### JTAG Clock Speed

Default, 40000

### ESP-IDF version

v4.4.3

### Problem Description

Programming using the S3 built in USB JTAG via `ESP-IDF Flash device` is very slow. It takes about 40 seconds to program a 3MB binary. This is unusually slow for JTAG programming using Octal memory.

Create a large application, at least 3MB. Build and flash the application in Visual Studio Code. 

### Debug Logs

```shell
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:E4:4F:00)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
Flashing c:/Users/jerem/Source/Repos//touch/build/bootloader/bootloader.bin at 0x0
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Target halted, PC=0x42056FCE, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x42056FCE, debug_reason=00000000
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 2575 KB
Info : Flash mapping 1: 0x2a0020 -> 0x42000020, 352 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 32768 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 32768 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 2575 KB
Info : Flash mapping 1: 0x2a0020 -> 0x42000020, 352 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu1.flash' size 32768 KB
Info : Using flash bank 'esp32s3.cpu1.flash' size 32768 KB
** Programming Started **
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : PROF: Erased 28672 bytes in 327.009 ms
Info : PROF: Data transferred in 235.022 ms @ 119.138 KB/s
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : PROF: Wrote 28672 bytes in 584.063 ms (data transfer time included)
** Programming Finished in 2019 ms **
** Verify Started **
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : PROF: Flash verified in 169.945 ms 
** Verify OK **
** Flashing done for bootloader/bootloader.bin in 2207 ms **
Flashing c:/Users/jerem/Source/Repos/touch/build/Touch.bin at 0x10000
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : PROF: Erased 3072000 bytes in 11453 ms
Info : PROF: Data transferred in 24270 ms @ 123.609 KB/s
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : PROF: Wrote 3072000 bytes in 24620 ms (data transfer time included)
** Programming Finished in 36512 ms **
** Verify Started **
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : PROF: Flash verified in 1832.97 ms 
** Verify OK **
** Flashing done for Touch.bin in 38377 ms **
Flashing c:/Users/jerem/Source/Repos/touch/build/partition_table/partition-table.bin at 0x8000
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : PROF: Erased 4096 bytes in 180.975 ms
Info : PROF: Data transferred in 48.135 ms @ 83.0996 KB/s
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : PROF: Wrote 4096 bytes in 377.033 ms (data transfer time included)
** Programming Finished in 964 ms **
** Verify Started **
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : PROF: Flash verified in 160.004 ms 
** Verify OK **
** Flashing done for partition_table/partition-table.bin in 1140 ms **
** Total programming time 41736 ms **
** Resetting Target **
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
0
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
```
```

### Expected behavior

I would like for faster flashing. I would have to wait over 5 minutes if I used the entire 32 MB. 

### Screenshots

SDK Configuration Editor:
![image](https://user-images.githubusercontent.com/13179985/203459049-8d7a5f21-fb16-46dd-9bc9-80f9c2c912d8.png)

JeremiahGillis, 2022-11-23:
Here is a detailed log when manually programming using `openocd -l openocd_log.txt -d3 -f board/esp32s3-builtin.cfg -c "program_esp build/Touch.bin 0x10000 verify reset exit"`

[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/10072666/openocd_log.txt)

erhankur, 2022-11-23:
@JeremiahGillis Thanks for reporting this. I'll check again. In the meantime can you try to load with the compress option?
```
-c "program_esp build/Touch.bin 0x10000 compress verify reset exit"
```

JeremiahGillis, 2022-11-23:
@erhankur Programming is faster. It saves about 14 seconds for the entire process. [openocd_log_compress.txt](https://github.com/espressif/openocd-esp32/files/10076921/openocd_log_compress.txt)

Note that both of the `openocd` terminal calls had `adapter speed` set to 26000. I was experimenting with various speeds. I didn't see any changes until I significantly lowered the speed to create a bottleneck. My original log using the flash button in VS Code was 40000.

As an FYI, I did try `idf.py flash` first at various baud rates including 10 Mbaud. This was even slower than openocd. I noticed in the logs `--flash_mode dout` which isn't correct and does not match menuconfig.

HSPalm, 2023-02-09:
I am experiencing the same, but with ESP32S2. I reach speeds at between 150 to 300 KB/s. Using the compress option works to compress the file, but for some reason the speed is lowered to about 90 KB/s so the total time is even a bit more.
@JeremiahGillis Did you ever reach higher speeds?
@erhankur Did you find out something?

Where might the limitation rely? It is frustrating. We pay our EMS per second during production testing, and erase/flash sequences that lasts minutes is very costly.

erhankur, 2023-02-09:
@HSPalm yes, compression is not the best booster for every file. It can reduce the binary load time, but decompression overhead might be higher.  What is your jtag clock frequency?  You can play with the `adapter speed <khz>` command. 
Or maybe you can remove `verify` option for the tests and skip reading flash.

In the stub loader, we are boosting the CPU frequency to the max level. The bottleneck is in the flash erase/write functions. I need to check if I can increase w/r speed of flash by changing the init settings. 

To be honest, I couldn't find a slot to look at this. Give me some more days. Sorry for the delay.

JeremiahGillis, 2023-02-09:
> Did you ever reach higher speeds?

No, nothing has changed for me. I spent some time on this before posting here. I also updated my project to ESP-IDF v5.0. This updated my OpenOCD version to `v0.11.0-esp32-20221026`. I still have the exact same behavior.

For now I am just dealing with it. Hopefully Espressif improves this since my device specifically has a lot of build in flash memory for large applications. Programming becomes a very long process when you include secure boot and flash encryption with keys generated on device.

> stub loader

Yes, please check. Are you using OPI mode for the `N32R8V`?

There may be issues in the bootloader (default one) or the stub program. I know for sure that ESP needs to work on the bootloader. Espressif is using parameters even in ESP-IDF v5.0 which shouldn't be used for mass production. I know this due to the log output at boot. https://github.com/espressif/esp-idf/issues/9586

HSPalm, 2023-02-10:
Thank you both for replying.

@erhankur 
> What is your jtag clock frequency? 
adapter_khz 20000
> To be honest, I couldn't find a slot to look at this. Give me some more days. Sorry for the delay.
Don't worry. I look forward to any findings.

@JeremiahGillis I don't know what a stub loader/program means. I will report back here if we find something.

JeremiahGillis, 2023-02-10:
@HSPalm Since you are on the S2, see this https://docs.espressif.com/projects/esptool/en/latest/esp32s2/esptool/flasher-stub.html

You can try with `no stub` to see if the original ROM is any different in speed. At any case, you should post your detailed log here in case it is needed. You can look at my comment above with the posted log for the command.

HSPalm, 2023-02-13:
> @HSPalm Since you are on the S2, see this https://docs.espressif.com/projects/esptool/en/latest/esp32s2/esptool/flasher-stub.html
>
> You can try with `no stub` to see if the original ROM is any [...]

Thanks for the tip. It says `erase_flash` is not available for the original bootloader. But are there any alternatives? Flashing an "empty" bin? We do at least one full flash erase during production because we have a separate HW-test firmware that jiggles all the IO for the inline test machine. Then we want a full erase after this to be on the safe side. An alternative would be to flash directly to RAM but I have not achieved this yet.

HSPalm, 2023-03-03:
@erhankur 
>The bottleneck is in the flash erase/write functions.

Hi. Were you able to look into this? From your previous message it looks like it is not OpenOCD related. It makes me wonder, is everyone using ESP32S2 experiencing these slow programming speeds? Or is there a workaroud using either JTAG or serial bootloader? The programming speed is a true bottleneck during production, and it's expensive. So I am very interested in any solution.

JeremiahGillis, 2023-11-22:
> The bottleneck is in the flash erase/write functions.

Just bumping this thread since it has been one year since I opened the issue. We have used over 2000 modules in the last year and this has been a big bottleneck during manufacturing. We have to limit the number of test fixtures we buy due to both cost and manufacturing floor space.

Can anything be done to improve these erase/write functions?

HSPalm, 2023-11-24:
Just to quantify it. Our European EMS charges 100 EUR per hour for a fully automatic inline ICT tester, and we spend two minutes per board doing 2x full erase/write cycles. That's 3.33 EUR per board just flashing. We've produced tens of thousands of products using ESP32S2 (chip, no module).

erhankur, 2023-11-24:
I understand. Unfortunately, I have no free time slot to look at this for now. But I will take it for the next quarter as a "must to do" issue. 

Octal Flash might be working in SPI mode so that you don't see the differences.  This is what I remember but I need to confirm.  
If my assumption is correct, considering only the programming use case, I can prepare a special test version for you. 

Sorry for the trouble.

gerekon, 2024-02-20:
Some update regarding flashing speed for ESP32-S3. From my experiments with the latest OpenOCD version. 

When flashing using builtin USB_JTAG interface (`-f board/esp32s3-builtin.cfg`) it takes 47 sec for flashing 3 MB binary.
```
PROF: Erased 3072000 bytes in 11156.7 ms
PROF: Data transferred in 34394.4 ms @ 87.2235 KB/s
[esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
PROF: Wrote 3072000 bytes in 34648.1 ms (data transfer time included)
** Programming Finished in 46922 ms **
```

When flashing using builtin USB_JTAG interface (`-f board/esp32s3-ftdi.cfg` which uses `adapter speed 20000`) it takes 20 sec for flashing 3 MB binary.
```
PROF: Erased 3072000 bytes in 11091.9 ms
PROF: Data transferred in 8731.55 ms @ 343.582 KB/s
[esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
PROF: Wrote 3072000 bytes in 8910.29 ms (data transfer time included)
** Programming Finished in 20337 ms **
```

Erase times are the same.

gerekon, 2024-02-20:
This is from the spec 
[image]

gerekon, 2024-02-20:
> Your test shows a 4x increase in data transfer. This definitely points to an area of improvement. Can someone look into the internal USB and JTAG block of the S3 for closer evaluation of the bottleneck?

The only difference in my tests is using JTAG pins connected to the host via FTDI in one case and using builtin USB_JTAG in 
 another case. Yes I agree that ESP USB_JTAG protocol could be less efficient than FTDI, but it is not so complicated to be much more heavy. In any case it would be good to have a look at it to compare its overhead with FTDI.
Since USB_JTAG on target is implemented in HW I do not think it is possible to improve something here for ESP32-S3, because as far as I now new versions of ESP32-S3 chips are not planned in Espressif.

> I've seen people flash these faster over UART at ~ 500 kbit/s.

Is this for direct HW UART pins connection, USB-UART bridge on evaluation board (which can actually be HS device) or builtin USB_SERIAL (which is FS device like JTAG)? What baudrates did they use? Also note, that flashing over JTAG involves a lot of transactions. There are many layers in this kind of communication as compared to what `esptool.py` does. So direct comparison is not completely correct IMHO because flash over JTAG has much more overhead (and this can be a showstopper in case of FS device). To get some brief understanding of possible overhead you can have a look at https://github.com/espressif/esp-idf/blob/master/components/app_trace/port/xtensa/port.c#L27. It is register-based protocol which involves many registers reads/writes which are done by means of JTAG transactions over USB.
From my experience, during testing of `apptrace` protocol, which ESP flash driver uses on top of JTAG, I saw significant data transfer speed degradation between chip and host when used FS USB hubs in between.

> @erhankur mentioned that the octal flash chips like mine may not be running in octal mode when flashing.

I will look at OPI mode support in flasher, but what I see from the stub's log is that in case of USB_JTAG data transfer tooks 70% of the time. See below 134444 us and 184427 us which were spent for transferring two 16KB blocks over JTAG and 50943 us spent for writing them into flash as one 32 KB block. So even if we will manage to improve flash access speed on target the overall result will not change significantly.
```
Got trace down buf 368 bytes @ 0x3fcba4d0 in 180389 us
STUB_D: Write flash @ 0x2e8000 sz 32768 in 50835 us
STUB_V: Req trace down buf 57344 bytes 3072000-3014656
STUB_V: Got trace down buf 16014 bytes @ 0x3fcaa640 in 1 us
STUB_V: Req trace down buf 41330 bytes 3072000-3030670
STUB_V: Got trace down buf 16382 bytes @ 0x3fcae4ce in 134444 us
STUB_V: Req trace down buf 24948 bytes 3072000-3047052
STUB_V: Got trace down buf 16382 bytes @ 0x3fcb24cc in 184427 us
STUB_D: Write flash @ 0x2f0000 sz 32768 in 50943 us
STUB_V: Req trace down buf 8566 bytes 3072000-3063434
STUB_V: Got trace down buf 8566 bytes @ 0x3fcb64ca in 47002 us
STUB_D: Write flash @ 0x2f8000 sz 24576 in 38178 us
STUB_D: Wrote 3072000 bytes @ 0x10000
STUB_D: exit 0
```

JeremiahGillis, 2024-02-21:
> Since USB_JTAG on target is implemented in HW

I figured this might be the case, but I wanted to ask in case there was some firmware involved. I understand why there may be no revisions correcting this. I just hope that Espressif considers improvements here for future SoCs. Programming speeds are important during product manufacturing.

> Is this for direct HW UART pins connection, USB-UART bridge on evaluation board (which can actually be HS device) or builtin USB_SERIAL (which is FS device like JTAG)? What baudrates did they use?

I was mistaken here when I mentioned 4x faster. I was thinking KB vs Kbits. Your built-in interface was 87.2235 KB/s which is faster than 500 kbit/s UART. Thanks for the link, it does help me understand what is going on.

> So even if we will manage to improve flash access speed on target the overall result will not change significantly.

Understood. There may be a few reasons to look at the stub program. I will add another comment here shortly with some flash programming details and results.

JeremiahGillis, 2024-02-21:
I spent this evening investigating this issue further and learned a few interesting things. I performed all development on Windows 11 in VS Code and flashed with `OpenOCD`. The test fixtures I designed and built used RPi CM4 running Raspberry Pi OS. The ESP32-S3 modules in the fixtures get flashed using `Esptool` over the native USB port.

I decided to compare flashing speeds between OpenOCD and Esptool in a ESP-IDF terminal in VS Code on my Windows computer. I used an ESP32-S3-DevKitC-1 v1.6 which as an ESP32-S3-WROOM-2 module for all testing. This has the native USB port along with a Silicon Labs CP210x USB to UART Bridge running at full-speed. My OpenOCD version for testing was `v0.12.0-esp32-20230921` and Esptool version was `4.7.0`. My bin size is different from my original post as I did not save the original. No USB hubs were used.

OpenOCD:
```
openocd -f board/esp32s3-builtin.cfg -c "program_esp build/Touch.bin 0x10000 exit"
Info : PROF: Erased 3690496 bytes in 12493.9 ms
Info : PROF: Data transferred in 30118.9 ms @ 119.659 KB/s
** Programming Finished in 45052 ms **

openocd -f board/esp32s3-builtin.cfg -c "program_esp build/Touch.bin 0x10000 compress exit"
Info : PROF: Erased 3690496 bytes in 12581.9 ms
Info : PROF: Compressed 3690496 bytes to 1276437 bytes in 125.121996ms
Info : PROF: Data transferred in 12228.1 ms @ 101.939 KB/s
Info : PROF: Wrote 3690496 bytes in 12582.1 ms (data transfer time included)
** Programming Finished in 27431 ms **
```

Esptool, USB CDC:
```
python -m esptool --no-stub write_flash --flash_freq 80m --flash_mode qio --flash_size 32MB 0x10000 build/Touch.bin
Erasing flash...
Took 12.40s to erase flash block
Wrote 3687424 bytes at 0x00010000 in 172.4 seconds (171.1 kbit/s)...

python -m esptool write_flash --flash_freq 80m --flash_mode qio --flash_size 32MB 0x10000 build/Touch.bin
Flash will be erased from 0x00010000 to 0x00394fff...
Compressed 3686960 bytes to 1265835...
Wrote 3686960 bytes (1265835 compressed) at 0x00010000 in 56.8 seconds (effective 518.8 kbit/s)...
```

Esptool, USB to UART:
```
python -m esptool -b 3000000 --no-stub write_flash --flash_freq 80m --flash_mode qio --flash_size 32MB 0x10000 build/Touch.bin
Erasing flash...
Took 12.46s to erase flash block
Wrote 3687424 bytes at 0x00010000 in 171.9 seconds (171.7 kbit/s)...

python -m esptool -b 2000000 write_flash --flash_freq 80m --flash_mode qio --flash_size 32MB 0x10000 build/Touch.bin
Flash will be erased from 0x00010000 to 0x00394fff...
Compressed 3686960 bytes to 1265835...
Wrote 3686960 bytes (1265835 compressed) at 0x00010000 in 56.4 seconds (effective 522.7 kbit/s)...
```

Here is the same testing on my RPi CM4. I compiled openocd from this repository match the latest release `v0.12.0-esp32-20230921`. My Esptool version was `4.7.0`.

OpenOCD:
```
openocd -f board/esp32s3-builtin.cfg -c "program_esp build/Touch.bin 0x10000 exit"
Info : PROF: Erased 3690496 bytes in 12562.7 ms
Info : PROF: Data transferred in 31982.5 ms @ 112.687 KB/s
Info : PROF: Wrote 3690496 bytes in 32238.7 ms (data transfer time included)

openocd -f board/esp32s3-builtin.cfg -c "program_esp build/Touch.bin 0x10000 compress exit"
Info : PROF: Erased 3690496 bytes in 12625.6 ms
Info : PROF: Compressed 3690496 bytes to 1276437 bytes in 433.778992ms
Info : PROF: Data transferred in 12305.4 ms @ 101.298 KB/s
Info : PROF: Wrote 3690496 bytes in 12611 ms (data transfer time included)
** Programming Finished in 26464 ms **

```

Esptool, USB CDC:
```
python -m esptool --no-stub write_flash --flash_freq 80m --flash_mode qio --flash_size 32MB 0x10000 build/Touch.bin
Erasing flash...
Took 12.42s to erase flash block
Wrote 3687424 bytes at 0x00010000 in 64.4 seconds (458.3 kbit/s)...

python -m esptool write_flash --flash_freq 80m --flash_mode qio --flash_size 32MB 0x10000 build/Touch.bin
Flash will be erased from 0x00010000 to 0x00394fff...
Compressed 3686960 bytes to 1265835...
Wrote 3686960 bytes (1265835 compressed) at 0x00010000 in 56.8 seconds (effective 519.7 kbit/s)...
```

Esptool, USB to UART:
```
python -m esptool -b 3000000 --no-stub write_flash --flash_freq 80m --flash_mode qio --flash_size 32MB 0x10000 build/Touch.bin
Erasing flash...
Took 12.54s to erase flash block
Wrote 3687424 bytes at 0x00010000 in 65.1 seconds (453.1 kbit/s)...

python -m esptool -b 3000000 write_flash --flash_freq 80m --flash_mode qio --flash_size 32MB 0x10000 build/Touch.bin
Flash will be erased from 0x00010000 to 0x00394fff...
Compressed 3686960 bytes to 1265835...
Wrote 3686960 bytes (1265835 compressed) at 0x00010000 in 56.4 seconds (effective 523.1 kbit/s)...
```

A few notes for the testing above.
* Esptool stub does not like the UART speed at 3 Mbaud. It would loose communicates at the beginning. I had to lower to 2 MBaud.
* Esptool stub automatically compresses the binary. I added the compress option in OpenOCD for comparison.

Summary
1. OpenOCD appears to be the fastest method for flashing.
2. There is a bottleneck in Windows vs Linux (RPi OS) when using the Esptool for uncompressed flashing.
3. I will be able to switch to OpenOCD from Esptool in my test fixtures. This will reduce my programming time by almost 50%.

gerekon, 2024-02-21:
Thanks for interesting results.
I tested flashing speed on Ubuntu and have different results for esptool
```
> esptool.py -b 3000000 write_flash --flash_freq 80m --flash_mode qio --flash_size 32MB 0x10000 /tmp/bin.dat
esptool.py vv4.8.dev1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_1v8)
Crystal is 40MHz
MAC: 68:b6:b3:3d:14:08
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 3000000
Changed.
Configuring flash size...
Flash will be erased from 0x00010000 to 0x002fdfff...
Compressed 3072000 bytes to 3072946...
Wrote 3072000 bytes (3072946 compressed) at 0x00010000 in 37.0 seconds (effective 664.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

gerekon, 2024-02-21:
and for OpenOCD `v0.12.0-esp32-20230921`
```
** Programming Started **
[esp32s3.cpu0] Target halted, PC=0x4038E3B6, debug_reason=00000001
[esp32s3.cpu0] Target halted, PC=0x4038E3B6, debug_reason=00000001
PROF: Erased 3072000 bytes in 11216.2 ms
PROF: Data transferred in 34833 ms @ 86.1251 KB/s
[esp32s3.cpu0] Target halted, PC=0x4038E3B6, debug_reason=00000001
PROF: Wrote 3072000 bytes in 35091.5 ms (data transfer time included)
** Programming Finished in 47397 ms **
```

gerekon, 2024-02-21:
For OpenOCD
> Info : PROF: Compressed 3690496 bytes to 1276437 bytes in 433.778992ms

Looks like compression in your case helps. <y experiment is done with randomly generated binary which is not compressed well.
```
PROF: Erased 3072000 bytes in 11232.1 ms
PROF: Compressed 3072000 bytes to 3072470 bytes in 70.955002ms
PROF: Data transferred in 34339.4 ms @ 87.3767 KB/s
[esp32s3.cpu0] Target halted, PC=0x4038E3B6, debug_reason=00000001
PROF: Wrote 3072000 bytes in 34596.9 ms (data transfer time included)
** Programming Finished in 46202 ms **
```

JeremiahGillis, 2024-02-21:
I did pull the pull the master branch of OpenOCD on my RPi CM4. I am seeing the same results as before, transferring at 100 KB/s.

> Looks like compression in your case helps.

I am definitely happy about this. It makes sense because I have a GUI display with a lot of screen assets. Compression also overcomes my Windows Esptool bottleneck. Too bad there is a decrease in actual throughput, likely due to the decompression algorithm.

JeremiahGillis, 2024-02-22:
> When flashing using JTAG pins

I just wanted to confirm that I also saw faster speeds on my DevKit when using the external JTAG pins. I was using a J-Link device which was limited to `adapter speed: 10000 kHz`.

```
openocd -f interface/jlink.cfg -c "adapter speed 10000" -f target/esp32s3.cfg -c "program_esp build/Touch.bin exit"
Info : PROF: Erased 3690496 bytes in 12450.3 ms
Info : PROF: Data transferred in 15404.7 ms @ 233.954 KB/s
Info : PROF: Wrote 3690496 bytes in 15644.6 ms (data transfer time included)
** Programming Finished in 30206 ms **
```

The transfer speeds actually slowed down with compression:
```
openocd -f interface/jlink.cfg -c "adapter speed 10000" -f target/esp32s3.cfg -c "program_esp build/Touch.bin compress exit"
Info : PROF: Erased 3690496 bytes in 12512.6 ms
Info : PROF: Compressed 3690496 bytes to 1276437 bytes in 125.320002ms
Info : PROF: Data transferred in 6641.53 ms @ 187.686 KB/s
Info : PROF: Wrote 3690496 bytes in 6950.17 ms (data transfer time included)
** Programming Finished in 21674 ms **
```

erhankur, 2024-03-05:
@JeremiahGillis now we all concluded that improving flash speed (if possible) will not change the overall result enough. 
So we can close this issue if you don't have any other questions.

HSPalm, 2024-03-05:
@erhankur The problem still persists and is a bug in the read/write speed of flash, or I misunderstand? IMO this issue is should be alive until we know if flash R/W speeds can be increased or not. I captured the wait times on a scope, blue trace is a data block from computer, and the idle time in between is where ESP writes block to flash, then ACKs (yellow trace) until new block can be transferred. My experience shows that the write time for ESP to flash is the same, so  increasing UART write speed does not impact ESP write speed much. I found the max limit on my system to be UART write speed of 2M, resulting in a real speed of about 800K which corresponds to the traces. I believe this is what you described earlier about the bottleneck.
[image]
[image]
sorry about the text image

gerekon, 2024-03-05:
@HSPalm Did you measure that wth OpenOCD? What is size of the written data block? Looks like it takes about 200 ms to write it to flash.
OpenOCD 32KB block transfers. And for ESP32-S3 N32R8V (used by the topic starter) it takes about 50-60 us to write it to flash (not accounting the transfer time). As far as I see you use ESP32-S2 with another flash chip.

JeremiahGillis, 2024-03-05:
I do not have any other questions about the S3. I only ask that feedback from this thread be shared with the hardware and firmware engineers at Espressif. Flash programming speed is important for mass production. Currently it would take almost 5 minutes to program the entire 32MB of the N32R8V.

You can see that data from @gerekon which shows over a 3x increase in throughput using an external JTAG device. The built in JTAG should be able to match or exceed that since USB full speed can reach 1 MB/s, not to mentions devices with octal flash memory. Hopefully the USB Serial speed could be improved at the same time for those who use Esptool.

I don't expect the S3 die to change, but it would be nice to see improvements in this area for future ESP devices.

gerekon, 2024-03-05:
@HSPalm Regarding your case...
if my assumption above (about slow flash chip on ESP32-S2) is correct the only thing which we can try is to increase flash frequency.

BTW. What debug adapter do you use? What JTAG freq?

HSPalm, 2024-03-05:
> @HSPalm Did you measure that wth OpenOCD? What is size of the written data block? Looks like it takes about 200 ms to write it to flash. OpenOCD uses 32KB block transfers. And for ESP32-S3 N32R8V (used by the topic starter) it takes about 50-60 us to write it to flash (not accounting the transfer time). As far as I see you use ESP32-S2 with another flash chip.

I'm sorry to confuse you. I'm flashing through UART but considered the experiment valid because it seems flash write speed is the limiting factor here as well. File size is 2730KB I believe.