espressif/openocd-esp32 #261: Could not run/continue: Don't know how to run. Try "help target". (from exec-run)

skanky-dev, 2022-11-30:
### Development Kit

Custom board

### Module or chip used

ESP32-WROOM-32E

### Debug Adapter

ESP_PROG

### OpenOCD version

v0.11.0-esp32-20220706

### Operating System

Win10 Pro

### Using an IDE ?

VS Code

### OpenOCD command line

F5

### JTAG Clock Speed

20000 kHz

### ESP-IDF version

esp-idf-v4.4.3

### Problem Description

Press F5 to debug project
Programs as expected, sets up to start debugging as expected and then VS Code shows pop up message "Could not run/continue: Don't know how to run. Try "help target". (from exec-run)"

This is an existing project I've taken over. After extensive searching I can't find any results for this pop up error message and after working through everything I can think of on the project config I can't find what the problem might be!

### Debug Logs

```shell
---------------------------
 ---------------------------
 ----- TERMINAL OUTPUT -----
 ---------------------------
 ---------------------------
 *  Executing task: openocd -s C:\Users\Dev\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\share\openocd\scripts -c 'adapter_khz 12000' -f interface\ftdi\esp32_devkitj_v1.cfg -f C:\Users\Dev\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\share\openocd\scripts\board\esp-wroom-32.cfg  

Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
adapter speed: 12000 kHz

adapter speed: 20000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[... 137 lines cut ...]
[New Thread 1073469596]
[New Thread 1073469248]
[New Thread 1073446224]
[New Thread 1073461380]
[New Thread 1073445612]
[Switching to Thread 1073468900]

Thread 1 "main" hit Temporary breakpoint 1, app_main () at C:/_PROJECTS/project1/firm/rpdintl-100-216_project1/main/rpd_main.c:654
warning: Source file is more recent than executable.
654	{
14-thread-info
Running executable
15-exec-run
16-thread-info
17-stack-info-depth --thread 1
18-stack-info-depth --thread 1
-gdb-exit
19-stack-list-frames --thread 1 0 2
20-stack-list-frames --thread 1 0 2
[Inferior 1 (Remote target) detached]
```

### Expected behavior

Debugging to start as normal

gerekon, 2022-12-01:
Hi @skanky-dev 

Actually the problem is not related to OpenOCD. Looks like VsCode debug session setup problem.
Do you use Espressif VsCode extention for debugging?  Looks like not and you use VsCode builtin debugger instead. [Here the description](https://code.visualstudio.com/docs/cpp/launch-json-reference) how to setup it. From my side I can tell that `-exec-run` is wrong command when you already run the program and stopped at `app_main`. `-exec-continue` or its equivalent `-interpreter-exec console "c"` should be used instead. But sorry I can not tell you why  VsCode builtin debugger use `-exec-run` in your case. I know that many people use it successfully for Espressif chips, so looks like configuration problem.

If you want to use Espressif VsCode extention for debugging, please have a look at [how to setup it](https://github.com/espressif/vscode-esp-idf-extension) and [how to debug with it](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/debugging.md) and [more details on debug configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md).

skanky-dev, 2022-12-01:
Hi, as far as I'm aware I'm already using the Espressif VsCode extension. This is my launch.json:

{
    "version": "0.2.0",
    "configurations": [        
        {
            "name": "OpenOCD Debugging",
            "type": "gdb",
            "request": "launch",
            "printCalls": true,
            "target": "./build/100_216_project1.elf",
            "cwd": "${workspaceFolder}",
            "autorun": [
                "target remote :3333",
                "mon reset halt",
                "mon program_esp ./build/100_216_project1.bin 0x10000",
                "flushregs",
                "thb app_main",
                "c"
            ],
            "preLaunchTask": "openocd_debug",
            "gdbpath": "C:\\Users\\Dev\\.espressif\\tools\\xtensa-esp32-elf\\esp-2021r2-patch5-8.4.0\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-gdb"
        }
    ]
}

gerekon, 2022-12-01:
No. It should have `"type": "espidf"` like below. See [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md)
```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "debugPort": 9998,
      "logLevel": 2,
      "mode": "manual",
      "verifyAppBinBeforeDebug": false,
      "tmoScaleFactor": 1,
      "initGdbCommands": [
        "target remote :3333",
        "symbol-file /path/to/program.elf",
        "mon reset halt",
        "flushregs",
        "thb app_main"
      ],
      "env": {
        "CUSTOM_ENV_VAR": "SOME_VALUE"
      }
    }
  ]
}
```

skanky-dev, 2022-12-01:
Thank you!!! I changed that and debugging is now working :-)

closed 2022-12-01 by gerekon