espressif/openocd-esp32 #263: JTAG Error: libusb_get_string_descriptor_ascii() failed with -1

michaeldakwar, 2022-12-10:
### Development Kit
None

### Module or Chip Used
ESP32-C3FN4

### Debug Adapter
On-board (GPIO18/19) directly through USB Micro-B

### OpenOCD version
Open On-Chip Debugger  v0.11.0-esp32-20220706

### Operating System
macOS Monterey 12.4

### Using an IDE ?
Visual Studio Code

### OpenOCD command line
Not shown in VSCode

### JTAG Clock Speed
N/A

### ESP-IDF version
5.0-rc3 (commit: 3f2f35bd5e1c3bab12def58de48c83fd8e93e431)

### Problem Description

I am unable to connect to the JTAG interface on pins GPIO18/19 of the chip. However, using Windows, I am able to connect to the JTAG interface. I had to change drivers and reinstall in order for it to work on Windows, so I know that my interface at least works. 
I am also able to connect through a USB/UART bridge on both macOS and Windows.

So no issues with the UART, only the JTAG

The output below is when I press the "ESP_IDF Build, Flash, and Monitor" button in VSCode.
The error shown below is "Error: libusb_get_string_descriptor_ascii() failed with -1", and then openOCD stops.

```
[OpenOCD]
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

[OpenOCD]
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
[OpenOCD]
Info : Listening on port 6666 for tcl connections
[OpenOCD]
Info : Listening on port 4444 for telnet connections
[OpenOCD]
❌ Error: libusb_get_string_descriptor_ascii() failed with -1
[OpenOCD]
❌ 
Error: Unsupported xlen: -1
Error: Unknown target arch!
[OpenOCD]
[Stopped] : OpenOCD Server
```
[image]

erhankur, 2023-03-02:
@michaeldakwar do you still have a problem? If yes, can you please try the latest release and share the result?

michaeldakwar, 2023-03-02:
Erhan,I no longer have the issue. The issue was my pins on the ESP32 were not all secured with solder. I fixed the solder and no more issues.  Regards,Michael

erhankur, 2023-03-02:
thanks for the info @michaeldakwar

closed 2023-03-02 by erhankur

AlvaroZimmermann, 2023-11-18:
@michaeldakwar  Hello, do you know which pins you had to solder? I am having the same exactly same problem with the ESP32