espressif/openocd-esp32 #264: setting gdb breakpoints at init causes PSRAM memory overwrite (IDFGH-8913)

zfields, 2022-12-13:


igrr, 2022-12-13:
@zfields Have transferred this issue to OpenOCD project so that it is tracked properly.

Regarding the successful read on 2nd attempt, there has been a similar issue we had fixed recently; however it looks like you are already using the latest release which includes the fix. Could you please confirm the version of OpenOCD you are running?

Regarding PSRAM support for ESP32-S3, it will likely be done in Q1 2023.

zfields, 2022-12-13:
Thanks @igrr!

> Could you please confirm the version of OpenOCD you are running?
I'm using OpenOCD v0.11.0-esp32-20221026.

> Regarding PSRAM support for ESP32-S3, it will likely be done in Q1 2023.
Thanks, that's great news!

> Have transferred this issue to OpenOCD project so that it is tracked properly.
No, I have not. Is there a way to "transfer" it using the GitHub UI, or do you mean reopen it there?

igrr, 2022-12-13:
Sorry for not phrasing this clearly. I meant that I have already transferred this issue to the OpenOCD project. You don't need to do anything.

erhankur, 2023-01-04:
Hi @zfields Can I ask you to test with the attached version?

[openocd-esp32-macos-0.11.0-esp32-20221026-86-g283b87d64.tar.gz](https://github.com/espressif/openocd-esp32/files/10346156/openocd-esp32-macos-0.11.0-esp32-20221026-86-g283b87d64.tar.gz)

zfields, 2023-01-05:
I'm working on something at this very moment, but I will test it ASAP and get back to you.

Thanks!

erhankur, 2023-03-02:
@zfields We provided the solution in the latest release. Can you please confirm your issue has been fixed?

zfields, 2023-03-02:
I'm sorry, but it does _**NOT**_ appear to be fixed.

### My Test

1. `rm -rf` of both `~/.espressif` and `~/esp/esp-idf`
2. reinstalled the 5.0.1-tagged release

    double-checking the versions:
    ```
    $ idf.py --version
    ESP-IDF v5.0.1
    $ which openocd
    /Users/rozzie/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin/openocd
    $ openocd --version
    Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
    $ which xtensa-esp32-elf-gdb
    /Users/rozzie/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin/xtensa-esp32-elf-gdb
    $ xtensa-esp32-elf-gdb --version
    GNU gdb (esp-gdb) 11.2_20220823
    ```
3. `rm -rf` of the `build` folder as a brute force `idf.py clean`
4. rebuilt and re-flashed the app
5. set the breakpoints
6. exactly same problem happens

### Repro Instructions

[sample_project.zip](https://github.com/espressif/openocd-esp32/files/10872713/sample_project.zip)

1. unzip this sample_project
2. open two terminal windows
3. do `. ./export.sh` in each
4. in each terminal, `cd` to the new `sample_project` directory
5. in one terminal, do an `idf.py build` and `idf.py flash`
6. in the same terminal, type `./serve.sh`
7. in the other terminal, type `./debug.sh`

when it hits the first breakpoint in `gdb`, look at the source code and follow the instructions

gerekon, 2023-03-02:
@zfields IDF is not updated to use the latest release. In your example you still use `v0.11.0-esp32-20221026`.
To test it you need to download and run it manually.

zfields, 2023-03-02:
I'm happy to test, but this is the message I responded to:

> We provided the solution in the latest release.

Can you please provide detailed instructions on what exactly you would like for me to test?

gerekon, 2023-03-02:
> I'm happy to test, but this is the message I responded to:

I suppose that was about the [latest OpenOCD release](https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20230221).

> Can you please provide detailed instructions on what exactly you would like for me to test?

It would be great of you checked your scenario described in the first message of this topic. But with the latest OpenOCD release. So your modified command should look like
```
$PATH_TO_LATEST_RELEASE/bin/openocd -s $PATH_TO_LATEST_RELEASE/share/openocd/scripts -f board/esp32s3-builtin.cfg -c "esp32s3.cpu1 configure -rtos FreeRTOS"
```

zfields, 2023-03-02:
Okay, so everything else will be the same, and you want me to additionally update/replace `openocd-esp32` with the latest release.

I'll will test this :point_up_2:, and get back to you.

zfields, 2023-03-02:
I downloaded the latest `openocd-esp32`, but I'm still having the issue.

```
$ which openocd-esp32/bin/openocd
openocd-esp32/bin/openocd
$ openocd-esp32/bin/openocd --version
Open On-Chip Debugger v0.12.0-esp32-20230221 (2023-02-21-11:31)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
```

> _downloaded from: https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20230221_

The problem happens when I say "c" to continue after setting the breakpoint.  On the gdb side it gives the warning that I can't insert the breakpoint, and on the openocd side it says "Warn" "address not writable".

**GDB Output:**

[image]

**OpenOCD Output:**
`openocd-esp32/bin/openocd -f board/esp32s3-builtin.cfg`
```
sample_project $ serve.sh
++ openocd-esp32/bin/openocd -f board/esp32s3-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20230221 (2023-02-21-11:31)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (68:B6:B3:2C:D6:10)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x4200CA15, debug_reason=00000001
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4037A900, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x40374400, debug_reason=00000001
Error: Failed to get flash maps (151587081)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 148034 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 148034 KB
Info : [esp32s3.cpu0] Target halted, PC=0x40374400, debug_reason=00000001
Error: Failed to get flash maps (151587081)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Using flash bank 'esp32s3.cpu0.irom' size 0 KB
Info : [esp32s3.cpu0] Target halted, PC=0x40374400, debug_reason=00000001
Error: Failed to get flash maps (151587081)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Using flash bank 'esp32s3.cpu0.drom' size 0 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Warn : negative reply, retrying
Info : Detected FreeRTOS version: (10.4.3)
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40041A79, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Target halted, PC=0x4200CA15, debug_reason=00000001
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4037A900, debug_reason=00000000
Info : Detected FreeRTOS version: (10.4.3)
Warn : address 0x4200ca36 not writable
Error: [esp32s3.cpu0] Error writing instruction to memory: -4
Error: [esp32s3.cpu0] Failed to write breakpoint instruction (-4)!
Error: [esp32s3.cpu0] Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
```

erhankur, 2023-03-02:
@zfields please make sure the `OPENOCD_SCRIPTS` env variable points to the new location. (path to the v0.12.0-esp32-20230221)

zfields, 2023-03-03:
Eureka! That worked! :+1: 

Do you know which official IDF release will contain the latest release of `openocd-esp32`, and approximately when that release is scheduled to occur?

Thanks,
Zak

erhankur, 2023-03-03:
Afaik, it will be in the IDF v5.1 release around June. Thanks for providing reproducible code and support to get this done.

closed 2023-03-03 by erhankur