espressif/openocd-esp32 #266: ESP32C3 Hart 0 unexpectedly reset!

JanWendler, 2022-12-19:
### Development Kit

ESP32-C3-DevKitM-1 v1.6

### Module or chip used

ESP32-C3

### Debug Adapter

Built in

### OpenOCD version

v0.11.0-esp32-20221026

### Operating System

Windows 10

### Using an IDE ?

no

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

40000

### ESP-IDF version

v4.4.3

### Problem Description

- build and flash project. (works)
- start openocd with: "openocd -f board/esp32c3-builtin.cfg" in a esp-idf console
- start riscv gdb with: "riscv32-esp-elf-gdb.exe .\build\blink.elf " or any other project
- connect with gdb with openocd with: "target extended-remote localhost:3333"
- set a breakpoint at app_main with: "break app_main"
- reset program with: "monitor reset halt"
- continue with: "c"

When I follow these steps I get the following output to the console:

PS D:\Projects\esp32blink> C:\Users\jan.wendler\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (60:55:F9:C1:0C:90)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 25 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 25 KB
Info : Using flash bank 'esp32c3.irom' size 0 KB
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 25 KB
Info : Using flash bank 'esp32c3.drom' size 28 KB
Warn : negative reply, retrying
Error: FreeRTOS: uxTopUsedPriority is not defined, consult the OpenOCD manual for a work-around!
Error: FreeRTOS: uxTopUsedPriority is not defined, consult the OpenOCD manual for a work-around!
Info : Hart 0 unexpectedly reset!
Info : Reset cause (3) - (Software core reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (3) - (Software core reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (3) - (Software core reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (3) - (Software core reset)
Info : Hart 0 unexpectedly reset!

No matter how deep I place the breakpoint, it is never reached because something is causing a reset.
I tested this with my personal project and the standard blinky project with the same result. Both programs work fine when flashed.

I also tried to debug my code in CLion, using openocd, with the same result.

### Debug Logs

```shell
The debug log is to long. I will provide it separately.
```

### Expected behavior

I expect to reach a breakpoint set somewhere in the program.

JanWendler, 2022-12-19:
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/10259899/openocd_log.txt)
[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/10259901/gdb_log.txt)
These are the debug logs

erhankur, 2023-01-02:
Can you also share the `idf.py monitor` logs? 
Flash mapping looks strange to me. There is nothing in the irom section.
```
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 25 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 25 KB
Info : Using flash bank 'esp32c3.irom' size 0 KB
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 25 KB
Info : Using flash bank 'esp32c3.drom' size 28 KB
```

JanWendler, 2023-01-04:
D:\Projects\esp32blink>idf.py monitor
Executing action: monitor
Serial port COM7
Connecting....
Detecting chip type... ESP32-C3
Running idf_monitor in directory d:\projects\esp32blink
Executing "C:\Users\jan.wendler\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.3/tools/idf_monitor.py -p COM7 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace d:\projects\esp32blink\build\blink.elf -m 'C:\Users\jan.wendler\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:/Espressif/frameworks/esp-idf-v4.4.3/\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM7 instead...
--- idf_monitor on \\.\COM7 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16c0
load:0x403cc710,len:0x930
load:0x403ce710,len:0x2d28
entry 0x403cc710
I (30) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (30) boot: compile time 14:27:26
I (30) boot: chip revision: 3
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=07d70h ( 32112) map
I (103) esp_image: segment 1: paddr=00017d98 vaddr=3fc8ba00 size=0142ch (  5164) load
I (108) esp_image: segment 2: paddr=000191cc vaddr=40380000 size=06e4ch ( 28236) load
I (120) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=14a04h ( 84484) map
I (137) esp_image: segment 4: paddr=00034a2c vaddr=40386e4c size=049e8h ( 18920) load
I (141) esp_image: segment 5: paddr=0003941c vaddr=50000010 size=00010h (    16) load
I (146) boot: Loaded app from partition at offset 0x10000
I (149) boot: Disabling RNG early entropy source...
I (166) cpu_start: Pro cpu up.
I (174) cpu_start: Pro cpu start user code
I (174) cpu_start: cpu freq: 160000000
I (174) cpu_start: Application information:
I (177) cpu_start: Project name:     blink
I (182) cpu_start: App version:      1
I (186) cpu_start: Compile time:     Dec 20 2022 13:06:46
I (192) cpu_start: ELF file SHA256:  ba249e2a3a0923da...
I (198) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (204) heap_init: Initializing. RAM available for dynamic allocation:
I (211) heap_init: At 3FC8DCC0 len 0004EA50 (314 KiB): DRAM
I (218) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (224) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (231) spi_flash: detected chip: generic
I (235) spi_flash: flash io: dio
W (239) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (252) sleep: Configure to isolate all GPIO pins in sleep state
I (259) sleep: Enable automatic switching of GPIO sleep configuration
I (266) cpu_start: Starting scheduler.
I (271) example: Example configured to blink addressable LED!
I (271) example: Turning the LED OFF!
I (1281) example: Turning the LED ON!
I (2281) example: Turning the LED OFF!
I (3281) example: Turning the LED ON!
I (4281) example: Turning the LED OFF!
I (5281) example: Turning the LED ON!
I (6281) example: Turning the LED OFF!
I (7281) example: Turning the LED ON!
I (8281) example: Turning the LED OFF!
I (9281) example: Turning the LED ON!

erhankur, 2023-01-16:
I did the same things using idf-master blink example but couldn't see the issue. 
Here are the logs for your reference. 
````
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)   
Saved PC:0x4004c626
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x16dc
load:0x403cc710,len:0x96c
load:0x403ce710,len:0x2e2c
entry 0x403cc710
I (24) boot: ESP-IDF v5.0-dirty 2nd stage bootloader
I (24) boot: compile time 22:27:57
I (24) boot: chip revision: v0.3
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (31) boot.esp32c3: SPI Mode       : DIO
I (36) boot.esp32c3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset  
 Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=083d0h ( 33744) map
I (97) esp_image: segment 1: paddr=000183f8 vaddr=3fc8b800 size=01538h (  5432) load
I (101) esp_image: segment 2: paddr=00019938 vaddr=40380000 size=066e0h ( 26336) load
I (114) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=16780h ( 92032) map
I (132) esp_image: segment 4: paddr=000367a8 vaddr=403866e0 size=04f74h ( 20340) load
I (136) esp_image: segment 5: paddr=0003b724 vaddr=50000010 size=00010h (    16) load
I (141) boot: Loaded app from partition at offset 0x10000       
I (144) boot: Disabling RNG early entropy source...
I (360) cpu_start: Pro cpu up.
I (369) cpu_start: Pro cpu start user code
I (369) cpu_start: cpu freq: 160000000 Hz
I (369) cpu_start: Application information:
I (372) cpu_start: Project name:     blink
I (377) cpu_start: App version:      v5.0-dirty
I (382) cpu_start: Compile time:     Jan 16 2023 22:25:45       
I (388) cpu_start: ELF file SHA256:  b0863a0f26e9d344...        
I (394) cpu_start: ESP-IDF:          v5.0-dirty
I (400) heap_init: Initializing. RAM available for dynamic allocation:
I (407) heap_init: At 3FC8DBE0 len 0004EB30 (314 KiB): DRAM     
I (413) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAMI (420) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM     
I (427) spi_flash: detected chip: generic
I (431) spi_flash: flash io: dio
W (435) spi_flash: Detected size(4096k) larger than the size in 
the binary image header(2048k). Using the size in the binary image header.
I (448) cpu_start: Starting scheduler.
I (453) example: Example configured to blink addressable LED!   
I (453) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (463) example: Turning the LED OFF!
I (1473) example: Turning the LED ON!
I (2473) example: Turning the LED OFF!
````

Openocd
````
PS C:\esp_\openocd-esp32\bin> .\openocd.exe -s ..\share\openocd\scripts\ -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026-87-gd269916f (2023-01-16-14:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
[... 40 lines cut ...]

For help, type "help".
Type "apropos word" to search for commands related to "word"... 
Reading symbols from .\build\blink.elf...
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x40000000 in ?? ()
(gdb) b app_main
Breakpoint 1 at 0x42006b8a: file C:/esp/esp-idf/examples/get-started/blink/main/blink_example_main.c, line 79.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) mon reset halt
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Reset cause (3) - (Software core reset)
(gdb) c
Continuing.
[New Thread 1070140388]
[New Thread 1070142272]
[New Thread 1070135880]
[Switching to Thread 1070140388]

Thread 2 "main" hit Breakpoint 1, app_main () at C:/esp/esp-idf/examples/get-started/blink/main/blink_example_main.c:79
79      {
````

Please try with the clean build and if you still see the issue send me your project directory.

erhankur, 2023-02-03:
@JanWendler is there any update?

JanWendler, 2023-02-03:
> @JanWendler is there any update?

Due to unforeseen circumstances I do not have access to the hardware or the computer from which this problem originated. I don't know when I will be able to test this. You can close the issue if you want.

erhankur, 2023-02-03:
Thanks. If you still have an issue, feel free to re-open this.

closed 2025-08-21 by Indastri