espressif/openocd-esp32 #267: JTAG debugging ESP32-Ethernet-Kit disconnects on OpenOCD startup

szekelyisz, 2022-12-23:
### Development Kit

ESP32-Ethernet-Kit V1.1

### Module or chip used

ESP32-WROVER-B

### Debug Adapter

ESP32-Ethernet-Kit built-in FT2232H

### OpenOCD version

v0.11.0-esp32-20221026 (2022-10-26-14:47)

### Operating System

Linux

### Using an IDE ?

Eclipse

### OpenOCD command line

openocd -s ${openocd_path}/share/openocd/scripts -f board/esp32-ethernet-kit-3.3v.cfg

### JTAG Clock Speed

20000 kHz

### ESP-IDF version

ESP-IDF v5.1-dev-2509-gcfef24863f

### Problem Description

When I start a debug session from Eclipse, my devkit's debugger instantly disconnects on the kernel level and so OpenOCD doesn't see it. This happens every time.

[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/10296970/openocd_log.txt)

Kernel log:
```
Dec 23 20:10:50 neo kernel: [ 4045.385336] usb 3-2.3.3.1.1: new high-speed USB device number 19 using xhci_hcd
Dec 23 20:10:50 neo kernel: [ 4045.489769] usb 3-2.3.3.1.1: New USB device found, idVendor=0403, idProduct=6010, bcdDevice= 7.00
Dec 23 20:10:50 neo kernel: [ 4045.489772] usb 3-2.3.3.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Dec 23 20:10:50 neo kernel: [ 4045.489773] usb 3-2.3.3.1.1: Product: Dual RS232-HS
Dec 23 20:10:50 neo kernel: [ 4045.489774] usb 3-2.3.3.1.1: Manufacturer: FTDI
Dec 23 20:10:50 neo kernel: [ 4045.504007] ftdi_sio 3-2.3.3.1.1:1.0: FTDI USB Serial Device converter detected
Dec 23 20:10:50 neo kernel: [ 4045.504029] usb 3-2.3.3.1.1: Detected FT2232H
Dec 23 20:10:50 neo kernel: [ 4045.504195] usb 3-2.3.3.1.1: FTDI USB Serial Device converter now attached to ttyUSB0
Dec 23 20:10:50 neo kernel: [ 4045.504331] ftdi_sio 3-2.3.3.1.1:1.1: FTDI USB Serial Device converter detected
Dec 23 20:10:50 neo kernel: [ 4045.504349] usb 3-2.3.3.1.1: Detected FT2232H
Dec 23 20:10:50 neo kernel: [ 4045.504463] usb 3-2.3.3.1.1: FTDI USB Serial Device converter now attached to ttyUSB1

[OpenOCD starts]

Dec 23 20:11:34 neo kernel: [ 4089.616057] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
Dec 23 20:11:34 neo kernel: [ 4089.616071] ftdi_sio 3-2.3.3.1.1:1.0: device disconnected
```

### Expected behavior

The debug session to start.

erhankur, 2023-01-02:
Looks like something wrong with the jtag connection. 
```
Debug: 840 21 core.c:328 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 841 21 core.c:1234 jtag_examine_chain(): DR scan interrogation for IDCODE/BYPASS
Debug: 842 21 core.c:328 jtag_call_event_callbacks(): jtag event: TAP reset
Info : 843 21 core.c:1285 jtag_examine_chain(): TAP esp32.cpu0 does not have valid IDCODE (idcode=0xb66ddbb6)
Info : 844 21 core.c:1133 jtag_examine_chain_display(): JTAG tap: esp32.cpu1 tap/device found: 0xdb36eddb (mfg: 0x6ed (<unknown>), part: 0xb36e, ver: 0xd)
Warn : 845 21 core.c:1133 jtag_examine_chain_display(): JTAG tap: esp32.cpu1       UNEXPECTED: 0xdb36eddb (mfg: 0x6ed (<unknown>), part: 0xb36e, ver: 0xd)
Error: 846 21 core.c:1133 jtag_examine_chain_display(): JTAG tap: esp32.cpu1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Debug: 847 21 core.c:1478 jtag_tap_init(): Created Tap: auto0.tap @ abs position 2, irlen 0, capture: 0x1 mask: 0x3
Info : 848 21 core.c:1133 jtag_examine_chain_display(): JTAG tap: auto0.tap tap/device found: 0xeddb36ed (mfg: 0x376 (Holtek Semiconductor Inc), part: 0xddb3, ver: 0xe)
Debug: 849 21 core.c:1478 jtag_tap_init(): Created Tap: auto1.tap @ abs position 3, irlen 0, capture: 0x1 mask: 0x3
Info : 850 21 core.c:1285 jtag_examine_chain(): TAP auto1.tap does not have valid IDCODE (idcode=0x36eddb36)
Debug: 851 21 core.c:1478 jtag_tap_init(): Created Tap: auto2.tap @ abs position 4, irlen 0, capture: 0x1 mask: 0x3
Info : 852 21 core.c:1133 jtag_examine_chain_display(): JTAG tap: auto2.tap tap/device found: 0x9b76ed9b (mfg: 0x6cd (<unknown>), part: 0xb76e, ver: 0x9)
```

Please check your application(you can try with hello world first), usb cable and dip switch state for the JTAG pins. They all must be at ON state.

szekelyisz, 2023-01-04:
Hi @erhankur I tried all of your suggestions, even flipped the dip switches there and back, no difference.

I don't see why a bad JTAG connection would cause a kernel level disconnect in the first place. My theory is that the disconnect happens first, that's why OpenOCD can't find the devkit.

Can somebody confirm that this is indeed a problem with my devkit and not a software bug? Here are all my versions:

* xtensa-esp-elf-gdb: GDB for Xtensa
  - 12.1_20221002 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 12.1_20221002 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2022r1-11.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2022r1-11.2.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2022r1-11.2.0 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - 15.0.0-23786128ae (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2022r1-11.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.24.0 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20221026 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20220823 (recommended, installed)

szekelyisz, 2023-01-14:
The problem is gone. I cleaned my sdkconfig, started from scratch, and it worked.

closed 2023-01-14 by szekelyisz