espressif/openocd-esp32 #269: When debugging with jtag, all variables show <unavailable> when entering the rtos task

xiaohuihere, 2023-02-02:
### Development Kit

ESP-WROVER-KIT-VE

### Module or chip used

ESP32-WROVER-E

### Debug Adapter

FT2232

### OpenOCD version

v0.11.0-esp32-20221026

### Operating System

window10

### Using an IDE ?

espressif IDF

### OpenOCD command line

C:\Users\22380\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe -f C:\Users\22380\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts\board\esp32-wrover-kit-3.3v.cfg

### JTAG Clock Speed

20000

### ESP-IDF version

v4.4.3

### Problem Description

# When debugging with jtag, all variable values are displayed as <unavailable> when entering the rtos task
While working on a project, I wanted to try to go into one of the rtos tasks to see the value of a variable. However, I found that all the variables in the task showed <unavailable>.
The following is a screenshot of the scenario.
[image]

### Debug Logs

```shell
openOCD:
PS C:\Users\22380> C:\Users\22380\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe -f C:\Users\22380\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts\board\esp32-wrover-kit-3.3v.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 26000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40168578, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32.cpu1] Target halted, PC=0x40168578, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32.cpu0] Target halted, PC=0x40168578, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x40168578, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu0.flash' size 4096 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu0.irom' size 640 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu0.drom' size 100 KB
Info : New GDB Connection: 1, Target esp32.cpu0, state: halted
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu0.irom' size 640 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu0.drom' size 100 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu1.irom' size 640 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu1.drom' size 100 KB
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : New GDB Connection: 1, Target esp32.cpu0, state: halted
Info : [esp32.cpu0] Target halted, PC=0x400D9231, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x40168578, debug_reason=00000000
Info : [esp32.cpu0] Target halted, PC=0x400DA0CF, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x400D4F46, debug_reason=00000000

vscode output:
[Debug Adapter]
DEBUG_ADAPTER_STARTED
2023-02-02 20:43:14,465 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
[Debug Adapter]
DEBUG_ADAPTER_READY2CONNECT
2023-02-02 20:43:14,469 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2023-02-02 20:43:25,132 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
[Debug Adapter]
DEBUG_ADAPTER_STARTED
2023-02-02 20:45:01,071 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
[Debug Adapter]
DEBUG_ADAPTER_READY2CONNECT
2023-02-02 20:45:01,075 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT

vscode debugconsole:
Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
Debug Adapter initialized
Debug Adapter is running
8
WARNING! This feature can't update UI after execution.
```

### Expected behavior

It should have shown the value of the variable in the left display bar, like the int variable a, which should have had a value of 1 or 0

### Screenshots

task:
[image]

Create a task：
[image]

erhankur, 2023-02-03:
This seems related to this issue. https://github.com/espressif/openocd-esp32/issues/246

xiaohuihere, 2023-02-03:
> 这似乎与这个问题有关。#246

I have ever use the -O0, but it doesn't work
[image]
By the way, here is the openocd log with -d3
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/10580170/openocd_log.txt)

Yecgaa1, 2023-02-13:
I have the same problem

gerekon, 2023-02-13:
@Yecgaa1 @xiaohuihere Do you see this in when debugging in VsCode only? Or you see the same behaviour in Eclipse too? I see that the last screenshot is from Eclipse while original one is from VsCode.

If you see this in VsCode only could you collect Debug Adapter's log. It should be shown in `Output` window. To enable verbose log you need to increase `logLevel` in `launch.json`. Please, see [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md#use-the-esp-idf-debug-adapter) for details.

cc @brianignacio5

xiaohuihere, 2023-02-17:
Yes, its still showing up that way on Eclipse. However after I changed the logLevel to 5 I get this full output and if it helps to find the problem then I would be very grateful.
[output.txt](https://github.com/espressif/openocd-esp32/files/10764016/output.txt)

xiaohuihere, 2023-02-17:
> @Yecgaa1 @xiaohuihere只在VsCode中调试时，您会看到这个吗？或者您在Eclipse中也看到了相同的行为？我看到到最后一张截图来自Eclipse，而最初截图来自VsCode。
> 
> 如果你只在 VsCode 中看到这个，你可以收集 Debug Adapter 的日志。它应该显示在`Output`窗口中。要启用详细日志，你需要`logLevel`添加`launch.json`。请参阅[此处](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md#use-the-esp-idf-debug-adapter)了解详细。
> 
> 抄送@brianignacio5

Yes, its still showing up that way on Eclipse. However after I changed the logLevel to 5 I get this full output and if it helps to find the problem then I would be very grateful.
[output.txt](https://github.com/espressif/openocd-esp32/files/10764024/output.txt)

erhankur, 2023-12-29:
@xiaohuihere are you able to fix the issue? Still needs help from our side?

erhankur, 2024-01-15:
Closed due to inactivity. Feel free to re-open if you have any questions.

closed 2024-01-15 by erhankur