espressif/openocd-esp32 #270: Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND (IDFGH-9317)

Zeni241, 2023-02-04:


erhankur, 2023-02-06:
@Zeni241 Your problem looks like a driver issue.  Please check this https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html#configure-usb-drivers

erhankur, 2023-02-11:
@Zeni241 Closing this one. You can reopen it if you have any other issues. Also please check previous driver issues. You will find something helpful for you.

closed 2023-02-11 by erhankur

torkleyy, 2023-09-07:
I had to follow this guide to get it to work:

https://community.platformio.org/t/esp32-pio-unified-debugger/4541/20

Tomblarom, 2024-03-07:
@torkleyy Yes that solved it dor me! Thanks :) **NOTE:** zadiag had to be executed as admin!

SpaceCheetah, 2024-07-11:
Following that guide just led to it not working at all, had to roll back to the windows ones. With those, I can flash with idf.py flash, but OpenOCD fails with error not found.

yamaan93, 2024-10-02:
is there no fix to this? the zadiag solution is only for chips without an on-chip debugger.....I'm also encountering this problem