espressif/openocd-esp32 #271: [Windows] Trace failing with error 22 on write

KonssnoK, 2023-02-05:
### Development Kit

ESP32S3-devkitC

### Module or chip used

ESP32S3

### Debug Adapter

USB Jtag

### OpenOCD version

v0.11.0-esp32-20221026

### Operating System

Windows

### Using an IDE ?

No

### OpenOCD command line

openocd.exe -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v4.4.4

### Problem Description

This issue seems to occur only on Windows devices.

```
Error: Incomplete block sz 28139, wr 1071
Error: Failed to write 28139 bytes to out file (22)! Written -1.
Error: Failed to write 28139 bytes to dest 0!
Error: Failed to process 28139 bytes!
Error: Failed to process trace block 16380 bytes!
Disconnect targets...
```

trace started with
```
mon reset halt
thb initTrax
c
mon esp apptrace start "file://$(TZ_OUT)" 0 -1 -1 0 0
mon resume
```

### Expected behavior

openocd should continue to write to the file indefinitely

erhankur, 2023-02-11:
@KonssnoK Can you send verbose OpenOCD debug logs?

KonssnoK, 2023-02-14:
Here they are
[openocdLog.log](https://github.com/espressif/openocd-esp32/files/10731739/openocdLog.log)

We actually spent a lot of time on this issue and it's related to TRAX buffer getting invalidated for unknown reasons.
This generates packets which are non consistent that leads to this error.

We do have an internal solution, but involves changing your ESP/OpenOCD protocol in app_trace.

I will ask a colleague to elaborate on my remarks

antoDimaggio, 2023-02-14:
Hello everybody.
As @KonssnoK told you we developed an internal solution that drops the corrupted apptrace blocks.
First of all it is important to tell you that we are using the Precepio Tracealyzer library. 
This is implemented by a single task pinned to core 0. The tracealyzer task periodically takes events from an internal FIFO and writes them into apptrace blocks, so concurrent access to the apptrace memory buffer may be ruled out.

We basically added a checksum and a fixed magic number to the packet header both on the target and on OpenOCD. OpenOCD verifies each apptrace packet and if the checksum is corrupted or the write and block size mismatch it drops the packet.

```
--- a/components/app_trace/app_trace_membufs_proto.c
+++ b/components/app_trace/app_trace_membufs_proto.c
@@ -23,6 +23,9 @@ typedef struct {
 #else
     uint16_t   block_sz; // size of allocated block for user data
     uint16_t   wr_sz;    // size of actually written data
+    uint32_t   magicHeader;  //fixed value 0xA5A5A5A5
+    uint32_t   count;     //packet counter value
+    uint32_t   crc;       //simple sum
 #endif
 } esp_tracedata_hdr_t;
```

With this patch we were able to see that 1-2% of the received packets are corrupted. Dropping the packets allows us to have a stable trace but we have too many missed events and we are not able to say where these packets get corrupted.

erhankur, 2023-02-14:
@KonssnoK, @antoDimaggio thanks for reporting this issue in detail. Actually, this is a known failure for the S3 chips.  We have a plan to fix it soon. I will inform you when we have a solution in the master branch.

KonssnoK, 2023-02-15:
@erhankur Do you have idea of what is failing?
Because from our small investigation, the TRAX is not touched by anything else than openOCD and app_trace_write (on single core).

gerekon, 2023-02-15:
@KonssnoK Actually we are going to check it with our HW team. Need to decide if it is a HW bug or not

KonssnoK, 2023-02-16:
@gerekon we didn't take a look at the wrong values but we know for sure that the corruption seems to be small (we reduced the header with crc content to 512B and it was still one every X occurrences).
I wouldn't be surprised if it was a single word getting mixed up.

KonssnoK, 2023-04-13:
is there any news?
@gerekon

erhankur, 2023-12-21:
@KonssnoK can you try with the latest OpenOCD release? We have a fix there.

erhankur, 2024-01-15:
@KonssnoK feel free to re-open if your issue continues.

closed 2024-01-15 by erhankur