espressif/openocd-esp32 #272: Debug esp32-s3 CPU1 while FreeRTOS ruining only on CPU0

5ami, 2023-02-08:
### OpenOCD version

v0.11.0-esp32-20220706

### Operating System

Windows

### Using an IDE ?

VScode

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### ESP-IDF version

v5.0

### Problem Description

I am using esp32-s3 by having the scheduler only in the PRO Core (using the option Run freeRTOS only on first core). Then I did some modification in the startup.c file to start the APP CPU without FreeRTOS.

So I have now one core running freeRTOS and the other one running without.

I set a High level 5 interrupt in the APP CPU triggered by a PWM counter and I am toggling a GPIO inside the ISR (In assembly). my Interrupt is working perfectly on the APP CPU.

however my problem is that when I try to run the Debugger, it can only pause on breakpoints set on the PRO CPU, when I set a break point on the APP CPU. the debugger pause at a _UserExceptionVector. like shown in the screenshot.
 

Or it keeps spinning and blocks sometimes.

it seems that the Debugger is only aware of the PRO_CPU not the APP_CPU. so How to solve that ?

### Debug Logs

```shell
User : 3 9 options.c:63 configuration_output_handler(): debug_level: 3
User : 4 9 options.c:63 configuration_output_handler(): 
Debug: 5 10 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 10 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 10 options.c:246 add_default_dirs(): exepath=C:/Users/U405172/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin
Debug: 8 10 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 9 10 configuration.c:44 add_script_search_dir(): adding C:\Users\U405172\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts
Debug: 10 10 configuration.c:44 add_script_search_dir(): adding C:/Users/U405172/AppData/Roaming/OpenOCD
Debug: 11 10 configuration.c:44 add_script_search_dir(): adding C:/Users/U405172/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin/../share/openocd/site
Debug: 12 10 configuration.c:44 add_script_search_dir(): adding C:/Users/U405172/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin/../share/openocd/scripts
Debug: 13 10 command.c:166 script_debug(): command - ocd_find board/esp32s3-builtin.cfg
Debug: 14 10 configuration.c:99 find_file(): found C:\Users\U405172\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/board/esp32s3-builtin.cfg
Debug: 15 13 command.c:166 script_debug(): command - ocd_find interface/esp_usb_jtag.cfg
Debug: 16 14 configuration.c:99 find_file(): found C:\Users\U405172\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/interface/esp_usb_jtag.cfg
Debug: 17 14 command.c:166 script_debug(): command - adapter driver esp_usb_jtag
Info : 18 14 transport.c:118 allow_transports(): only one transport option; autoselect 'jtag'
Debug: 19 14 command.c:166 script_debug(): command - espusbjtag vid_pid 0x303a 0x1001
Info : 20 14 esp_usb_jtag.c:899 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Debug: 21 14 command.c:166 script_debug(): command - espusbjtag caps_descriptor 0x2000
Info : 22 14 esp_usb_jtag.c:910 esp_usb_jtag_caps_descriptor(): esp_usb_jtag: capabilities descriptor set to 0x2000
Debug: 23 14 command.c:166 script_debug(): command - adapter speed 40000
Debug: 24 14 adapter.c:180 adapter_config_khz(): handle adapter khz
Debug: 25 14 adapter.c:144 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 26 14 adapter.c:144 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 27 14 command.c:166 script_debug(): command - ocd_find target/esp32s3.cfg
Debug: 28 14 configuration.c:99 find_file(): found C:\Users\U405172\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/target/esp32s3.cfg
Debug: 29 15 command.c:166 script_debug(): command - transport select jtag
Warn : 30 15 transport.c:287 jim_transport_select(): Transport "jtag" was already selected
Debug: 31 15 command.c:166 script_debug(): command - ocd_find bitsbytes.tcl
Debug: 32 15 configuration.c:99 find_file(): found C:\Users\U405172\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/bitsbytes.tcl
Debug: 33 15 command.c:166 script_debug(): command - expr 1 << $x
Debug: 34 15 command.c:166 script_debug(): command - expr $x + 1
[... 331 lines cut ...]
Debug: 330 25 xtensa.c:2292 xtensa_build_reg_cache(): Special reg 'mecr' (161) does not exist
Debug: 331 25 xtensa.c:2292 xtensa_build_reg_cache(): Special reg 'mevaddr' (162) does not exist
Debug: 332 25 semihosting_common.c:118 semihosting_common_init():  
Debug: 333 25 semihosting_common.c:118 semihosting_common_init():  
Error: 334 66 libusb_helper.c:190 jtag_libusb_open(): libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Error: 335 66 libusb_helper.c:190 jtag_libusb_open(): libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: 336 67 esp_usb_jtag.c:651 esp_usb_jtag_init(): esp_usb_jtag: could not find or open device!
Debug: 337 67 command.c:556 run_command(): Command 'init' failed with error code -4
User : 338 68 command.c:619 command_run_line(): 
Debug: 339 68 esp_xtensa.c:133 esp_xtensa_target_deinit(): start
Debug: 340 68 xtensa.c:2476 xtensa_target_deinit(): start
Debug: 341 68 target.c:2239 target_free_all_working_areas_restore(): freeing all working areas
Debug: 342 68 esp_xtensa.c:133 esp_xtensa_target_deinit(): start
Debug: 343 68 xtensa.c:2476 xtensa_target_deinit(): start
Debug: 344 68 target.c:2239 target_free_all_working_areas_restore(): freeing all working areas
```

### Expected behavior

I expect the debigger to pause at the break point set on the APP_CPU

### Screenshots

 
[image]

5ami, 2023-02-08:
Sorry I tried To delete that one from https://github.com/espressif/esp-idf/issues/10724. as this is I guess more related to OpenOCD-esp32 repo then esp-idf one

erhankur, 2023-02-11:
I am not sure if it will help but it is worth trying.  Can you replace target create commands inside `esp32s3.cfg` like below? 

```
target create $_TARGETNAME_0 $_CHIPNAME -endian little -chain-position $_TARGETNAME_0 -coreid 0 -rtos $_RTOS
target create $_TARGETNAME_1 $_CHIPNAME -endian little -chain-position $_TARGETNAME_1 -coreid 1
```

If you can send an example binary and some instructions to create the issue, I can also give it a try.

5ami, 2023-02-13:
Hi @erhankur. 
I have already tried making changes   to the esp32s3.cfg but with no success. I still get the  _UserExceptionVector. 
I will try try to prepare a setup to reproduce the issue, as booting the APP_CPU without freeRTOS isn't supported in ESP-IDF. and the setup isn't stable during the boot, but once it boots it gets stable.

5ami, 2023-02-16:
Hi @erhankur, please check the code here: https://github.com/5ami/BM_Setup_ESP32s3 , I have sent you an invite.

erhankur, 2023-02-16:
Thanks @5ami I will test it soon.

5ami, 2023-02-26:
@erhankur any updates?

erhankur, 2023-03-02:
hi @5ami , I was a bit busy this week. But I just did a quick try with your application. Looks like there are smp design limitations. I will share my detailed analysis.

5ami, 2023-03-02:
Thanks @erhankur, waiting for your analysis.

5ami, 2023-05-11:
@erhankur any updates on this issue ?

erhankur, 2024-01-15:
Unfortunately, I don't have enough time to investigate this deeply. If you have any findings let us know and we will be happy to update related parts of OpenOCD.

closed 2024-01-15 by erhankur