espressif/openocd-esp32 #273: After updated OpenOCD there is FLASH Programming Error

LisaTamanti1965, 2023-02-10:
### Development Kit

ESP32 Devkit

### Module or chip used

ESP32 WROVER B with 8MB FLASH and 8MB SPIRAM

### Debug Adapter

ESP32DevKit

### OpenOCD version

20230202

### Operating System

Windows

### Using an IDE ?

Visual Studio

### OpenOCD command line

C:\Users\bope02\AppData\Local\VisualGDB\EmbeddedDebugPackages\com.sysprogs.esp32.core\bin\openocd.exe -c "gdb_port 58397" -c "telnet_port 58395" -f interface/ftdi/esp32_devkitj_v1.cfg -c "adapter_khz 3000" -f target/esp32.cfg -c "echo VisualGDB_OpenOCD_Ready"

### JTAG Clock Speed

3000

### ESP-IDF version

ESP-IDF:          v4.4.1

### Problem Description

I'm using Visual Studio IDE with Visual GDB.
Before to update to OpenOCD 20230202 everything was functioning (20220503)

I'm using a personal partition table:

I (69) boot:  0 otadata          OTA data         01 00 00010000 00002000
I (76) boot:  1 ota_0            OTA app           00 10 00020000 00271000
I (84) boot:  2 ota_1            OTA app           00 11 002a0000 00271000
I (91) boot:  3 nvs              WiFi data           01 02 00511000 001f4000

Now when I try to DEBUG   I have a:  FLASH Programming failed with this LOG:

"JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)\n"
@"JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)\n"
@"esp32.cpu0: Debug controller was reset.\n"
@"esp32.cpu0: Core was reset.\n"
@"esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000\n"
@"esp32.cpu0: Core was reset.\n"
@"esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000\n"
@"esp32.cpu1: Debug controller was reset.\n"
@"esp32.cpu1: Core was reset.\n"
@"esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000\n"
@"timed out while waiting for target halted\n"
@"esp32.cpu0: Target halted, PC=0x4000C2DD, debug_reason=00000000\n"
@"xtensa_wait_algorithm: not halted 0, pc 0x4000c2dd, ps 0x60025\n"
@"Failed to wait algorithm (-302)!\n"
@"Algorithm run failed (-302)!\n"
@"**Too many flash mappings 11072892! Must be 2.\n**"
@"Failed to get flash mappings (-4)!\n"
@"Target is already running an algorithm\n"
@"Failed to start algorithm (-4)!\n"
@"Failed to get flash size!\n"
@"Target is already running an algorithm\n"
@"Failed to start algorithm (-4)!\n"
@"Failed to get flash size!\n"
@"Failed to probe flash, size 0 KB\n"
@"auto_probe failed\n"
@"Failed to find bank 'esp32.cpu1.flash'!\n"
@"\n"
^done
-target-disconnect
=thread-group-exited,id="i1"
^done
-gdb-exit
^exit

I tryed to restore old OpenOCD (20220503)   version but without success.

### Debug Logs

```shell
Before to start DEBUG I make a test  that seems to be OK (below LOG from TEST)

VisualGDB_OpenOCD_Ready
Info : Listening on port 6666 for tcl connections
Info : Listening on port 58395 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : clock speed 3000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : starting gdb server for esp32.cpu0 on 58397
Info : Listening on port 58397 for gdb connections
Info : accepting 'gdb' connection on tcp/58397
Info : esp32.cpu0: Target halted, PC=0x401EB37E, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x401EB37E, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : esp32.cpu0: Target halted, PC=0x4009260E, debug_reason=00000001
Error: Failed to get flash maps (4294967290)!
Warn : Failed to get flash mappings (-4)!
Info : esp32.cpu0: Target halted, PC=0x4009260E, debug_reason=00000001
Info : esp32.cpu0: Target halted, PC=0x4009260E, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32.cpu0.flash' size 8192 KB
Info : esp32.cpu0: Target halted, PC=0x4009260E, debug_reason=00000001
Error: Failed to get flash maps (4294967290)!
Warn : Failed to get flash mappings (-4)!
Info : esp32.cpu0: Target halted, PC=0x4009260E, debug_reason=00000001
Info : Using flash bank 'esp32.cpu0.irom' size 0 KB
Info : esp32.cpu0: Target halted, PC=0x4009260E, debug_reason=00000001
Error: Failed to get flash maps (4294967290)!
Warn : Failed to get flash mappings (-4)!
Info : esp32.cpu0: Target halted, PC=0x4009260E, debug_reason=00000001
Info : Using flash bank 'esp32.cpu0.drom' size 0 KB
Info : New GDB Connection: 1, Target esp32.cpu0, state: halted
Info : dropped 'gdb' connection
shutdown command invoked
```

### Expected behavior

I expected to continue to debug my target as before updated OpenOCD.
I tryed to restore previous OpenOCD version but without success.

erhankur, 2023-02-11:
@LisaTamanti1965 Where did you get this version? 20230202 

If you get it from sysprogs openocd fork, then please create an issue there. 

Or you can try with our latest [release](https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20221026)

LisaTamanti1965, 2023-02-13:
Yes I got that version from sysprogs..I already asked to them ..but they answer to ask to you..

But I have a question...what does it mean these logs from openOCD?:

Too many flash mappings 11072892! Must be 2.\n"
@"Failed to get flash mappings (-4)!\n"
@"Target is already running an algorithm\n"
@"Failed to start algorithm (-4)!\n"
@"Failed to get flash size!\n"
@"Target is already running an algorithm\n"

erhankur, 2023-02-13:
Can you share the idf application opening logs? 
And can you also try to use our latest release?

LisaTamanti1965, 2023-02-13:
This is  the idf application opening logs when I download it with FLASH TOOLS on my target:

I (27) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (27) boot: compile time 15:28:40
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 otadata          OTA data         01 00 00010000 00002000
I (75) boot:  1 storage          Unknown data     01 82 00012000 00006000
I (82) boot:  2 ota_0            OTA app          00 10 00020000 00271000
I (90) boot:  3 ota_1            OTA app          00 11 002a0000 00271000
I (97) boot:  4 nvs              WiFi data        01 02 00511000 001f4000
I (105) boot: End of partition table
I (109) boot_comm: chip revision: 1, min. application chip revision: 0
I (116) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=35868h (219240) map
I (204) esp_image: segment 1: paddr=00055890 vaddr=3ffbdb60 size=058cch ( 22732) load
I (213) esp_image: segment 2: paddr=0005b164 vaddr=40080000 size=04eb4h ( 20148) load
I (222) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=12da20h (1235488) map
I (669) esp_image: segment 4: paddr=0018da48 vaddr=40084eb4 size=19488h (103560) load
I (712) esp_image: segment 5: paddr=001a6ed8 vaddr=50000000 size=00010h (    16) load
I (727) boot: Loaded app from partition at offset 0x20000
I (727) boot: Disabling RNG early entropy source...
I (739) psram: This chip is ESP32-D0WD
I (741) spiram: Found 64MBit SPI RAM device
I (741) spiram: SPI RAM mode: flash 40m sram 40m
I (744) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (751) cpu_start: Pro cpu up.
I (755) cpu_start: Starting app cpu, entry point is 0x400816f8
I (748) cpu_start: App cpu up.
I (1646) spiram: SPI SRAM memory test OK
I (1655) cpu_start: Pro cpu start user code
I (1655) cpu_start: cpu freq: 160000000
I (1655) cpu_start: Application information:
I (1658) cpu_start: Project name:     FAAC-IOT
I (1663) cpu_start: App version:      1_4_0-67-g77a20a1-dirty
I (1669) cpu_start: Compile time:     Feb  9 2023 11:34:14
I (1676) cpu_start: ELF file SHA256:  b29365d68dad5dd5...
I (1682) cpu_start: ESP-IDF:          v4.4.1-dirty
I (1688) heap_init: Initializing. RAM available for dynamic allocation:
I (1695) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1701) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1707) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1713) heap_init: At 3FFC54E8 len 0001AB18 (106 KiB): DRAM
I (1719) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1726) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1732) heap_init: At 4009E33C len 00001CC4 (7 KiB): IRAM
I (1739) spiram: Adding pool of 4084K of external SPI memory to heap allocator
I (1748) spi_flash: detected chip: gd
I (1751) spi_flash: flash io: dio
I (1757) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1770) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations

erhankur, 2023-02-13:
Your application is at offset 0x20000 

You should tell offset to the OpenOCD like below; 

-c "init; halt; esp appimage_offset 0x20000"

Details are here. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#flash-mappings-vs-sw-flash-breakpoints

erhankur, 2023-02-20:
@LisaTamanti1965 Your issue should be fixed now.  You can re-open if you still have a problem.

closed 2023-02-20 by erhankur