espressif/openocd-esp32 #274: gdb does not show stack trace

arbrauns, 2023-02-17:
### Development Kit

ESP32-DevKitC-V4

### Module or chip used

ESP32-WROVER-IE

### Debug Adapter

JLink

### OpenOCD version

v0.11.0-esp32-20221026 (2022-10-26-14:47)

### Operating System

Ubuntu 22.04.1 LTS

### Using an IDE ?

-

### OpenOCD command line

openocd -f interface/jlink.cfg -f target/esp32.cfg -c 'adapter speed 1000'

### JTAG Clock Speed

1000

### ESP-IDF version

v5.0

### Problem Description

1. `cd examples/get-started/hello_world/`
2. `idf.py build`
3. `idf.py flash`
4. `openocd -d3 -f interface/jlink.cfg -f target/esp32.cfg -c 'adapter speed 1000' &| openocd.log`
5.
```
$ cat gdbinit
target remote :3333
set remote hardware-watchpoint-limit 2
mon reset halt
flushregs
thb app_main
```
6. `xtensa-esp32s2-elf-gdb -ex "set remotelogfile gdb_log.txt" -x gdbinit build/hello_world.elf`
7.
```
(gdb) bt
#0  app_main () at $HOME/dev/misc/esp-idf/examples/get-started/hello_world/main/hello_world_main.c:15
#1  0x800882e2 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

### Debug Logs

```shell
https://gist.github.com/arbrauns/8f0480c9f82ccdfa9d6031a4575cf9a1
```

### Expected behavior

A working backtrace as in the example [here](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/api-guides/jtag-debugging/debugging-examples.html#jtag-debugging-examples-command-line-01):

```
(gdb) bt
#0  app_main () at /home/user-name/esp/blink/main/./blink.c:43
#1  0x400d057e in main_task (args=0x0) at /home/user-name/esp/esp-idf/components/esp32/./cpu_start.c:339
```

arbrauns, 2023-02-20:
I don't know how unwinding works on xtensa, but it seems to try to get the previous stack frame's PC from register `a0`:

```
(gdb) info registers
pc             0x400d519c          0x400d519c <app_main>
sar            0x0                 0
ps             0x10                16
threadptr      0xc2bcfffe          3267166206
pwrctl         0x0                 0
pwrstat        0x0                 0
eristat        0x0                 0
cs_itctrl      0x0                 0
cs_claimset    0x0                 0
cs_claimclr    0x60620             394784
cs_lockaccess  0x0                 0
cs_lockstatus  0x3ffb43b0          1073431472
cs_authstatus  0x40000954          1073744212
fault_info     0x40000a28          1073744424
trax_id        0x40000af8          1073744632
trax_control   0x40000c68          1073745000
trax_status    0x0                 0
trax_data      0x0                 0
trax_address   0x0                 0
trax_pctrigger 0x18000             98304
trax_pcmatch   0x0                 0
trax_delay     0x0                 0
trax_memstart  0x1300124c          318771788
trax_memend    0x40080000          1074266112
pmg            0x4                 4
pmpc           0x2                 2
pm0            0x3092d41           50933057
pm1            0xcdcd              52685
pmctrl0        0x0                 0
pmctrl1        0x0                 0
pmstat0        0x0                 0
pmstat1        0x3145173           51663219
ocdid          0x0                 0
ocd_dcrclr     0x0                 0
ocd_dcrset     0x0                 0
ocd_dsr        0x0                 0
a0             0x800882e2          -2146925854
a1             0x3ffb5520          1073435936
a2             0x3                 3
a3             0x60e23             396835
a4             0x60e20             396832
a5             0x10                16
a6             0x0                 0
a7             0x0                 0
a8             0x8008a690          -2146916720
a9             0x3ffb54f0          1073435888
a10            0x3ffaffe4          1073414116
a11            0x3ffaffe4          1073414116
a12            0x3ffaff5c          1073413980
a13            0xfffffffe          -2
a14            0x80000             524288
a15            0xcdcd              52685
```

arbrauns, 2023-02-20:
Oh, d'oh, I found the issue - I simply used `xtensa-esp32s2-elf-gdb` from the tutorial without realizing that it's only for esp32s2, backtrace works fine with `xtensa-esp32-elf-gdb`:

```
#0  app_main () at $HOME/dev/misc/esp-idf/examples/get-started/hello_world/main/hello_world_main.c:15
#1  0x400e55bd in main_task (args=<optimized out>) at $HOME/dev/misc/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131
#2  0x40087fec in vPortTaskWrapper (pxCode=0x400e5550 <main_task>, pvParameters=0x0) at $HOME/dev/misc/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```

Would it be possible to add a big fat error when you use the wrong gdb?

erhankur, 2023-03-02:
cc @Lapshin , do you think is it possible?

Lapshin, 2023-03-13:
@erhankur , this should be done by adding a chip name into ELF notes section + checking this record on GDB startup.

But I don't think this is a critical issue to do. Because IDF provides the command `idf.py gdb` which choose the right GDB to execute.

Also, for this particular case, it could be `idf.py gdb --gdbinit gdbinit` (put `set remotelogfile gdb_log.txt` into `gdbinit` file)

cc @arbrauns

closed 2023-05-04 by erhankur