espressif/openocd-esp32 #275: error: invalid command name adapter

FabricioPazCosta, 2023-02-20:
### Development Kit

ESP32-DEVKITC

### Module or chip used

ESP-WROOM-32

### Debug Adapter

ESP-PROG

### OpenOCD version

openocd-0.12.0

### Operating System

Windows

### Using an IDE ?

Visual Studio Code

### OpenOCD command line

openocd -f board/esp32-wrover-kit-3.3v.cfg

### JTAG Clock Speed

20000

### ESP-IDF version

esp-idf-v5.0

### Problem Description

PS C:\Espressif\frameworks\esp-idf-v5.0\examples\get-started\blink> openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f /target/esp32.cfg
over-kit-3.3v.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.htmlC:/openocd/bin/..//scripts/interface/ftdi/esp32_devkitj_v1.cfg:7: Error: invalid command name "adapter"
in procedure 'script'
at file "embedded:startup.tcl", line 60
at file "C:/openocd/bin/..//scripts/board/esp32-wrover-kit-3.3v.cfg", line 11at file "C:/openocd/bin/..//scripts/interface/ftdi/esp32_devkitj_v1.cfg", line 7

### Expected behavior

user-name@computer-name:~/esp/esp-idf$ openocd -f board/esp32-wrover-kit-3.3v.cfg
Open On-Chip Debugger  v0.10.0-esp32-20190708 (2019-07-08-11:04)
Licensed under GNU GPL v2
For bug reports, read
        https://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).

### Screenshots

[image]

erhankur, 2023-02-20:
You are using new config files with the incompatible OpenOCD version.  I would suggest to install OpenOCD from scratch with the ESP-IDF install script.

erhankur, 2023-03-02:
@FabricioPazCosta I don't see an issue here. Feel free to reopen if you see something different in the next steps.

closed 2023-03-02 by erhankur