espressif/openocd-esp32 #276: new version of "openocd-esp32" never works

EduardShaid, 2023-02-23:
### OpenOCD version

v0.12.0-esp32-20230221

### Operating System

Win 10

### Using an IDE ?

Yes

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### ESP-IDF version

v4.4.4

### Problem Description

Can someone explain why when I install in Espressif-IDE a new version of "openocd-esp32", for example, I now installed version v0.12.0-esp32-20230221 instead of v0.11.0-esp32-20221026, then it never works?

### Debug Logs

```shell
Open On-Chip Debugger v0.12.0-esp32-20230221 (2023-02-21-11:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Flashing C:/Espressif/frameworks/esp-idf-v4.4.4/workspace/restful_server/build/bootloader/bootloader.bin at 0x0
Info : esp_usb_jtag: serial (7C:DF:A1:E6:94:80)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Target halted, PC=0x420076A8, debug_reason=00000001
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x420808FE, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Target halted, PC=0x400559CC, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Error: Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Error: Failed to get flash size!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32s3.cpu0.flash'!
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Error: Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Error: Failed to get flash size!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
shutdown command invoked
```

### Expected behavior

no

### Screenshots

no

igrr, 2023-02-23:
Could you please check, what is the value of OPENOCD_SCRIPTS variable when you are launching `openocd`?

EduardShaid, 2023-02-23:
Yes, sorry, I have this:

-s ${openocd_path}/share/openocd/scripts -f board/esp32s3-builtin.cfg

erhankur, 2023-02-23:
@EduardShaid some questions;

1. What kind of S3 board do you have? Flash size, with PSRAM or not? 
2. Where is your application located? By default, OpenOCD reads the partition table at 0x8000. Is that the same for you? Please check here. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#flash-mappings-vs-sw-flash-breakpoints
3. Can you paste idf application console output to understand what is going on the application side? 
4. Can you run OpenOCD with `-c "esp stub_log on" -l oocd.log` and attach `oocd.log` file here?

EduardShaid, 2023-02-23:
The question is different: I just change the location of openocd-esp32 (Workspace OpyenOCD Path) from the old, working path

C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin

on a new path

C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230221\openocd-esp32\bin

and openocd-esp32 stops working.

erhankur, 2023-02-23:
But the new OpenOCD is running as I can see from the log you shared. So to understand what is the issue with the new version, I need some answers. Our internal tests have passed with the idf 4.4, 5.0, 5.1 and we believe we didn't break anything. But can be some cases that we missed.  e.g. Looking at the release notes, directly related to S3, we have fixed the flasher stub PSRAM issue. Now I would like to be sure if this fix is related to your case or not.

gerekon, 2023-02-28:
@EduardShaid Unfortunately we are still unable to reproduce your problem. Could you give us some hints?

edbek, 2023-02-28:
It is rather strange that this simple problem does not reproduce for you.
But if this is the one, then this topic can be closed. 
Thank you for your time.

igrr, 2023-02-28:
I just wanted leave a comment since I ran into what looks like the same issue, locally. I was also overriding the workspace OpenOCD path to the new version.

However I found that the environment variable OPENOCD_SCRIPTS (which can be set via workspace preferences > C++ > Build > Environment) was still pointing to the old copy of OpenOCD, and OpenOCD was loading some of the .cfg files from that copy.

After removing OPENOCD_SCRIPTS variable from C++ > Build > Environment, i was able to start a debugging session with the latest OpenOCD release.

EduardShaid, 2023-03-01:
Thanks a lot for the advice!
Yes, everything is working now.

erhankur, 2023-03-02:
@EduardShaid thanks for your support. Closing this issue.

closed 2023-03-02 by erhankur