espressif/openocd-esp32 #278: libusb_get_string_descriptor_ascii() failed with -9 error for JTAG Debugging ESP32S3

miczag80, 2023-03-21:
### Development Kit

ESP32-S3-DevKitC-1

### Module or chip used

ESP32-S3-WROOM1

### Debug Adapter

USB Serial/JTAG

### OpenOCD version

v0.11.0-esp32-20221026

### Operating System

Windows

### Using an IDE ?

Visual Studios Code

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

5.0.1

### Problem Description

1. I started by opening up visual studio code, and installing the Espressif IDF v1.6.0 extension, with my ESP32S3 Device plugged in to my machine

2. I opened windows PowerShell and pasted the following line to install required drivers (downloaded successfully):

Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe; .\idf-env.exe driver install --espressif

3. I opened the command palette (View > Command Palette) and typed in the command 'ESP-IDF: Show Examples Projects,' then selected the current framework, which opened the example projects 

4. I clicked 'blink' under the get-started examples and then created a new project using example blink

5. From here, without editing any of the code, in the lower left of the screen in the bar along the bottom I select the correct COM port, the device, and the debug type, which for me in this case was COM9, esp32s3, and JTAG. 

6. I built the code, and after a successful build, I clicked the 'ESP-IDF Flash Device' button.

7. A prompt that indicates OpenOCD is not running pops up, I click 'open'

8. In the output, I get the error show in the debug logs below. 

### Debug Logs

```shell
[OpenOCD]
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
[OpenOCD]
debug_level: 2

[OpenOCD]
Info : only one t[OpenOCD]
ransport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
[OpenOCD]
Warn : Transport "jtag" w[OpenOCD]
as already selected
[OpenOCD]
Info : Listening on port 6666 for tcl conne[OpenOCD]
ctions
Info : Listening on port 4444 for telnet connections
[OpenOCD]
❌ Error: libusb_get_string_de[OpenOCD]
scriptor_ascii() failed with -9

[OpenOCD]
[Stopped] : OpenOCD Server
```

### Expected behavior

Successful Debugging

### Screenshots

[image]

erhankur, 2023-05-04:
@miczag80 Can you update your OpenOCD with the latest release? We have upgraded `libusb` version which might be helpful  to fix that error.

erhankur, 2023-06-12:
@miczag80 Did you fix the issue? Can we close this?

rapeauxa, 2023-08-09:
Can we reopen this? I have a similar issue with the same error message when attempting to use OpenOCD through esp32s3 USB-JTAG interface. The interface registers correctly for Windows (correct pid-vid). I have updated OpenOCD manually using the latest version and this did not fix the issue.

erhankur, 2023-08-09:
@rapeauxa Can you please share the OpenOCD logs with -d3 option. First of all I would like to be sure about version and the other settings. As an additional , which windows version do you have?

rapeauxa, 2023-08-09:
Many thanks for the prompt answer. Using Windows 10 Pro. Will provide logs tomorrow, first thing (I don't have development system with me today, apologies). openOCD version: v0.12.0-esp32-20230419 (2023-04-18-22:02)

rapeauxa, 2023-08-10:
[openocd_log_esp32s3_usbjtag.txt](https://github.com/espressif/openocd-esp32/files/12310953/openocd_log_esp32s3_usbjtag.txt)
Here is the debug level 3 log for running openocd

MrPec, 2023-08-10:
Same problem here. 

Custom board, ESP32-C3-MINI-1
GPIO9 is high measured as high
Nothing else connected to GPIO18 / GPIO19.

Version info:
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Device appears as 
1. "Serielles USB-Gerät (COM5)", 10.0.19041.2130
2. "USB JTAG/serial debug unit",libwdi, 6.1.7600.16386

Things worth to mention:
1. I can't open the serial port COM5 with Putty or Arduino IDE
2. I can read out board info in Arduino IDE (with Serial number etc - where does it come from?)
3. If I pull GPIO9 low on startup, it works. After flashing any firmware it also works without pulling GPIO9 low (?)

Full log with d3:
```
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 2 options.c:52 configuration_output_handler(): debug_level: 3User : 4 5 options.c:52 configuration_output_handler():
Debug: 5 7 options.c:233 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 10 options.c:234 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 14 options.c:235 add_default_dirs(): exepath=C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin
Debug: 8 19 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 20 configuration.c:33 add_script_search_dir(): adding C:/Users/unknown-user/AppData/Roaming/OpenOCD
Debug: 10 24 configuration.c:33 add_script_search_dir(): adding C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/site
Debug: 11 31 configuration.c:33 add_script_search_dir(): adding C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts
Debug: 12 36 command.c:152 script_debug(): command - ocd_find board/esp32c3-builtin.cfg
Debug: 13 38 configuration.c:88 find_file(): found C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts/board/esp32c3-builtin.cfg
Debug: 14 47 command.c:152 script_debug(): command - ocd_find interface/esp_usb_jtag.cfg
Debug: 15 50 configuration.c:88 find_file(): found C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts/interface/esp_usb_jtag.cfg
Debug: 16 57 command.c:152 script_debug(): command - adapter driver esp_usb_jtag
Info : 17 60 transport.c:107 allow_transports(): only one transport option; autoselecting 'jtag'
Debug: 18 63 command.c:152 script_debug(): command - espusbjtag vid_pid 0x303a 0x1001
Info : 19 66 esp_usb_jtag.c:894 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Debug: 20 69 command.c:152 script_debug(): command - espusbjtag caps_descriptor 0x2000
Info : 21 72 esp_usb_jtag.c:905 esp_usb_jtag_caps_descriptor(): esp_usb_jtag: capabilities descriptor set to 0x2000
Debug: 22 76 command.c:152 script_debug(): command - adapter speed 40000
Debug: 23 78 adapter.c:253 adapter_config_khz(): handle adapter khz
Debug: 24 81 adapter.c:217 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 25 84 adapter.c:217 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 26 87 command.c:152 script_debug(): command - ocd_find target/esp32c3.cfg
Debug: 27 89 configuration.c:88 find_file(): found C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts/target/esp32c3.cfg
Debug: 28 97 command.c:152 script_debug(): command - transport select jtag
Warn : 29 99 transport.c:280 handle_transport_select(): Transport "jtag" was already selected
Debug: 30 102 command.c:152 script_debug(): command - ocd_find bitsbytes.tcl
Debug: 31 104 configuration.c:88 find_file(): found C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts/bitsbytes.tcl
Debug: 32 112 command.c:152 script_debug(): command - ocd_find memory.tcl
Debug: 33 115 configuration.c:88 find_file(): found C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts/memory.tcl
Debug: 34 122 command.c:152 script_debug(): command - ocd_find mmr_helpers.tcl
Debug: 35 125 configuration.c:88 find_file(): found C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts/mmr_helpers.tcl
Debug: 36 132 command.c:152 script_debug(): command - ocd_find target/esp_common.cfg
Debug: 37 136 configuration.c:88 find_file(): found C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg
Debug: 38 143 command.c:152 script_debug(): command - ocd_find target/esp_version.cfg
Debug: 39 146 configuration.c:88 find_file(): found C:/Users/unknown-user/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts/target/esp_version.cfg
Debug: 40 154 command.c:152 script_debug(): command - version
Debug: 41 156 command.c:152 script_debug(): command - add_help_text program_esp write an image to flash, address is only required for binary images. verify, reset, exit, compress, restore_clock and encrypt are optional
[... cut ...]
Debug: 108 454 command.c:152 script_debug(): command - esp32c3 configure -event gdb-flash-write-end reset halt
Debug: 109 458 command.c:152 script_debug(): command - esp32c3 cget -event gdb-attach
Debug: 110 461 target.c:1657 handle_target_init_command(): Initializing targets...
Debug: 111 465 riscv.c:444 riscv_init_target(): riscv_init_target()
Debug: 112 467 semihosting_common.c:107 semihosting_common_init():
Error: 113 485 libusb_helper.c:373 jtag_libusb_get_serial(): libusb_get_string_descriptor_ascii() failed with -9
```

MrPec, 2023-08-10:
I added the following line to the .cfg File:
adapter serial 40:4C:CA:C9:20:8C

I got that number from here:
[image]

Unfortunately, it can't find the adapter:
```
...
Debug: 111 503 target.c:1657 handle_target_init_command(): Initializing targets...
Debug: 112 506 riscv.c:444 riscv_init_target(): riscv_init_target()
Debug: 113 509 semihosting_common.c:107 semihosting_common_init():
Error: 114 526 libusb_helper.c:104 string_descriptor_equal(): libusb_get_string_descriptor_ascii() failed with 'LIBUSB_ERROR_PIPE'
Info : 115 531 libusb_helper.c:201 jtag_libusb_open(): No device matches the serial string
Error: 116 537 esp_usb_jtag.c:646 esp_usb_jtag_init(): esp_usb_jtag: could not find or open device!
```

rapeauxa, 2023-08-10:
My device does not seem to enumerate with a MAC address, I cannot find it in driver details unlike @MrPec 
I have 2 devices listed as 'USB JTAG/Serial debug unit' in device manager with VID 303A and PID 1001, the interface numbers seem to be 00 and 02.

MrPec, 2023-08-10:
@rapeauxa but shouldn't the first one (00) iterate as a serial port, like COMn?
Thats how it is for me, however I can't even open that serial port (it says "A fatal error occurred: Could not open COM5, the port doesn't exist") even it is here:

rapeauxa, 2023-08-10:
@MrPec my device does not enumerate as a COM Port, I don't have any listed in device manager. For what it's worth I followed instructions on setting up flashing and debugging using openocd using https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html . This may have been a mistake.

rapeauxa, 2023-08-17:
Hello, an update on my side - for what it's worth we've been able to confirm part of the issue was due to connecting from the USB-JTAG and USB_UART chip on the same USB bus, which must have caused conflicts between both interfaces. We have reverted this to only using USB UART for programming. Of course this prevents usage of the debugger, but we will make do without. @MrPec please check if your custom board also has this particular connection configuration.

MrPec, 2023-08-17:
Hey @rapeauxa , thanks for the advice. Not sure I understand. Our custom board basically only routes IO18 and IO19 directly to an USB connector. There is no other uart bridge chip involved - could you please explain what exactly you did to fix it? Or do you mean you simply disabled one of the two interfaces in Windows that appeared when you connect to the ESP via IO18/19?

MrPec, 2023-08-17:
Yes, it does now. By the way, we use a ESP32-C3-MINI-1-N4 module, not a S3. Only workaround so far is to manually short IO9 with GND during Reset - then it works just fine, also for subsequent programming. Very weird, as if the factory image has a problem....

erhankur, 2023-08-17:
@MrPec Are you suggesting that connecting IO9 to the ground resolves the issue? Have you tried this on a different operating system as well? I intend to update the documentation and would appreciate knowing if this problem is exclusive to Windows.  Additionally, We've observed that having IO8 in a floating state could potentially lead to another issue during the flash probe. But this is not Windows specific.

closed 2023-11-07 by erhankur

RamyaDidde, 2023-11-28:
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Error: libusb_get_string_descriptor_ascii() failed with -9
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1
Facing this issue, can anyone help me with this. Using ESp32 S3 C1 devkit using USB port for uploading and debugging purpose.. while trying to upload the code facing this issue.

KonssnoK, 2023-12-22:
> Hello, an update on my side - for what it's worth we've been able to confirm part of the issue was due to connecting from the USB-JTAG and USB_UART chip on the same USB bus, which must have caused [...]

I also confirm that the same -9 issue was caused by attaching jtag and terminal to the same laptop.
Once i moved the esp-prog to another computer openocd can connect easily

erikthegamer1242, 2024-02-04:
I can also say this fixed it for me, just having the JTAG connected, with UART disconnected