espressif/openocd-esp32 #279: CMSIS-DAP doesn't have hidapi support built in release

jschoch, 2023-03-23:
### Development Kit

esp32-devkit c

### Module or chip used

wroom32

### Debug Adapter

CMSIS-DAP

### OpenOCD version

openocd-esp32-win32-0.12.0-esp32-20230313

### Operating System

windows

### Using an IDE ?

CLI

### OpenOCD command line

.\bin\openocd.exe -f .\share\openocd\scripts\interface\cmsis-dap2.cfg --debug

### JTAG Clock Speed

default

### ESP-IDF version

N/A

### Problem Description

.\bin\openocd.exe -f .\share\openocd\scripts\interface\cmsis-dap2.cfg --debug

PS C:\Users\jesse\Downloads\openocd-esp32-win32-0.12.0-esp32-20230313\openocd-esp32> .\bin\openocd.exe -f .\share\openocd\scripts\interface\cmsis-dap2.cfg --debug
Open On-Chip Debugger v0.12.0-esp32-20230313 (2023-03-13-09:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 2 options.c:52 configuration_output_handler(): debug_level: 3
User : 4 4 options.c:52 configuration_output_handler():
Debug: 5 5 options.c:233 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 8 options.c:234 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 11 options.c:235 add_default_dirs(): exepath=C:/Users/jesse/Downloads/openocd-esp32-win32-0.12.0-esp32-20230313/openocd-esp32/bin
Debug: 8 16 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 18 configuration.c:33 add_script_search_dir(): adding C:/Users/jesse/AppData/Roaming/OpenOCD
Debug: 10 20 configuration.c:33 add_script_search_dir(): adding C:\Users\jesse/.config/openocd
Debug: 11 22 configuration.c:33 add_script_search_dir(): adding C:\Users\jesse/.openocd
Debug: 12 24 configuration.c:33 add_script_search_dir(): adding C:/Users/jesse/Downloads/openocd-esp32-win32-0.12.0-esp32-20230313/openocd-esp32/bin/../share/openocd/site
Debug: 13 29 configuration.c:33 add_script_search_dir(): adding C:/Users/jesse/Downloads/openocd-esp32-win32-0.12.0-esp32-20230313/openocd-esp32/bin/../share/openocd/scripts
Debug: 14 33 command.c:155 script_debug(): command - ocd_find .\share\openocd\scripts\interface\cmsis-dap2.cfg
Debug: 15 36 configuration.c:88 find_file(): found .\share\openocd\scripts\interface\cmsis-dap2.cfg
Debug: 16 39 command.c:155 script_debug(): command - adapter driver cmsis-dap
Debug: 17 41 command.c:155 script_debug(): command - adapter speed 1000
Debug: 18 43 adapter.c:253 adapter_config_khz(): handle adapter khz
Debug: 19 45 adapter.c:217 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 20 47 adapter.c:217 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 21 49 command.c:155 script_debug(): command - cmsis_dap_vid_pid 0x0483 0x572a 0x046d 0xc52b 0x046d 0xc08b
Debug: 22 52 command.c:155 script_debug(): command - cmsis_dap_backend hid
Error: 23 53 cmsis_dap.c:2149 cmsis_dap_handle_backend_command(): invalid backend argument to cmsis_dap_backend <backend>
Debug: 24 57 command.c:155 script_debug(): command - transport select jtag
User : 25 60 options.c:52 configuration_output_handler(): jtagUser : 26 62 options.c:52 configuration_output_handler(): 
Info : 27 64 server.c:298 add_service(): Listening on port 6666 for tcl connections
Info : 28 66 server.c:298 add_service(): Listening on port 4444 for telnet connections
Debug: 29 69 command.c:155 script_debug(): command - init
Debug: 30 70 command.c:155 script_debug(): command - target init
Debug: 31 72 command.c:155 script_debug(): command - target names
Debug: 32 74 target.c:1657 handle_target_init_command(): Initializing targets...
Debug: 33 104 cmsis_dap_usb_bulk.c:171 cmsis_dap_usb_open(): enumerating interfaces of 0x046d:0xc08b
Debug: 34 111 cmsis_dap_usb_bulk.c:171 cmsis_dap_usb_open(): enumerating interfaces of 0x046d:0xc52b
Debug: 35 118 cmsis_dap_usb_bulk.c:151 cmsis_dap_usb_open(): found product string of 0x0483:0x572a 'CMSIS-DAP'
Debug: 36 120 cmsis_dap_usb_bulk.c:171 cmsis_dap_usb_open(): enumerating interfaces of 0x0483:0x572a
Debug: 37 125 cmsis_dap_usb_bulk.c:218 cmsis_dap_usb_open(): found interface 0 string 'CMSIS-DAP CDC'
Debug: 38 127 cmsis_dap_usb_bulk.c:231 cmsis_dap_usb_open(): skipping interface 0, has only 1 endpoints
Debug: 39 131 cmsis_dap_usb_bulk.c:218 cmsis_dap_usb_open(): found interface 1 string 'CMSIS-DAP DCI'
Debug: 40 133 cmsis_dap_usb_bulk.c:281 cmsis_dap_usb_open(): skipping interface 1, class 10 subclass 0 protocol 0
Debug: 41 137 cmsis_dap_usb_bulk.c:218 cmsis_dap_usb_open(): found interface 2 string 'CMSIS-DAP'
Debug: 42 139 cmsis_dap_usb_bulk.c:238 cmsis_dap_usb_open(): skipping interface 2, endpoint[0] is not bulk out
Error: 43 143 cmsis_dap.c:281 cmsis_dap_open(): unable to find a matching CMSIS-DAP device
Debug: 44 145 command.c:545 run_command(): Command 'init' failed with error code -4
User : 45 148 command.c:608 command_run_line():

### Expected behavior

does not make error like so:

Error: 23 53 cmsis_dap.c:2149 cmsis_dap_handle_backend_command(): invalid backend argument to cmsis_dap_backend <backend>

paulfertser, 2023-03-24:
Please consider using https://github.com/openocd-org/openocd/blob/master/.github/workflows/snapshot.yml for your windows releases as these commands are actually supported and tested regularly, and if there's any problem with them let's work together on fixing.

erhankur, 2023-03-27:
@jschoch can you please give it a try with the attached windows package? I have added HIDAPI support. I would like to know if there is something missing.

[openocd-esp32-win32-0.12.0-esp32-20230313-3-gec9619f02.zip](https://github.com/espressif/openocd-esp32/files/11076069/openocd-esp32-win32-0.12.0-esp32-20230313-3-gec9619f02.zip)

jschoch, 2023-03-27:
I have tested with radio operator's STM32F103C8T6_CMSIS-DAP_SWO, MCU-Link, and DAP 103 ( i think https://github.com/devanlai/dap42).

DAP 103 and STM32F103C8T6_CMSIS-DAP_SWO work with HID and auto modes.

MCU-Link doesn't work with hid mode but i'm not sure if it is supported or not.

erhankur, 2023-03-30:
No idea what is wrong with MCU-Link but it seems binary is fine in terms of Hidapi support.  8f172239578b124cdebeaae583065287f1ddff93 is the commit you are interested in. In the next OpenOCD release, we will have hidapi support by default. 

@paulfertser The GH action you provided has an issue now.  You probably saw my patches regarding it. 

https://review.openocd.org/c/openocd/+/7551
https://review.openocd.org/c/openocd/+/7552

I think I can close this issue for now.

closed 2023-03-30 by erhankur