espressif/openocd-esp32 #280: Why is Espressif ESP32 Dev Module deprecated for OpenOCD ?

rob040, 2023-03-25:
### OpenOCD version

Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)

### Operating System

Windows 10

### Using an IDE ?

Platform IO

### OpenOCD command line

unknown

### ESP-IDF version

v4.4.1

### Problem Description

When using a board similar to **Espressif ESP32 Dev Module** with ESP32-WROOM (or ESP-WROOM32) module,  and starting debug, the following message appears:

### **WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.**

----
Configuration:

platformio.ini:
```ini
[env:esp32dev_dbg]
platform = espressif32
board = esp32dev
framework = arduino
```
and starting debugging, the following ominous message appears:

```
WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
```
This change to boards/esp-wroom-32.cfg which introduced this warning happened in this repo on commit b9ac088 by [gerekon](https://github.com/espressif/openocd-esp32/commits?author=gerekon) committed Sep 26, 2019, so quite some time ago.

I now worry that OCD support will be removed soon.

What is the reasoning behind the deprecation of this chip, being still popular, although its the first of the ESP 32 series.

### Expected behavior

Continue support for ESP32 WROOM series

igrr, 2023-03-25:
This `boards/esp-wroom-32.cfg` was deprecated because "wroom-32" is not a "board", but a module. In case of the ESP32, there is no configuration in OpenOCD which is specific to a particular module. The only meaninful configuration can be done for a chip or a development board. Therefore instead of the deprecated config file we have introduced config files for:
- the ESP32 chip (`target/esp32.cfg`)
- various development boards built around ESP32 (e.g. `target/esp32-wrover-kit-3.3v.cfg`).

The list of the available config files for ESP32 series of chips and boards can be found here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#configuration-of-openocd-for-specific-target.

In for your use case you can pick `target/esp32.cfg` and any other config file which describes the debug adapter.

rob040, 2023-03-26:
OK. Thanks for your answer. 
Indeed, some boards could mount different ESP32 modules, such as Espressif does with their [ESP32-DevKitC V4 board](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html#overview).
However, There are so many boards that use the ESP32-WROOM module, and to distinguish it clearly from the WROVER series (and S or C series), I would advise keeping the WROOM config file, albeit its unlucky naming (ESP-WROOM-32, i.s.o. ESP32-WROOM-32) where Espressif started with.

Do you know that PlatformIO lists 63(!) boards that reference this OpenOCD debug configuration file!
It would result in quite some complaints if the config file would indeed be removed.

igrr, 2023-03-26:
> There are so many boards that use the ESP32-WROOM module, and to distinguish it clearly from the WROVER series

From OpenOCD perspective, there is no need to. These two modules only differ in that WROVER has PSRAM, which OpenOCD can detect at run time. So `target/esp32.cfg` should work fine for both ESP32 WROOM and WROVER modules.

> and S or C series

These already have their own config files, such as target/esp32c3.cfg, target/esp32s3.cfg and so on.

> It would result in quite some complaints if the config file would indeed be removed.

PlatformIO does not, as far as i know, automatically upgrade OpenOCD to a new version. I suppose that whoever performs this upgrade will notice the change and also replace the names of config files in board definitions.

This config file has been deprecated more than 3 years ago, which seems to be enough time to update PlatformIO board definitions for the affected boards.

(That all being said, i don't think we have a concrete plan for removing this config file. Thank you very much for bringing this issue up, we will keep it in mind and will try to not cause any unnecessary breakage!)

closed 2023-12-29 by erhankur