espressif/openocd-esp32 #281: ESP32S3 JTAG USB/Serial does not print output using esp-rs/esp-println

jayv, 2023-04-14:
### Development Kit

Heltec WIFI LoRa 32 V3

### Module or chip used

ESP32S3

### Debug Adapter

custom JTAG USB Serial cable

### OpenOCD version

v0.12.0-esp32-20230313

### Operating System

Ubuntu 22.10 / Linux 6.2

### Using an IDE ?

Clion + CLI

### OpenOCD command line

openocd -c 'set ESP_RTOS none' -f board/esp32s3-builtin.cfg -c "init; reset halt"

### JTAG Clock Speed

default

### ESP-IDF version

v5.1-dev-4557-gf404fe96b1

### Problem Description

I'm new to ESP32S3. I made a JTAG USB/Serial cable and have used it to connect both with `espflash` as well as via `openocd` (using the [clion openocd plugin](https://plugins.jetbrains.com/plugin/18760-openocd--esp32-support-for-embedded-development)) to my device. 

Using rust bare metal ([Rust ESP book](https://esp-rs.github.io/book/introduction.html)) when I try to emit `println` logs via [esp-println](https://github.com/esp-rs/esp-println) crate it works fine using [esp-rs/espflash](https://github.com/esp-rs/espflash) to compile/flash/monitor the device, I get to see the output in my console. 

However when I tried running the same progran using the debugger in clion via openocd, it doesn't emit anything to my console. I noticed that the `USB_DEVICE/JFIFO_ST/OUT_FIFO_FULL` register bit is set HI upon start of the program (at entrypoint)... 

Is this behavior not expected/supported or am I missing something?

### Debug Logs

```shell
Pasted here (1MB) https://controlc.com/96d993c9
```

### Expected behavior

logs emitted print in the console when emitted via jtag serial as they do with `espflash`

igrr, 2023-04-14:
@jayv OpenOCD only communicates with the "JTAG" part of the USB_SERIAL_JTAG peripheral. It doesn't communicate with the serial port part. If you are using CLion, you can install the serial port monitor plugin and open the serial port separately. It will work independently from OpenOCD.

jayv, 2023-04-14:
Aha that's a very good reason :) That worked, thanks a lot this wasn't obvious.

[image]