espressif/openocd-esp32 #282: Cannot add breakpoint

MoreTore, 2023-04-21:
### Development Kit

ESP32-S2-DevKitM-1 1.0

### Module or chip used

ESP32-S2-mini-2

### Debug Adapter

ESP-Prog JTAG

### OpenOCD version

v0.11.0-esp32-20221026 (2022-10-26-14:48)

### Operating System

Windows 11

### Using an IDE ?

VSCode

### OpenOCD command line

Press F5 in vscode hello world sample project

### JTAG Clock Speed

20000

### ESP-IDF version

5.0

### Problem Description

I start the hello world program by pressing F5. It will break on entry and then I press continue. After a some time I add a breakpoint and I get the the error:
```

[Debug Adapter]
2023-04-21 03:05:50,038 - Debug Adapter (main) - DEBUG - Poll target restart
2023-04-21 03:05:50,038 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Poll target restart\n", "source": {}}, "seq": 178}

[Debug Adapter]
2023-04-21 03:05:50,045 - Gdb - DEBUG - MI<-:
[{'message': 'error',
  'payload': {'msg': 'Cannot execute this command while the target is '
                     'running.nUse the "interrupt" command to stop the '
                     'targetnand then try again.'},
  'stream': 'stdout',
  'token': None,
  'type': 'result'}]
2023-04-21 03:05:50,045 - Gdb - DEBUG - RESULT: error {'msg': 'Cannot execute this command while the target is running.nUse the '
        '"interrupt" command to stop the targetnand then try again.'}
[Debug Adapter]
2023-04-21 03:05:50,045 - Debug Adapter (main) - ERROR - unhashable type: 'dict'
Traceback (most recent call last):
  File "c:\Users\faste\.vscode\extensions\espressif.esp-idf-extension-1.6.1\esp_debug_adapter\debug_adapter\command_processor.py", line 700, in on_setBreakpoints_request
    try_set_once(get_good_path(source.path), src_line, condition)
  File "c:\Users\faste\.vscode\extensions\espressif.esp-idf-extension-1.6.1\esp_debug_adapter\debug_adapter\command_processor.py", line 676, in try_set_once
    raise e
  File "c:\Users\faste\.vscode\extensions\espressif.esp-idf-extension-1.6.1\esp_debug_adapter\debug_adapter\command_processor.py", line 674, in try_set_once
    return self.da.source_break_add(source_path, line, condition=condition)
  File "c:\Users\faste\.vscode\extensions\espressif.esp-idf-extension-1.6.1\esp_debug_adapter\debug_adapter\debug_adapter.py", line 523, in source_break_add
    bp_num = self._gdb.add_bp("{}:{}".format(src, line), ignore_count=0, cond=condition)
  File "c:\Users\faste\.vscode\extensions\espressif.esp-idf-extension-1.6.1\esp_debug_adapter\debug_adapter\debug_backend\debug_backend\gdb.py", line 487, in add_bp
    raise DebuggerError('Failed to insert BP!')
debug_adapter.debug_backend.debug_backend.defs.DebuggerError: Failed to insert BP!

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\faste\.vscode\extensions\espressif.esp-idf-extension-1.6.1\esp_debug_adapter\debug_adapter\command_processor.py", line 52, in __call__
    on_request(protocol_message)
  File "c:\Users\faste\.vscode\extensions\espressif.esp-idf-extension-1.6.1\esp_debug_adapter\debug_adapter\command_processor.py", line 706, in on_setBreakpoints_request
    bp.update({{'message': e}})
TypeError: unhashable type: 'dict'
2023-04-21 03:05:50,045 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "unhashable type: 'dict'\n", "source": {}}, "seq": 179}

2023-04-21 03:05:50,046 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "unhashable type: 'dict'\n", "source": {}}, "seq": 180}

[Debug Adapter]
2023-04-21 03:05:50,148 - Debug Adapter (main) - DEBUG - Poll target. Wait state 1
2023-04-21 03:05:50,148 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Poll target. Wait state 1\n", "source": {}}, "seq": 181}

```

When I try to kill the program I get an error:

```
[Debug Adapter]
2023-04-21 03:08:24,828 - Debug Adapter(Command Processor) - DEBUG - Got json: {
    "arguments": {
        "restart": false,
        "terminateDebuggee": true
    },
    "command": "disconnect",
    "seq": 15,
    "type": "request"
}
2023-04-21 03:08:24,828 - Debug Adapter (main) - DEBUG - Adapter is Stopping...
2023-04-21 03:08:24,828 - Debug Adapter (main) - DEBUG - Stopping target poller
2023-04-21 03:08:24,828 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got json: {\n    \"arguments\": {\n        \"restart\": false,\n        \"terminateDebuggee\": true\n    },\n    \"command\": \"disconnect\",\n    \"seq\": 15,\n    \"type\": \"request\"\n}\n", "source": {}}, "seq": 686}

2023-04-21 03:08:24,828 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug Adapter stopped\n", "source": {}}, "seq": 687}

2023-04-21 03:08:24,828 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Adapter is Stopping...\n", "source": {}}, "seq": 688}

2023-04-21 03:08:24,828 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Stopping target poller\n", "source": {}}, "seq": 689}

[Debug Adapter]
2023-04-21 03:08:25,012 - Debug Adapter (main) - DEBUG - Closing the GDB process
2023-04-21 03:08:25,012 - Gdb - DEBUG - MI->: -gdb-exit
[OpenOCD]
Info : dropped 'gdb' connection
[Debug Adapter]
2023-04-21 03:08:25,013 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the GDB process\n", "source": {}}, "seq": 690}

2023-04-21 03:08:25,014 - Gdb - DEBUG - MI<-:
[{'message': 'exit',
  'payload': None,
  'stream': 'stdout',
  'token': None,
  'type': 'result'},
 {'message': None,
  'payload': 'Cannot execute this command while the target is running.\\n',
  'stream': 'stdout',
  'type': 'log'},
 {'message': None,
  'payload': 'Use the \\"interrupt\\" command to stop the target\\n',
  'stream': 'stdout',
  'type': 'log'},
 {'message': None,
  'payload': 'and then try again.\\n',
  'stream': 'stdout',
  'type': 'log'}]
2023-04-21 03:08:25,014 - Gdb - DEBUG - RESULT: exit None
[Debug Adapter]
Exception in thread Thread-318:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
Exception in thread Thread-319:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "threading.py", line 870, in run
  File "subprocess.py", line 1366, in _readerthread
  File "subprocess.py", line 1366, in _readerthread
OSError: [Errno 22] Invalid argument
[Debug Adapter]
2023-04-21 03:08:25,017 - Debug Adapter (main) - DEBUG - list index out of range
2023-04-21 03:08:25,017 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "list index out of range\n", "source": {}}, "seq": 691}

2023-04-21 03:08:25,017 - Debug Adapter (main) - DEBUG - Closing the Socket
2023-04-21 03:08:25,018 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the Socket\n", "source": {}}, "seq": 692}

2023-04-21 03:08:25,018 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
2023-04-21 03:08:25,018 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Stopping of the Writer thread\n", "source": {}}, "seq": 693}

2023-04-21 03:08:25,018 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
2023-04-21 03:08:25,018 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2023-04-21 03:08:25,018 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2023-04-21 03:08:25,018 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
[Debug Adapter]
OSError: [Errno 22] Invalid argument
[Stopped] : ESP-IDF Debug Adapter

```

settings.json 
```

{
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.adapterTargetName": "esp32s2",
  "idf.openOcdConfigs": [
    "interface/ftdi/esp32_devkitj_v1.cfg",
    "target/esp32s2.cfg"
  ],
  "files.associations": {
    "esp_flash.h": "c"
  },
  "idf.ninjaArgs": [
    "-j 20"
  ],
  "idf.portWin": "COM13",
  "idf.flashType": "JTAG", // 20 threads for ninja build
  "idf.cmakeCompilerArgs": [
    "-DCMAKE_BUILD_TYPE=Debug",
    "Ninja",
    "-DPYTHON_DEPS_CHECKED=1",
    "-DESP_PLATFORM=1"
  ]

}

```

### Expected behavior

I expect the esp to halt on my breakpoint. 

### Screenshots

[image]
[image]
[image]

MoreTore, 2023-04-21:
[debug.log](https://github.com/espressif/openocd-esp32/files/11293775/debug.log)

MoreTore, 2023-04-21:
[hello_world.zip](https://github.com/espressif/openocd-esp32/files/11293783/hello_world.zip)

AndrewCapon, 2023-04-29:
I'm seeing the same here, with a S3:

```
023-04-29 16:22:43,763 - Debug Adapter (main) - ERROR - unhashable type: 'dict'
Traceback (most recent call last):
  File "/Volumes/T7/.vscode/extensions/espressif.esp-idf-extension-1.6.1/esp_debug_adapter/debug_adapter/command_processor.py", line 700, in on_setBreakpoints_request
    try_set_once(get_good_path(source.path), src_line, condition)
  File "/Volumes/T7/.vscode/extensions/espressif.esp-idf-extension-1.6.1/esp_debug_adapter/debug_adapter/command_processor.py", line 676, in try_set_once
    raise e
  File "/Volumes/T7/.vscode/extensions/espressif.esp-idf-extension-1.6.1/esp_debug_adapter/debug_adapter/command_processor.py", line 674, in try_set_once
    return self.da.source_break_add(source_path, line, condition=condition)
  File "/Volumes/T7/.vscode/extensions/espressif.esp-idf-extension-1.6.1/esp_debug_adapter/debug_adapter/debug_adapter.py", line 523, in source_break_add
    bp_num = self._gdb.add_bp("{}:{}".format(src, line), ignore_count=0, cond=condition)
  File "/Volumes/T7/.vscode/extensions/espressif.esp-idf-extension-1.6.1/esp_debug_adapter/debug_adapter/debug_backend/debug_backend/gdb.py", line 487, in add_bp
    raise DebuggerError('Failed to insert BP!')
debug_adapter.debug_backend.debug_backend.defs.DebuggerError: Failed to insert BP!
```

erhankur, 2023-05-04:
@brianignacio5 can you have a look?

brianignacio5, 2023-05-05:
@erhankur Sure thing !

Edit:
It is not broken the debug adapter. The thing is that the user is trying to add a breakpoint while the program is running and not interrupting the execution. In the terminal you would usually run the interrupt command but the debug adapter when you set a breakpoint doesn't interrupt the program execution. Need to add the workflow of interrupting the program if executing before adding breakpoint.

This is a duplicate of https://github.com/espressif/vscode-esp-idf-extension/issues/958 and the issue is not related to openocd-esp32. We can close it here @erhankur

erhankur, 2023-05-05:
Thanks @brianignacio5

closed 2023-05-05 by erhankur