espressif/openocd-esp32 #284: [esp32s3] Change cfg setting to act by default as appimage_offset

KonssnoK, 2023-04-24:
Hello guys,
in our project we use a partition table which allows the bootloader to use its maximum size (0x10000) instead of the default one (0x8000).

When running openocd to debug, we simply add
`esp32s3 appimage_offset 0x100000`
as command and we are fine.

Problem is when we want to use an external tool which interfaces with app_trace. I can't find a why to put the appimage_offset in the correct position without the core resetting at some point (and losing the setting)

Where can i change such value in the openocd scripts 
xtensa-core-esp32s3.cfg
xtensa-core-esp32.cfg

so that i don't have to issue the appimage_offset command anymore?

Thanks

erhankur, 2023-04-24:
Do I understand correctly, running OpenOCD with the command line argument doesn't work for you?

`-c "init; halt; esp appimage_offset 0x100000; resume`

`-c "init; halt; esp appimage_offset 0x100000; esp apptrace start con: 0; resume"`

You need to send `halt` command before `appimage_offset`.  You don't need to reset the device. Then send `resume` to continue running.

KonssnoK, 2023-04-24:
@erhankur no the problem is not using appimage_offset,
the question is : how to have 0x10000 as default value in the scripts :)

erhankur, 2023-04-26:
You can add it to `-event gdb-attach` and try. It is inside esp32.cfg and esp32s3.cfg
```
$_TARGETNAME_0 configure -event gdb-attach {
	$_TARGETNAME_0 xtensa smpbreak BreakIn BreakOut
	# necessary to auto-probe flash bank when GDB is connected
	halt 1000
	
        esp appimage_offset 0x100000
	# main flash bank
	flash probe 0
	# IROM
	flash probe 1
	# DROM
	flash probe 2
}
```

KonssnoK, 2023-04-28:
thanks! i'll give it a try ! :)

erhankur, 2023-06-12:
@KonssnoK Did you try it? Can we close the issue?

KonssnoK, 2023-06-14:
sorry, we had to shift priorities.. let's close, in case i'll open another ticket if necessary... thanks again