espressif/openocd-esp32 #285: Sporadic failure to read memory when OpenOCD launched in PIPE mode

valeros, 2023-04-25:
### OpenOCD version

v0.12.0-esp32-20230419

### Operating System

Windows

### Using an IDE ?

No

### OpenOCD command line

-f interface/esp_usb_jtag.cfg -f target/esp32s3.cfg

### ESP-IDF version

5.0.1

### Problem Description

Hi there!

While trying to integrate the latest OpenOCD to PlatformIO I've bumped into a sporadic issue when OpenOCD fails with a `Error: memread32: read_memory: failed to read memory` error if launched in the PIPE mode via the `gdb_port pipe` command. 
Tested with an ESP32-S3-DevKitC and the `esp_usb_jtag` interface, but other ESP32 boards (with different debug probes) also fail occasionally with the same error. 

Even though it's hard to reproduce the issue reliably, here is a minimal project if needed:

[openocd-0.12-20230419.zip](https://github.com/espressif/openocd-esp32/files/11322995/openocd-0.12-20230419.zip)

Just update toolchain location in the `run_openocd.py` file and OpenOCD location in `.pioinit`. Then run `python run_openocd.py` several times.

Some observations I believe worth mentioning:

- The issue only occurs on Windows, Linux works just fine
- Tested OpenOCD v0.12 with other hardware (STMicro, Nordic) and it seems there are no issues with the PIPE mode
- Any previous release from this repository based on OpenOCD v0.11 also works correctly.

Maybe you have some ideas what's happening here? Any hints are very welcome. 

### Debug Logs

```shell
Sadly, I cannot attach detailed logs because once logging is enabled the issue become unreproducible. But here output from the CLI:

GNU gdb (crosstool-NG esp-2021r2-patch5) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=xtensa-esp32s3-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from firmware.elf...
Initializing remote target...
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz
Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz
Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (7C:DF:A1:E0:0B:AC)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x4201FC7A, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x4201FC7A, debug_reason=00000000
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Warn : [esp32s3.cpu0] target not halted
Error executing event gdb-attach on target esp32s3.cpu0:
C:/tool-openocd-esp32-012-20230419/share/openocd/scripts/memory.tcl:87: Error: memread32: read_memory: failed to read memory
in procedure 'esp32s3_memprot_is_enabled' called at file "C:/tool-openocd-esp32-012-20230419/share/openocd/scripts/target/esp32s3.cfg", line 155
in procedure 'get_mmr_bit' called at file "C:/tool-openocd-esp32-012-20230419/share/openocd/scripts/target/esp32s3.cfg", line 48
in procedure 'memread32' called at file "C:/tool-openocd-esp32-012-20230419/share/openocd/scripts/mmr_helpers.tcl", line 79
at file "C:/tool-openocd-esp32-012-20230419/share/openocd/scripts/memory.tcl", line 87
Warn : No symbols for FreeRTOS!
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Error: error during select: Unknown error
.pioinit:3: Error in sourced command file:
Remote communication error.  Target disconnected.: Success.
```

### Expected behavior

A successfully launched debug session.

erhankur, 2023-05-04:
@valeros Thanks for sharing the binaries. I will try to reproduce the issue. In the meantime, can you run openocd with -d3 option and attach the logs ?

valeros, 2023-05-08:
@erhankur That's actually a tricky part. Once I enable logging the issue becomes impossible to reproduce. I was hoping maybe you have superficial idea of possible causes based to the error message.

erhankur, 2023-05-09:
@valeros I tried several times without debug but couldn't reproduce the issue. I used latest master branch on macos.

Unfortunately, no clue why you see halt issue just before reading the memory.  Can you add reset halt to you .pioinit.
for example:
```
target extended-remote | "$OOCD_DIR/src/openocd" -s "$OOCD_DIR/tcl" -f "interface/esp_usb_jtag.cfg" -f "target/esp32s3.cfg" -c "init; reset halt; gdb_port pipe; tcl_port disabled; telnet_port disabled" -d2
```

I will let you know once I try on windows.

erhankur, 2023-12-28:
No progress on my side. Since can not be reproducible I do not plan to investigate more. Sorry for not coming up with a magical solution.

closed 2023-12-28 by erhankur