espressif/openocd-esp32 #286: Cannot debug ESP32 through esp-usb-bridge

AlbertoLopSie, 2023-04-26:
### Development Kit

ESP32-DevkitC

### Module or chip used

ESP32WROOM32

### Debug Adapter

esp-usb-bridge

### OpenOCD version

v0.12.0-esp32-20230313 (2023-03-13-09:08)

### Operating System

Mac OSX 10.15.7 (Catalina)

### Using an IDE ?

VSCode

### OpenOCD command line

openocd -d3 -f board/esp32-bridge.cfg

### JTAG Clock Speed

default

### ESP-IDF version

5.1

### Problem Description

ESP32-Devkit connected to [esp-usb-bridge](https://github.com/espressif/esp-usb-bridge) as the [schematic](https://github.com/espressif/esp-usb-bridge/blob/master/images/schematics.pdf) suggest.

Trying to debug on VSCode with PlatformIO IDE extension

PlatformIO version
   Core: 6.1.6
  Home 3.4.3

platformio.ini:
``` ini
upload_protocol = esp-bridge

; debug_tool = esp-bridge       ; replaced by custom tool to allow debug logging
debug_tool = custom

debug_speed = 1000
debug_server = 
  ${platformio.packages_dir}/tool-openocd-esp32/bin/openocd
  -d3
  -f
  board/esp32-bridge.cfg
```

Uploading is working fine. Debugging is not.

More info on this  [esp-usb-bridge issue](https://github.com/espressif/esp-usb-bridge/issues/10)

See [log on my gdrive](https://drive.google.com/file/d/16JIYCN2utahjGFnBNCdrpHx43CIEvMfG/view?usp=share_link)

I guess the interesting lines are the end, after `PlatformIO: Initializing remote target...`. It takes about 5-10 minutes to be produced (I guess some process is insisting in getting 4 bytes out of 2 and retrying).

I also have on another window the log generated by `esp-usb-bridge` if you think is useful (I don't see anything wrong there)

**Any clue about this??**

### Debug Logs

```shell
Too long to be accepted here. Get [it from my gdrive](https://drive.google.com/file/d/16JIYCN2utahjGFnBNCdrpHx43CIEvMfG/view?usp=share_link)
```

### Expected behavior

I should be able to start a debugging session

### Screenshots

N/A

AlbertoLopSie, 2023-04-27:
Following a suggestion from [dobairoland](https://github.com/dobairoland) on a [related issue](https://github.com/espressif/esp-usb-bridge/issues/10) on  [espressif/esp\-usb\-bridge](https://github.com/espressif/esp-usb-bridge) I increased the debug level of the openocd execution.  I coudn't be able to doit inside the PlatformIO extension I'm using, but managed to do it on a terminal shell:

``` sh 
$ ./openocd -d4 -l /Users/alberto/dev/ESP32/trh-esp32/openocd.log -f board/esp32-bridge.cfg
Open On-Chip Debugger v0.12.0-esp32-20230313 (2023-03-13-09:08)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 1 options.c:52 configuration_output_handler(): debug_level: 4
User : 4 1 options.c:52 configuration_output_handler(): 
Debug: 5 1 command.c:155 script_debug(): command - log_output /Users/alberto/dev/ESP32/trh-esp32/openocd.log
``` 

After a 5-10 minutes running, the log accumulated almost half a GB, so I had to cancel it.

Here is the [link to the log](https://drive.google.com/file/d/16JsMRr9UyWA07Mlanu4Xj4Reca0Rj_rD/view?usp=share_link) on my GDrive.

Up to the end of the logfile (before the terminating signal) there is a pattern that repeats over and over:

``` sh
Debug: 7115831 768263 bitq.c:247 bitq_execute_queue(): scan end in 12
Debug: 7115832 768263 bitq.c:248 bitq_execute_queue(): scan ir
Debug: 7115833 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115834 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115835 768263 bitq.c:106 bitq_state_move(): tap_set_state(IRSHIFT)
Debug: 7115836 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115837 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
----------------------- [ MANY LINES LIKE THIS] -----------------------
Debug: 7115846 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115847 768263 bitq.c:185 bitq_scan_field(): tap_set_state(IRPAUSE)
Debug: 7115848 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115849 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115850 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115851 768263 bitq.c:106 bitq_state_move(): tap_set_state(RUN/IDLE)
Debug: 7115852 768263 bitq.c:247 bitq_execute_queue(): scan end in 12
Debug: 7115853 768263 bitq.c:248 bitq_execute_queue(): scan dr
Debug: 7115854 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115855 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115856 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115857 768263 bitq.c:106 bitq_state_move(): tap_set_state(DRSHIFT)
Debug: 7115858 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115859 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
----------------------- [ MANY LINES LIKE THIS] -----------------------
Debug: 7115867 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115868 768263 bitq.c:187 bitq_scan_field(): tap_set_state(DRPAUSE)
Debug: 7115869 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115870 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115871 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115872 768263 bitq.c:106 bitq_state_move(): tap_set_state(RUN/IDLE)
Debug: 7115873 768263 bitq.c:247 bitq_execute_queue(): scan end in 12
Debug: 7115874 768263 bitq.c:248 bitq_execute_queue(): scan dr
Debug: 7115875 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115876 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115877 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115878 768263 bitq.c:106 bitq_state_move(): tap_set_state(DRSHIFT)
Debug: 7115879 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115880 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
----------------------- [ MANY LINES LIKE THIS] -----------------------
Debug: 7115912 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115913 768263 bitq.c:187 bitq_scan_field(): tap_set_state(DRPAUSE)
Debug: 7115914 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115915 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115916 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115917 768263 bitq.c:106 bitq_state_move(): tap_set_state(RUN/IDLE)
Debug: 7115918 768263 bitq.c:247 bitq_execute_queue(): scan end in 12
Debug: 7115919 768263 bitq.c:248 bitq_execute_queue(): scan ir
Debug: 7115920 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115921 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115922 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7115923 768263 bitq.c:51 bitq_in_proc(): bitq in EOF
[... 115 lines cut ...]
Debug: 7116123 768267 bitq.c:106 bitq_state_move(): tap_set_state(RUN/IDLE)
Debug: 7116124 768267 bitq.c:247 bitq_execute_queue(): scan end in 11
Debug: 7116125 768267 bitq.c:248 bitq_execute_queue(): scan ir
Debug: 7116126 768267 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7116127 768267 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7116128 768268 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7116129 768268 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7116130 768268 bitq.c:106 bitq_state_move(): tap_set_state(IRSHIFT)
Debug: 7116131 768268 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7116132 768268 bitq.c:51 bitq_in_proc(): bitq in EOF
Debug: 7116133 768268 bitq.c:185 bitq_scan_field(): tap_set_state(IRPAUSE)
Debug: 7116134 768268 esp_usb_jtag.c:564 esp_usb_jtag_flush(): esp_usb_jtag: Flush!
Debug: 7116135 768268 esp_usb_jtag.c:441 esp_usb_jtag_send_buf(): esp_usb_jtag: sent 23 bytes.
[... cut ...]
Debug: 7116141 768272 core.c:967 default_interface_jtag_execute_queue():   8b out: 0a
Debug: 7116142 768272 core.c:962 default_interface_jtag_execute_queue(): JTAG DR SCAN to RUN/IDLE
Debug: 7116143 768272 core.c:967 default_interface_jtag_execute_queue():   32b out: 00000000
Debug: 7116144 768272 core.c:972 default_interface_jtag_execute_queue():   32b  in: 00000000
Debug: 7116145 768272 core.c:962 default_interface_jtag_execute_queue(): JTAG IR SCAN to IRPAUSE
Debug: 7116146 768272 core.c:967 default_interface_jtag_execute_queue():   1b out: 00
```

AlbertoLopSie, 2023-04-29:
I searched across **this repo** for the `qSupported` error the debug console trows:

``` sh
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
.pioinit:11: Error in sourced command file:
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
``` 

And found:
 - An entry inside the [Troubleshooting FAQ](https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ#gdb-reports-unrecognized-item-timeout-in-qsupported-response)
- An [old PR](https://github.com/espressif/openocd-esp32/pull/134) saying something about remote bitbanging (no idea about)
 - [14 issues](https://github.com/search?q=repo%3Aespressif%2Fopenocd-esp32+qSupported&type=issues)
 - [2 code files](https://github.com/search?q=repo%3Aespressif%2Fopenocd-esp32+qSupported&type=code)

Searching `vMustReplyEmpty` gives similar results.

Does anything of this make any sound?

AlbertoLopSie, 2023-04-29:
SOLVED!!!!

YOU WON'T BELIEVE THE REASON!!

[ESP\-PROG \- openocd path contains "undefined" \- Advanced Solutions / Debugging \- PlatformIO Community](https://community.platformio.org/t/esp-prog-openocd-path-contains-undefined/23799)

closed 2023-05-02 by erhankur

ivankravets, 2023-05-02:
@AlbertoLopSie how did you solve this issue?