espressif/openocd-esp32 #287: External JTAG connection (i.e., Segger Jlink Ultra+)is not working

Vijay9441, 2023-05-05:
### Development Kit

ESP32-S3

### Module or chip used

ESP32-S3

### Debug Adapter

Segger Jlink Ultra +

### OpenOCD version

v0.11.0-esp32-20221026 (2022-10-26-14:48)

### Operating System

Window 10

### Using an IDE ?

No

### OpenOCD command line

openocd -f interface/jlink.cfg -f target/esp32s3.cfg

### JTAG Clock Speed

100Khz

### ESP-IDF version

esp-idf-v5.0.1

### Problem Description

I tried to connect the device with External JTAG with Segger J link.
I'm getting 
Error: [esp32s3.cpu1] Timed out waiting for CPU to be reset, target state=2.

I'm sure the my JTAG connections are correct and I made DIS_JTAG_JTAG bit and STRAO_JTAG_SEL bit ie.GPIO3 connected to Ground.
Please suggest for further .

### Expected behavior

JTAG program should work.

### Screenshots

[image]

erhankur, 2023-05-05:
@Vijay9441 can you set the clock speed and try?  e.g.
```
-f interface/jlink.cfg -c "adapter speed 5000" -f target/esp32s3.cfg
```

Vijay9441, 2023-05-08:
Hi,
I trying your suggested command, I'm getting this.
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf5.0_py3.8_env\Scripts
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.0.1
Checking Python compatibility
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: 1.22.0-61-gab8375a-5.2.0. To use it, run 'C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v5.0.1\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool esp32ulp-elf found in PATH: 2.28.51.20170517. To use it, run 'C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v5.0.1\tools\idf_tools.py export --prefer-system'

Name                           Value
----                           -----
OPENOCD_SCRIPTS                C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\s...
IDF_CCACHE_ENABLE              1
ESP_ROM_ELF_DIR                C:\Espressif\tools\esp-rom-elfs\20220823\
IDF_PYTHON_ENV_PATH            C:\Espressif\python_env\idf5.0_py3.8_env
ESP_IDF_VERSION                5.0
IDF_DEACTIVATE_FILE_PATH       C:\Users\vijaygu\AppData\Local\Temp\tmpov3dpf9sidf_17164

Added to PATH
-------------
C:\Espressif\frameworks\esp-idf-v5.0.1\components\esptool_py\esptool
C:\Espressif\frameworks\esp-idf-v5.0.1\components\app_update
C:\Espressif\frameworks\esp-idf-v5.0.1\components\espcoredump
C:\Espressif\frameworks\esp-idf-v5.0.1\components\partition_table
C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin
C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin
C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin
C:\Espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin
C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin
C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin
C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
C:\Espressif\tools\cmake\3.24.0\bin
C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin
C:\Espressif\tools\ninja\1.10.2\
C:\Espressif\tools\idf-exe\1.0.3\
C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64
C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
C:\Espressif\frameworks\esp-idf-v5.0.1\tools
Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.0.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.0.1\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of idf-driver, idf-python-wheels use command 'python.exe C:\Espressif\frameworks\esp-idf-v5.0.1\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\Espressif\frameworks\esp-idf-v5.0.1\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build

PS C:\Espressif\frameworks\esp-idf-v5.0.1> -f interface/jlink.cfg -c "adapter speed 5000" -f target/esp32s3.cfg
-f : The term '-f' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the
spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ -f interface/jlink.cfg -c "adapter speed 5000" -f target/esp32s3.cfg
+ ~~
    + CategoryInfo          : ObjectNotFound: (-f:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
[... 54 lines cut ...]
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link Ultra V5 compiled Mar 28 2023 17:00:28
Info : Hardware version: 5.00
Info : VTarget = 3.357 V
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40378B02, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32s3.cpu1] Target halted, PC=0x40378B02, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (1) - (Power on reset)

Please can you confirm weather it is correct??

erhankur, 2023-05-08:
Yes, the latest one looks correct. 

```
openocd -f interface/jlink.cfg -c "adapter speed 5000" -f target/esp32s3.cfg
```

Do you see any other issues in the next steps during debugging?

Vijay9441, 2023-05-08:
Ok thanks. After this When I try to type anything , it is not showing on the command line.
Please can you send the detailed flow of open ocd?? Please suggest.

erhankur, 2023-05-08:
OK, now you need to launch GDB for debugging. 

I suggest you to read https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html

And maybe for general information https://hackaday.com/2012/09/27/beginners-look-at-on-chip-debugging/

Also, you can look at official OpenOCD documentation. https://openocd.org/doc/html/index.html

Vijay9441, 2023-05-08:
Thanks.
We are using the External JTAG in openocd.
Is this works for ESP32S3, Please confirm.
[image]

erhankur, 2023-05-08:
According to the log you already get it worked. In the next steps, there should not be an issue with GDB, since OpenOCD will handle the target specific functionality.

closed 2023-05-10 by erhankur