espressif/openocd-esp32 #289: Flashing over jtag: timed out while waiting for target halted

UKenda, 2023-05-15:
### OpenOCD version

v0.11.0-esp32-20221026

### Operating System

Windows

### Using an IDE ?

vs-code

### OpenOCD command line

openocd -f board/esp-wroom-32-jlink.cfg

### ESP-IDF version

5.0.1

### Problem Description

### Description
Problem occurred when my specific firmware is flashed, then the same or other firmware flashing over JTAG fails. However, I can flash it over UART successfully.  However, if I flash some other firmware, then I can flash over JTAG until the specific firmware is flashed.
The firmware work correctly, and the JTAG worked on several projects without problems even on the same and different HW.
When I made this new firmware which include WIFI stack the JTAG flashing fails if I remove the WIFI stack from the project the JTAG starts working, similar is happening when including BT stack. It seems to me that it brakes when the size of the code exceeds some amount. 
Also, the debugger works, so I realy have the problem oly whit flashing. 

### The output of openocd:
```
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
C:\Users\User\.espressif501\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe
[OpenOCD]
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
[OpenOCD]
Info : accepting 'tcl' connection on tcp/6666
[OpenOCD]
Flashing c:/Users/User/esp/projects-5.0.1/FLARM_WIFI_Link/b[OpenOCD]
uild/bootloader/bootloader.bin at 0x1000
[OpenOCD]
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG [OpenOCD]
tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[OpenOCD]
Info : [esp32.cpu0] requesting target halt and executing[OpenOCD]
 a soft reset
[OpenOCD]
Info : [esp32.cpu0] Debug controller was reset.
[OpenOCD]
Info : [esp32.cpu0] Core was reset.
[OpenOCD]
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
[OpenOCD]
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
[OpenOCD]
Info : [esp32.cpu1] requesting target halt and executing a[OpenOCD]
 soft reset
[OpenOCD]
Info : [esp32.cpu0] Core was reset.
[OpenOCD]
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reas[OpenOCD]
on=00000000
[OpenOCD]
Info : [esp32.cpu1] Debug controller was reset.
[OpenOCD]
Info : [esp32.cpu1] Core was reset.
[OpenOCD]
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[OpenOCD]
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
[OpenOCD]
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
[OpenOCD]
** Programming Started **
[OpenOCD]
❌ Error: timed out while waiting for target halted
[OpenOCD]
Info : [esp32.cpu0] Target halted, PC=0x40088825, debug_reason=00[OpenOCD]
000000
Err[OpenOCD]
❌ or: xtensa_wait_algorithm: not halted 0, pc 0x40088825, ps 0x60023
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run flasher stub (-302)!
[OpenOCD]
** Flashing Failed **
-1

```

my cosutm esp-wroom-32-jlink.cfg file:
```
adapter speed 4000
source [find interface/jlink.cfg]
set ESP32_FLASH_VOLTAGE 3.3
source [find target/esp32.cfg]
```

### Expected behavior

I have tried to change the 'adapter speed' in .cfg file but no luck.
I will be happy with any help.

erhankur, 2023-05-22:
@UKenda can you please try with the latest release? Do you see the same issue with one of the wifi or bl idf examples?

UKenda, 2023-05-23:
Yes, the latest release (v0.12.0-esp32-20230419) solves my problem, thanks.

erhankur, 2023-05-23:
Thanks for informing.

closed 2023-05-23 by erhankur

erhankur, 2024-09-09:
Can you try adding the `no_clock_boost` option. Please try from the command line first. Something like this.

```
 openocd.exe -f <config file> -c "program_esp_bins build flasher_args.json compress verify no_clock_boost; reset; exit"
 ```

FrancoisB-HEX, 2024-09-09:
> Can you try adding the `no_clock_boost` option. Please try from the command line first. Something like this.
> 
> ```
>  openocd.exe -f <config file> -c "program_esp_bins build flasher_args.json compress verify no_clock_boost; reset; exit"
> ```

Hi and thanks,

I just tried adding `no_clock_boost` as a `program_esp` argument and that works fine. I will investigate adding this to my IDE OpenOCD config somehow.