espressif/openocd-esp32 #290: OpenOCD - GDB Session

Duolabs, 2023-05-24:
### Development Kit

Custom ESP32S3 board

### Module or chip used

ESP32S3-WROOM-1-N8R2

### Debug Adapter

Built in JTAG

### OpenOCD version

pen On-Chip Debugger v0.12.0-esp32-20230313

### Operating System

Ubuntu

### Using an IDE ?

Eclipse

### OpenOCD command line

-s ${openocd_path}/share/openocd/scripts -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

5.0.1

### Problem Description

**Debug Large applications OpenOCD fails to startup due to a timeout error.**

Scenario:

**Debugger {tab}**
[X] Start GDB Sesssion

**Startup {tab}**
OpenOCD Options:
[X] Flash every time with application binaries

When the programming time of the module exceeds 25 seconds the error raises as follows:
_**Error in services launch sequence
 Starting OpenOCD timed out.**_

This issue is widely spread on the net and many users are reporting it in the same way as we do.

If we short the source code and let the module program below 25 seconds everything works fine. Once you cross the boundary of 25 seconds it fails all of the times.

The problems seems to be on a HARDCODED 25 seconds timeout.
We've tried to force the time out in different points of the Eclipse IDE:

GDB Client setup
Commands:
set mem inaccessible-by-default off
**set remotetime 50** 

**NO EFFECT**

Windows->Preferences->Espressif->GDB server launch timeous(s) 
50

**NO EFFECT**

I am pretty sure that this damned 25 seconds is HARDCODED somewhere and I cannot find where it is.

To the best of my knownledge there is NO WAY to change this timeout and it always triggers when the Application Flashing time exceeds such limit. The problem is existing in IDF 5.0.1 such as in old 4.x.x versions.

Yet again please do not come up with suggestions like drivers or such as it is a matter of fact that with 24.899 ms it works and when I reach 26.243 it fails.

Please help us as it is extremely frustrating as our application is around 1.5MB which takes pretty much that time and please consider that this problem is running around since months on forums.

### Expected behavior

OpenOCD should not timeout and work properly.

erhankur, 2023-05-24:
@alirana01 can you have a look? 
cc @kolipakakondal

Duolabs, 2023-05-25:
Hallo. Anybody in charge of this?

igrr, 2023-05-25:
This looks like a relevant issue: https://github.com/espressif/idf-eclipse-plugin/issues/423#issuecomment-1038997809

> Eclipse > Prefernces > Espressif > GDB server launch timeout

Could you please try increasing the "GDB Server launch timeout" option value and see if that helps?

Duolabs, 2023-05-25:
Hallo @igrr 

Nice to see you again....

As I mentioned earlier we tested it over and over before posting. Of course I've tried this before and it never works.

On this comment by @kolipakakondal 
https://github.com/espressif/idf-eclipse-plugin/issues/423#issuecomment-1038997809
He says that: "Hello, _We have increased the default timeout for the OpenOCD server from 15 sec to 25 sec with the v2.4.1_" which makes me think that it is hardcoded. 

Anyhow the test is easy to be carried. If you set it to 3 seconds you should have a timeout right away as the flasher usually does program the whole device on time. That being said you will notice that the timeout will trigger after 25 seconds (if not hooked) regardless of the time you set.
Make the test yourself and you will notice that it won't trigger in 3 seconds. 
I am 100% sure that "GDB Server launch timeout" is not taken into consideration.

I've also tried to set it manually as it follows:
Edit Configuration->Startup->Initialization commands:
_mon reset halt
flushregs
set remote hardware-watchpoint-limit 2
**set remotetimeout 30**_
Regardless of the command sequence nothing happens

Then I've tried on: 
Edit Configuration->Debugger->GBD Client Setup
[Commands]
set mem inaccessible-by-default off
**set remotetimeout 30**
Regardless of the command sequence nothing happens

Here comes some screenshots as per reference.
[image]
![Screen2](https://github.com/espressif/openocd-esp32/assets/8825675/07c
[image]
cac1d-5788-408e-a6fb-7ccd32284140)

My guess is that something has changed in your OpenOCD or GDB client plugins along the way.

Duolabs, 2023-05-25:
Hallo @igrr 

Nice to see you again....

As I mentioned earlier we tested it over and over before posting. Of course I've tried this before and it never works.

On this comment by @kolipakakondal 
https://github.com/espressif/idf-eclipse-plugin/issues/423#issuecomment-1038997809
He says that: "Hello, _We have increased the default timeout for the OpenOCD server from 15 sec to 25 sec with the v2.4.1_" which makes me think that it is hardcoded. 

Anyhow the test is easy to be carried. If you set it to 3 seconds you should have a timeout right away as the flasher usually does program the whole device on time. That being said you will notice that the timeout will trigger after 25 seconds (if not hooked) regardless of the time you set.
Make the test yourself and you will notice that it won't trigger in 3 seconds. 
I am 100% sure that "GDB Server launch timeout" is not taken into consideration.

I've also tried to set it manually as it follows:
Edit Configuration->Startup->Initialization commands:
_mon reset halt
flushregs
set remote hardware-watchpoint-limit 2
**set remotetimeout 30**_
Regardless of the command sequence nothing happens

Then I've tried on: 
Edit Configuration->Debugger->GBD Client Setup
[Commands]
set mem inaccessible-by-default off
**set remotetimeout 30**
Regardless of the command sequence nothing happens

[image]

[image]

[image]

kolipakakondal, 2023-05-26:
Hi @Duolabs We apologize for the inconvenience. Let us investigate and get back to you on this.

Duolabs, 2023-05-26:
Hi @kolipakakondal 
Thank you for your answer. Here comes the Error log 
[ErrorLog.txt](https://github.com/espressif/openocd-esp32/files/11572702/ErrorLog.txt)

Duolabs, 2023-05-26:
I will do it right away!

Duolabs, 2023-05-26:
@kolipakakondal unfortunately the change did not make any effect. Still 25 seconds.
Question: does it make any difference if we run Eclipse in Ubuntu in SUDO compared to normal user?
FYI we are running it SUDO mode.

Same problem here come the log:

```
!ENTRY com.espressif.idf.core 1 0 2023-05-26 16:38:11.882
!MESSAGE Open On-Chip Debugger v0.12.0-esp32-20230313 (2023-03-13-09:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

!ENTRY org.eclipse.embedcdt.debug.gdbjtag.core 4 1 2023-05-26 16:38:11.884
!MESSAGE /home/ubuntu/esp/openocd-esp32/bin/openocd -c gdb_port 3333 -c telnet_port 4444 -c tcl_port 6666 -s /home/ubuntu/esp/openocd-esp32/share/openocd/scripts -f board/esp32s3-builtin.cfg -c program_esp_bins /home/ubuntu/Desktop/_ESP32_WORKSPACE_/ESP32_Basic_Project_0.3.8/build flasher_args.json verify reset -c echo "Started by GNU MCU Eclipse"

!ENTRY org.eclipse.cdt.dsf 4 10005 2023-05-26 16:38:37.880
!MESSAGE Request for monitor: 'RequestMonitor (org.eclipse.cdt.dsf.concurrent.RequestMonitorWithProgress@1b44275): Status ERROR: org.eclipse.embedcdt.debug.gdbjtag.core code=5010 Starting OpenOCD timed out.' resulted in an error.
!SUBENTRY 1 org.eclipse.embedcdt.debug.gdbjtag.core 4 5010 2023-05-26 16:38:37.880
!MESSAGE Starting OpenOCD timed out.

!ENTRY org.eclipse.cdt.dsf.gdb 4 5012 2023-05-26 16:38:37.882
!MESSAGE Error in services launch sequence
!STACK 1
org.eclipse.core.runtime.CoreException: Starting OpenOCD timed out.
	at org.eclipse.cdt.dsf.concurrent.Sequence.abortExecution(Sequence.java:599)
	at org.eclipse.cdt.dsf.concurrent.Sequence$2.handleErrorOrWarning(Sequence.java:448)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleFailure(RequestMonitor.java:433)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleCompleted(RequestMonitor.java:393)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor$2.run(RequestMonitor.java:317)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:833)
!SUBENTRY 1 org.eclipse.embedcdt.debug.gdbjtag.core 4 5010 2023-05-26 16:38:37.883
!MESSAGE Starting OpenOCD timed out.
```

sigmaaa, 2023-05-26:
Hi @Duolabs,

If you were to set the timeout value in the Espressif Preference Page to a lower value, such as 5 seconds, would you faster encounter a timeout error? It would be appreciated if you could test this scenario to confirm that the setting does not have any impact in your specific case.

Duolabs, 2023-05-26:
@sigmaaa 

> Make the test yourself and you will notice that it won't trigger in 3 seconds.

That's what I did at first and of course it ALWAYS triggers at 25 seconds no matter what.
I made further tests spreading **set remotetimeout** option here and there and nothing changes.
I am a bit doubtful about the user priviledges as I am always running Eclipse in **sudo** mode but I still have no clue where this is coming from.

Question #1: when I set the GDB Timeout in preferences menu on which file is it stored?
Question #2: do you have a 25 seconds value as default value somewhere in your code?

Please help as after 6+ plus of programming 24/7 on this chip this problem starts to become a nightmare!

Duolabs, 2023-05-26:
Addition:
I exported Eclipse preferences to a file and the value is written correctly:
**/instance/com.espressif.idf.core/fGdbServerLaunchTimeout=60**

Suggestion: can you please tell me where to set the default value (25) that come us when you "Restore default values". I am not much familiar with Eclipse plugs-ins and I don't know if it is hardcoded in the .jar file or on an external file.
Just in case it is hardcoded can you provide me with a default value of 60 seconds just to test?

sigmaaa, 2023-05-26:
> That's what I did at first and of course it ALWAYS triggers at 25 seconds no matter what.

I mean to try to set it to a lower value with the installed new artifact that @kolipakakondal has provided. 

> Addition:
> I exported Eclipse preferences to a file and the value is written correctly:
> **/instance/com.espressif.idf.core/fGdbServerLaunchTimeout=60**

in the artifact, that @kolipakakondal sent, fGdbServerLaunchTimeout should be related to com.espressif.idf.ui (instead of com.espressif.idf.com). Do you have this  /instance/com.espressif.idf.ui/fGdbServerLaunchTimeout = * in exported preferences?

in product log I noticed, that IDF Eclipse Plugin Version: 2.7.0.202211101524, but with provided artifact it should be IDF Eclipse Plugin Version: 2.10.0.202305261108. Did you already revert a patch? 

> Just in case it is hardcoded can you provide me with a default value of 60 seconds just to test?

The default value is pre-set and cannot be easily changed. However, we can provide you a customized build with a modified default value for you to test. Meanwhile, I was conducting some tests using the artifact available at https://github.com/espressif/idf-eclipse-plugin/suites/13176286359/artifacts/717024151 (It's slightly different from the one shared by @kolipakakondal). Based on my findings, when I set the server gdb timeout too low using this artifact, I encountered a timeout exception. However, I was able to resolve this issue by adjusting the setting to its normal value.

also, I can confirm that this problem (bug) exists on the master branch. 
@Duolabs can you check with this artifact, please:
https://github.com/espressif/idf-eclipse-plugin/suites/13176286359/artifacts/717024151
also, if you want, the same artifact but as Espressif-IDE for linux:
https://github.com/espressif/idf-eclipse-plugin/suites/13176286359/artifacts/717024152

If it doesn't work I will send you a build with a modified default value to test. Thank you

Duolabs, 2023-05-26:
**OK... there we go! It worked.**

It was my bad at the beginning as IDF Eclipse Plugin Version: 2.10.0.202305261108 was not installed properly and it was still running on my old 2.7.0.202211101524.

I deeply tested it also with a large application of 3.2MB and it works like a charm. The whole flashing cycle takes around 45secs and I was never been able to debug it in one shot.

I can also FLASH_IN_PROJECT the SPIFFS with spiffs_create_partition_image which was unimmaginable since the early days of that project.

I would like to deeply thank you guys for the support and thought I’d give you my two pence worth on this tweak.

Although the problem has been solved my vivid recommendation is to keep an eye on such troublesome issues as it seems like the tools versions as misaligned with the whole toolchain. To give you a better idea we moved from version 4.x.x of IDF to never 5.0.1 and I bet the plugin got stuck to version 2.7.0 which was causing the problem.
We'd never been able to get this to work without your support and whatever happened during the update (to 5.0.1) I guess it did not affect the plugin.
The bottom line is that I am so freaking happy to be finally able to debug without facing hurdes now!
Thank you and long life to ESP32!

closed 2023-05-29 by Duolabs