espressif/openocd-esp32 #292: Timeout with CMSIS-DAP on ESP32

gudvinr, 2023-06-27:
### Development Kit

None

### Module or chip used

ESP-WROOM-32 (wemos d1 mini32)

### Debug Adapter

DAPLink (WCH CH32F103)

### OpenOCD version

v0.12.0-esp32-20230419

### Operating System

Arch Linux

### Using an IDE ?

VSCode (PlatformIO)

### OpenOCD command line

/path/to/openocd/bin/openocd -d3 -c "gdb_port pipe; tcl_port disabled; telnet_port disabled" -s /path/to/openocd/share/openocd/scripts -f "interface/cmsis-dap.cfg" -c "cmsis_dap_vid_pid 0xc251 0xf001; cmsis_dap_backend hid" -c "set ESP32_FLASH_VOLTAGE 3.3" -f "target/esp32.cfg"

### JTAG Clock Speed

500 (and 5000 too)

### ESP-IDF version

v4.4.4 (arduino-esp32 v2.0.9)

### Problem Description

1. Attach daplink adapter and the board to PC
2. Run openocd
3. Wait

I am using platformio with following board config:
```ini
platform = espressif32@^6.3.2  ; latest default
platform_packages =
	tool-openocd-esp32@^2.1200.20230419 ; default is 2.1100.20220706
debug_tool = custom
debug_server =
  $PLATFORMIO_CORE_DIR/packages/tool-openocd-esp32/bin/openocd
  -d3
  -c "gdb_port pipe; tcl_port disabled; telnet_port disabled"
  -s $PLATFORMIO_CORE_DIR/packages/tool-openocd-esp32/share/openocd/scripts/
  -f "interface/cmsis-dap.cfg"
  -c "cmsis_dap_vid_pid 0xc251 0xf001; cmsis_dap_backend hid"
  -c "set ESP32_FLASH_VOLTAGE 3.3"
  -f "target/esp32.cfg"
debug_speed = 500
upload_protocol = esptool
board = wemos_d1_mini32
board_build.f_cpu = 240000000L
```

udev rules are set for daplink to work under normal user:
```
# CMSIS-DAP compatible adapters
ATTRS{product}=="*CMSIS-DAP*", MODE="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}="1"
ATTRS{product}=="*CMSIS_DAP*", MODE="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}="1"
```

`cmsis_dap_vid_pid` set to `c251` `f001` according to lsusb:  
```
Bus 001 Device 055: ID c251:f001 Keil Software, Inc. CMSIS_DAP
```

CMSIS-DAP with `cmsis_dap_backend hid` seems to be supported since v0.12 as seen in #237 so I can't use default openocd-esp32 that comes with espressif32 platform atm. To work around that I set debug tool in platformio to custom.

I don't use daplink for upload though. It is handled by esptool since I connect to PC via second USB anyway.

### Debug Logs

```shell
[openocd-esp32-verbose-debug.txt](https://github.com/espressif/openocd-esp32/files/11877283/openocd-esp32-verbose-debug.txt)
```

### Expected behavior

I kinda expect it to work and jump into debugging.

erhankur, 2023-06-28:
@gudvinr Based on the log, it appears that the chip is resetting unexpectedly while running the stub algorithm. Could you please disable flash support and try again? If it succeeds, we can then focus on identifying the specific issue.

```bash
/path/to/openocd/bin/openocd -d3 \
-c "gdb_port pipe; tcl_port disabled; telnet_port disabled" \
-s /path/to/openocd/share/openocd/scripts \
-f "interface/cmsis-dap.cfg" \
-c "cmsis_dap_vid_pid 0xc251 0xf001; cmsis_dap_backend hid" \
-c "set ESP32_FLASH_VOLTAGE 3.3" \
-c "set ESP_FLASH_SIZE 0" \
-f "target/esp32.cfg"
```

gudvinr, 2023-06-30:
> Could you please disable flash support and try again?

Sure. Now it does not reset but there are still errors:
[openocd-log-flash0.txt](https://github.com/espressif/openocd-esp32/files/11915406/openocd-log-flash0.txt)

Also since there are errors related to `esp_common.cfg`, I am also attaching the one that comes bundled in pio package:
[esp_common.cfg.txt](https://github.com/espressif/openocd-esp32/files/11915398/esp_common.cfg.txt)

gudvinr, 2023-07-26:
@erhankur any follow up on this one?

dphzz, 2023-08-02:
Hello, I am also having the same issue, I was using [wuxx](https://github.com/wuxx/nanoDAP/tree/master) nanoDAP firmware on an STM32F103C8T6 and openOCD keeps reporting that the target gets reset continuously. 

[openocd_esp32_flash0_timeout1000.txt](https://github.com/espressif/openocd-esp32/files/12242891/openocd_esp32_flash0_timeout1000.txt)

erhankur, 2023-08-03:
@dphzz I did some test [here](https://github.com/espressif/openocd-esp32/issues/237#issuecomment-1219990478)

Reading my notes, you might need to increase reset timeout from the source code.  If it helps, I can consider to make it configurable.

dphzz, 2023-08-03:
Thank you for the quick reply, 
Do I have to recompile openOCD from source to adjust that or there is some way to do it from the .cfg file?

erhankur, 2023-08-03:
This is not configurable yet. You have to compile it from the source. If you can't, tell me your OS, and I will share the compiled binary.

dphzz, 2023-08-03:
@erhankur I am running on Windows 10 64bit version.

erhankur, 2023-08-03:
[openocd-esp32-win32-0.12.0-esp32-20230419-192-g9ef398ef8.zip](https://github.com/espressif/openocd-esp32/files/12249100/openocd-esp32-win32-0.12.0-esp32-20230419-192-g9ef398ef8.zip)

Please replace this package with yours. including script files. 

For a reference; I increased this timeout from 100ms to 2000ms now. 
https://github.com/espressif/openocd-esp32/blob/master/src/target/espressif/esp32.c#L229 

Lets see if it helps.

dphzz, 2023-08-03:
@erhankur I run the new openOCD in the terminal, the problem is still the same. It doesn't seem like the timeout was increased at all. 
Here is the new log file. 
[openocd_esp32_flash0_timeout2000.txt](https://github.com/espressif/openocd-esp32/files/12249248/openocd_esp32_flash0_timeout2000.txt)

erhankur, 2023-08-03:
[openocd-esp32-win32-0.12.0-esp32-20230419-193-gd1405eab6.zip](https://github.com/espressif/openocd-esp32/files/12249353/openocd-esp32-win32-0.12.0-esp32-20230419-193-gd1405eab6.zip)

Ok. last try. If it doesn't help, I will re-test with my setup.

dphzz, 2023-08-03:
@erhankur The problem still persists. 
It worth noting that I was not using the commercial nanoDAP from wuxx like the [previously post](https://github.com/espressif/openocd-esp32/issues/237#issuecomment-1219990478). I was using an STM32 Bluepill (C8 variant) flashed with [wuxx's v2.3 daplink firmware](https://github.com/wuxx/nanoDAP/blob/master/firmware/v2.3/DAPLink/flash_image_20210104.hex) so maybe the problem wasn't from openocd but from the actual firmware itself but I am not sure. 
[debug_openocd_esp32_flash0_timeout2000.txt](https://github.com/espressif/openocd-esp32/files/12249444/d
[simple_openocd_esp32_flash0_timeout2000.txt](https://github.com/espressif/openocd-esp32/files/12249447/simple_openocd_esp32_flash0_timeout2000.txt)
ebug_openocd_esp32_flash0_timeout2000.txt)

erhankur, 2023-08-03:
At least there is no reset timeout now. I don't know now why your board resetting consistently. Reset reason seems odd. You can try disabling watchdogs. You can make sure, your application works fine without debugger.  Or if any chance to connect other debuggers? Sorry, not much familiar with DAP probes.

dphzz, 2023-08-03:
@erhankur I did use a jlink to connect to my esp32 running debugger on platformIO and it works fine.  
As for the DAP probes, I am running a simple program which print Hello World every second to serial monitor, and when the DAP probe attempt connect to the probe, the serial monitor is garbled into random characters. 

[image]

gudvinr, 2023-08-03:
@erhankur I have [FTDI-ish debugger](https://github.com/sipeed/RV-Debugger-BL702) that works with openocd-esp32 and [modified config](https://github.com/espressif/openocd-esp32/pull/142#issuecomment-1652351342) without modifications.

I might test your build with daplink too but I can't really build it myself right now (x86_64 linux).

erhankur, 2023-08-04:
@gudvinr you can try 
[openocd-esp32-linux-amd64-0.12.0-esp32-20230419-193-gd1405ea.tar.gz](https://github.com/espressif/openocd-esp32/files/12259052/openocd-esp32-linux-amd64-0.12.0-esp32-20230419-193-gd1405ea.tar.gz)

arthurfprecht, 2023-10-01:
Hello! Any follow up on this issue? I have a similar setup from @dphzz - a nanoDAPlink flashed on a STM32F103CB Blue Pill. My ESP32 is a WROOM Dev module. Using the Windows compiled OpenOCD provided here a few posts ago, I don't get the reset timeouts, but the ESP32 keeps being reset, just as @dphzz. On an interesting note, if I put the ESP32 in bootloader mode, the reset messages go away (but the debugging still doesn't work, gdb fails after some time. 
[OpenOCD log.txt](https://github.com/espressif/openocd-esp32/files/12778488/OpenOCD.log.txt)
[GDB log.txt](https://github.com/espressif/openocd-esp32/files/12778489/GDB.log.txt)

erhankur, 2024-01-09:
@arthurfprecht @gudvinr sorry for the late response. I don't know the latest status of this but to avoid timeout issues, did you try using faster adapter speed like 5000 khz?

erhankur, 2024-02-24:
@gudvinr I am closing this. Feel free to re-open if you need something from our side.