espressif/openocd-esp32 #293: esp32s3 vscode openocd debug Warn : address 0x42008342 not writable

YuLianChen, 2023-08-03:
### Development Kit

esp32s3

### Module or chip used

esp32s3

### Debug Adapter

usb jtag

### OpenOCD version

Open On-Chip Debugger v0.11.0-esp32-20221026

### Operating System

ubuntu

### Using an IDE ?

vscode

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

40000 kHz

### ESP-IDF version

4.4.5

### Problem Description

1. Open a shell window and enter the command openocd -f board/esp32s3-builtin.cfg
2. Press F5 to start debugging
3. Then it will report an error

### Debug Logs

```shell
openocd -f board/esp32s3-builtin.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:47)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:E9:EC:CC)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 35 KB
Info : Flash mapping 1: 0x20020 -> 0x3c070020, 132 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 35 KB
Info : Flash mapping 1: 0x20020 -> 0x3c070020, 132 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 0 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 35 KB
Info : Flash mapping 1: 0x20020 -> 0x3c070020, 132 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 136 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Warn : address 0x42008342 not writable
Error: [esp32s3.cpu0] Error writing instruction to memory: -4
Error: [esp32s3.cpu0] Failed to write breakpoint instruction (-4)!
Error: [esp32s3.cpu0] Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
```

### Expected behavior

debug run correctly

erhankur, 2023-08-03:
Your flash mapping looks strange to me.  I expect something like 
```
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 42 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 94 KB
```

Can you full-clean and rebuild from scratch. I remember some users had similar problem and rebuild fixed it.

YuLianChen, 2023-08-09:
Sorry for not replying for so long, I seem to have found the reason, it seems to be related to the use of ota partition table, when I use a simple code, such as helloword, it can run, but when I add the partition table, he started to report an error, I don't know how to solve this problem
full-clean and rebuild is useless
Below is my partition table
# Name,   Type, SubType, Offset,   Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,        0x4000,
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
ota_0,    app,  ota_0,   ,        2M,
ota_1,    app,  ota_1,   ,        2M,
storage,  data, spiffs,  ,        10M,

erhankur, 2023-08-09:
I tested with your partition table with default boot loader size. No error so far. Can you send your sdkconfig also? And what do you see in the idf console when you start your application? This is mine.
```
I (35) boot: ESP-IDF v5.2-dev-2055-g81d1f674ee-dirty 2nd stage bootloader
I (35) boot: compile time Aug  9 2023 11:49:32
I (36) boot: Multicore bootloader
I (40) boot: chip revision: v0.1
I (44) boot.esp32s3: Boot SPI Speed : 80MHz
I (49) boot.esp32s3: SPI Mode       : SLOW READ
I (54) boot.esp32s3: SPI Flash Size : 16MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (82) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (90) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  3 ota_0            OTA app          00 10 00010000 00200000
I (105) boot:  4 ota_1            OTA app          00 11 00210000 00200000
I (112) boot:  5 storage          Unknown data     01 81 00410000 00a00000
I (120) boot: End of partition table
I (124) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=32440h (205888) map
I (182) esp_image: segment 1: paddr=00042468 vaddr=3fc97200 size=05850h ( 22608) load
I (188) esp_image: segment 2: paddr=00047cc0 vaddr=40374000 size=08358h ( 33624) load
I (198) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=a19ech (661996) map
I (356) esp_image: segment 4: paddr=000f1a14 vaddr=4037c358 size=0ae0ch ( 44556) load
I (377) boot: Loaded app from partition at offset 0x10000
I (378) boot: Disabling RNG early entropy source...
I (389) cpu_start: Multicore app
I (389) cpu_start: Pro cpu up.
I (390) cpu_start: Starting app cpu, entry point is 0x403753c0
```

closed 2023-12-28 by erhankur