espressif/openocd-esp32 #294: XIAO ESP32-C3 Failed to get flash maps

tutoduino, 2023-08-11:
### Development Kit

XIAO ESP32-C3

### Module or chip used

ESP32-C3

### Debug Adapter

Builtin

### OpenOCD version

v0.12.0-esp32-20230419-191-g4da79b3e

### Operating System

Linux

### Using an IDE ?

No, command line

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

Default

### ESP-IDF version

v5.1

### Problem Description

1/ Start OpenOCD
tutoduino@my-computer:~/esp/openocd-esp32$ openocd -f board/esp32c3-builtin.cfg

2/ Start GDB with hello_world example
tutoduino@my-computer:~/esp/hello_world$ riscv32-esp-elf-gdb -x gdbinit build/hello_world.elf 

3/ OpenOCD raised errors
Failed to get flash maps!
Failed to get flash size!

4/ GDB close remote connection

### Debug Logs

```shell
Open On-Chip Debugger v0.12.0-esp32-20230419-191-g4da79b3e (2023-08-11-13:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (D4:F9:8D:03:EF:80)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] datacount=2 progbufsize=16
Info : [esp32c3] Examined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Error: Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
Error: Failed to get flash size!
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```

### Expected behavior

OpenOCD should not raise the error and GDB should allow debuging of the program.

tutoduino, 2023-08-11:
The problem does not appear if I use the suggested option "gdb_memory_map disable" : 
openocd -f board/esp32c3-builtin.cfg -c "gdb_memory_map disable"

tutoduino, 2023-08-16:
> Hi @tutoduino some users had the same problem and they fixed it by changing the state of the strapping pins. [#241 (comment)](https://github.com/espressif/openocd-esp32/issues/241#issuecomment-1400230555)

Indeed, setting pin D8 state to high during boot is indeed a good solution. Though documentation is not very clear regarding this...
I updated my tutorial to mention this!
Thanks @erhankur.

erhankur, 2023-08-16:
Thank you for verifying. I will leave this issue open until I update the documentation.

IronMan-Nam, 2024-08-20:
`./openocd -f board/esp32c3-builtin.cfg -c "program_esp C:\Users\NAM\Desktop\hello_world\build\hello_world.bin 0x8000 verify exit"
Open On-Chip Debugger v0.12.0-esp32-20240726 (2024-07-26-15:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
WARNING!!! OpenOCD binary version (Open On-Chip Debugger v0.12.0-esp32-20240726 (2024-07-26-15:19)) doesn't match the .cfg file version (v0.12.0-esp32-20230313)
** program_esp input args <0x8000 verify exit> **
Info : esp_usb_jtag: serial (24:58:7C:AE:EA:C4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] datacount=2 progbufsize=16
Info : [esp32c3] Examined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : [esp32c3] Examination succeed
Info : [esp32c3] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] Reset cause (3) - (Software core reset)
** Programming Started **
Info : [esp32c3] Found 8 triggers
Warn : Failed to get flash maps (-9)!
Warn : Failed to read flash size!
Warn : Failed to get flash mappings (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32c3.flash'!
** Clock configuration set failed **
shutdown command invoked
C:/Espressif/tools/openocd-esp32/v0.12.0-esp32-20230313/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg:9: Error:
at file "C:/Espressif/tools/openocd-esp32/v0.12.0-esp32-20230313/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg", line 9
PS C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230313\openocd-esp32\bin>`

I'm trying flash new program to ESP32-C3 on kit https://dl.artronshop.co.th/ESP32-C3%20SuperMini%20datasheet.pdf

but it not success. above result I try to download application.
Please help me.

IronMan-Nam, 2024-08-20:
```
PS C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230313\openocd-esp32\bin> ./openocd -f board/esp32c3-builtin.cfg -c "program_esp C:\Users\NAM\Desktop\hello_world\build\hello_world.bin 0x8000 verify exit"
Open On-Chip Debugger v0.12.0-esp32-20240726 (2024-07-26-15:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
WARNING!!! OpenOCD binary version (Open On-Chip Debugger v0.12.0-esp32-20240726 (2024-07-26-15:19)) doesn't match the .cfg file version (v0.12.0-esp32-20230313)
** program_esp input args <0x8000 verify exit> **
Info : esp_usb_jtag: serial (24:58:7C:AE:EA:C4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] datacount=2 progbufsize=16
Info : [esp32c3] Examined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : [esp32c3] Examination succeed
Info : [esp32c3] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] Reset cause (3) - (Software core reset)
** Programming Started **
Info : [esp32c3] Found 8 triggers
Warn : Failed to get flash maps (-9)!
Warn : Failed to read flash size!
Warn : Failed to get flash mappings (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32c3.flash'!
** Clock configuration set failed **
shutdown command invoked
C:/Espressif/tools/openocd-esp32/v0.12.0-esp32-20230313/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg:9: Error:
at file "C:/Espressif/tools/openocd-esp32/v0.12.0-esp32-20230313/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg", line 9
```

I'm trying flash new program to ESP32-C3 on kit https://dl.artronshop.co.th/ESP32-C3%20SuperMini%20datasheet.pdf

but it not success. above result I try to download application.
Please help me.