espressif/openocd-esp32 #295: Failed To init Debug adapter (Failed to get flash maps, failed to get flash size)

5ami, 2023-08-21:
### Development Kit

Customs Board

### Module or chip used

ESP32s3 Chip 

### Debug Adapter

esp32s3 on-chip JTAG 

### OpenOCD version

v0.11.0-esp32-20220706

### Operating System

Windows

### Using an IDE ?

VScode with ESP-IDF extension 

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

40000 kHz

### ESP-IDF version

v5.0

### Problem Description

I have a costume board with esp32s3 chip (not a wroom). I could flash software for it using the USB/JTAG, however I cant get the Debuger connected to it. I get Failed To init Debug adapter (Failed to get flash maps, failed to get flash size).

With the same menuconfig settings, same environment  and same software uploaded, the debugger works with other esspressif boards. 

The only difference is that My costume board has  a 512KB flash. and the log from openOCD says: 
[OpenOCD]
❌ Error: Failed to get flash maps (4294967295)!
Warn : Failed to get flash mappings (-4)!

[OpenOCD]
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001

[OpenOCD]
❌ Error: Failed to get flash size!

[OpenOCD]
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debu
[OpenOCD]
g_reason=00000001

[OpenOCD]
❌ Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
[OpenOCD]
❌ 

### Debug Logs

```shell
Here is the full Log:

[OpenOCD]
[... 251 lines cut ...]

[Debug Adapter]
2023-08-21 12:03:57,538 - Debug Adapter (main) - DEBUG - Stopping target poller
2023-08-21 12:03:57,538 - Debug Adapter (main) - DEBUG - Closing the GDB process
2023-08-21 12:03:57,538 - Debug Adapter (main) - DEBUG - Closing the Socket
2023-08-21 12:03:57,539 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug Adapter stopped\n", "source": {}}, "seq": 11}

[Debug Adapter]
2023-08-21 12:03:57,539 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
2023-08-21 12:03:57,539 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Adapter is Stopping...\n", "source": {}}, "seq": 12}

2023-08-21 12:03:57,539 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
2023-08-21 12:03:57,539 - Debug Adapter (main) - DEBUG - Adapter is Stopped

[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2023-08-21 12:03:57,539 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2023-08-21 12:03:57,540 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect

[Stopped] : ESP-IDF Debug Adapter
```

### Expected behavior

The debugger should connect.

5ami, 2023-08-22:
Extra note: I have a costume partition table in csv format. 
[image]

erhankur, 2023-08-24:
@5ami Before getting into detail, can you update your OpenOCD to the latest release and share the verbose logs with -d3 option?
Also, it might be useful to see your idf application boot logs.

erhankur, 2024-01-03:
@5ami do you still have an issue?

erhankur, 2024-01-15:
Thanks for reporting. Feel free to re-open if you still have an issue.

closed 2024-01-15 by erhankur