espressif/openocd-esp32 #296: Error: timed out while waiting for target halted  , which with my own debugger(CH347)

Yecgaa1, 2023-08-31:
### Development Kit

ESP32-DEVKIT

### Module or chip used

ESP32-WROVER-E

### Debug Adapter

CH347 with jtag mode

### OpenOCD version

build with master branch

### Operating System

Windows

### Using an IDE ?

cmd?

### OpenOCD command line

C:\Users\xtx\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe -c "tcl_port disabled" -s C:\Users\xtx\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts -f C:\GitProject\ESP32\hello_world\ESP32-ch347.cfg -c "program_esp C:/GitProject/ESP32/hello_world/cmake-build-debug/bootloader/bootloader.bin 0x0 verify" -c "program_esp C:/GitProject/ESP32/hello_world/cmake-build-debug/partition_table/partition-table.bin 0x8000 verify" -c "program_esp C:/GitProject/ESP32/hello_world/cmake-build-debug/hello_world.bin 0x10000 verify" -c "" -c "echo \"** Program Flash Complete! **\"" -c shutdown

### JTAG Clock Speed

4000 and 10000 also tried

### ESP-IDF version

5.0.2

### Problem Description

I used the code for this repository, then added support for CH347 ( as interfaces ) and compiled it, and then this problem occurred

It may be a tough question, but I do need your help, and I ' d like to provide more information you need.

Here is the openocd log
[log.zip](https://github.com/espressif/openocd-esp32/files/12482594/log.zip)

### Expected behavior

Flash the ESP32 With Jtag

Yecgaa1, 2023-08-31:
https://github.com/xutongxin1/openocd-esp32 
Here is my fork repository from this repository, thanks

Yecgaa1, 2023-08-31:
Add the content in ESP32-ch347.cfg

gerekon, 2023-09-05:
Also looks like ch347 is not supported by OpenOCD out of the box. I can find neither `ch347.c` nor `ch347.cfg` in the repo. So I am afraid we can not help you more in this case.

Yecgaa1, 2023-09-05:
> 

Here is the photo I'm wired
I use ESP-WROVER-KIT_V4.1 and CH347T-EVT-R0-1v1 to test
In my photo, I removed the jumper cap from the ESP32 to prevent the on-board FT2232 from interfering with the CH347 signal
[image]
[image]
Also, I saw the openocd prompt, does this mean that jtag communication can be successfully established, but cannot be reset
[image]

Yecgaa1, 2023-09-05:
> Also looks like ch347 is not supported by OpenOCD out of the box. I can find neither `ch347.c` nor `ch347.cfg` in the repo. So I am afraid we can not help you much in this case.

Please forgive me for not providing the address of this file.
https://github.com/xutongxin1/openocd-esp32/blob/master/src/jtag/drivers/ch347.c
If you need more information, I'll give it to you as soon as possible

Yecgaa1, 2023-09-12:
After communicating with the engineer of WDH Company (manufacturer of CH347), we think there may be some problems with esp_common.cfg, and we need your help

Yecgaa1, 2023-09-17:
The manufacturer of the CH347 says it will take some time and is not currently incorporated into the upper warehouse

I used openocd-esp32 compiled by myself, ESP32-Wrover-kit, and FT2232 on board successfully burned, the following is the log
```
C:/Users/xtx/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin/openocd.exe -f board/esp32-wrover-kit-3.3v.cfg -c "program_esp C:/GitProject/ESP32/hello_world/cmake-build-debug/bootloader/bootloader.bin 0x1000 verify exit"
Open On-Chip Debugger v0.12.0-dirty (2023-08-30-22:52)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
** program_esp input args <0x1000 verify exit> **
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32.cpu0 on 3333
Error: couldn't bind gdb to socket on port 3333: No error
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000001
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 79 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 468 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu0.flash' size 4096 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 79 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 468 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu1.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu1.flash' size 4096 KB
** Programming Started **
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : PROF: Erased 28672 bytes in 485.528 ms
Info : PROF: Data transferred in 93.996 ms @ 297.885 KB/s
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : PROF: Wrote 28672 bytes in 343.118 ms (data transfer time included)
** Programming Finished in 1638 ms **
** Verify Started **
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : PROF: Flash verified in 51.919 ms
** Verify OK **
shutdown command invoked
```

closed 2023-09-18 by gerekon

EasyDevKits, 2023-10-03:
@Yecgaa1 for your info: I'm trying to do the same. Getting the JTAG part of the CH347 working for ESP32. But for me it seams that at least the Shift-IR is not working correctly. I've analyzed the JTAG signals with a logic analyzer and have seen, that the signal output is different then the signal output with a FTDI2232. If you like we can work together to get the CH347 working for ESP32.

My repo is here: https://github.com/EasyDevKits/openocd-easydevkits

Yecgaa1, 2023-10-07:
> @Yecgaa1 for your info: I'm trying to do the same. Getting the JTAG part of the CH347 working for ESP32. But for me it seams that at least the Shift-IR is not working correctly. I've analyzed the [...]

I'm glad you have that idea. How can I help you？
I don't know much about Jtag signals, but I am in touch with the CH347 manufacturer engineers

EasyDevKits, 2023-10-07:
Thanks @Yecgaa1, I've made progress and narrows down the issue.

If you compare the FTDI log with the CH347 log there is one crucial line missing in the CH347 log:

`Info : [esp32.cpu1] Debug controller was reset.`

With the CH347 chip you will never get to the point where the debug controller (the TAP controller) for the CPU1 (the second ESP32 core) will be reset. With CH347 you get this only for CPU0 (the first ESP32 core). But as far as I know you need to get both CPU cores into the debugging mode. Otherwise the cpu0 will halt and the cpu1 is running.
The message 

`Info : [esp32.cpu1] Core was reset.`

says only that the core was reset but runs after the reset. At the end we need also for cpu1 the log message

`Info : [esp32.cpu1] Target halted, PC=...`

I've searched how it's possible to take both CPU cores into debugging mode. If one TAP controller (debug controller) is reset, they send a signal to the other cores to enter the debugging mode. This corresponds to this log line:

```
Info : 1779 784 xtensa.c:2235 xtensa_poll(): [esp32.cpu0] Debug controller was reset.
Debug: 1780 785 xtensa.c:871 xtensa_smpbreak_write(): [esp32.cpu0] write smpbreak set=0x30000 clear=0x600000
```

They send a smp break signal. This results in a few JTAG commands. The bits and bytes that are send to the ESP32 are the same regardless if you use a FTDI or CH347. But the reaction of the ESP32 via JTAG differs. The next 32 bits that are read for the FTDI case are 0x0733bff2, but in the CH347 case the next 32 bits that are read are 0x0e677fe4.

I've not yet decoded that data that comes back from JTAG interface. But I've no idea why it differs. Maybe there is a timing problem?

But anyways, I think it would be very helpful if you could get more information from the WCH engineers why they think it's a bigger problem to get CH347 running with ESP32.

And if you like we can also have contact via email: info@easydevkits.com

EasyDevKits, 2023-10-08:
It took me a while but finally I found that 0x0733bff2 that I've mentioned in the previous comment is the ESP32 device id. After that I've found the when I left shift the value 0x0733bff2 by one bits I get exactly the value 0x0e677fe4 that I've also mentioned in my previous comment. So it's only a bug in the ch347 source code in the function CH347_Read_Scan

```
for (i = 0; i < cmd->num_fields; i++) {
    /* if neither in_value nor in_handler
     * are specified we don't have to examine this field
     */
    LOG_DEBUG("fields[%i].in_value[%i], offset: %d", i, cmd->fields[i].num_bits, offset);
    if (cmd->fields[i].in_value) {
        ...
	bit_count += cmd->fields[i].num_bits;
    }
}
```

This code advance the bit_count only for the fields that have data. But for the fields without data the code should skip the bits because they are already read. So **bit_count += cmd->fields[i].num_bits;** needs to be outside of the if condition.

I've now an enhanced version that's working with the ESP32 here: https://github.com/EasyDevKits/openocd-easydevkits

Yecgaa1, 2023-10-10:
Yes, according to my test, now CH347 can be used to burn debugging ESP32, thank you for your efforts, I will feedback the bug to the WCH engineers, if there is any problem I will contact you again