espressif/openocd-esp32 #299: Error: Target not examined yet

diplfranzhoepfinger, 2023-10-18:
espressif/esp-idf#11086

\### Answers checklist.

\- \[X\] I have read the documentation \[ESP-IDF Programming Guide\](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.  
\- \[X\] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.  
\- \[X\] I have searched the issue tracker for a similar issue and not found a similar issue.

\### General issue report

Hi,

Try debugging with JTAG with ESP-PROG

get this Error:

!\[image\](https://user-images.githubusercontent.com/37779037/228867528-3a0ea087-622f-4bcb-b1f6-dc2c772bdedd.png)

  
get this Log:

\`\`\`  
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)  
Licensed under GNU GPL v2  
For bug reports, read  
http://openocd.org/doc/doxygen/bugs.html  
Error: libusb\_open() failed with LIBUSB\_ERROR\_NOT\_SUPPORTED  
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo\_sample\_edge falling"  
Info : clock speed 20000 kHz  
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)  
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)  
Error: Unexpected OCD\_ID = 00000000  
Error: Unexpected OCD\_ID = 00000000  
Error: Unexpected OCD\_ID = 00000000  
Error: Unexpected OCD\_ID = 00000000  
Warn : target esp32s3.cpu1 examination failed  
Error: Unexpected OCD\_ID = 00000000  
Error: Unexpected OCD\_ID = 00000000  
Info : starting gdb server for esp32s3.cpu0 on 3333  
Info : Listening on port 3333 for gdb connections  
Error: Target not examined yet  
\`\`\`

erhankur, 2023-10-18:
@diplfranzhoepfinger Can you please try with the latest OpenOCD release and IDF Hello World application?  You can use IDF master branch and install the latest OpenOCD release. 

If you still see the issue, please share the OpenOCD logs (with -d3 level) and IDF application console output.

erhankur, 2023-11-29:
@diplfranzhoepfinger any update?

erhankur, 2023-12-02:
Closed due to inactivity. Feel free to reopen if you still have an issue.

closed 2023-12-02 by erhankur

diplfranzhoepfinger, 2023-12-03:
> @diplfranzhoepfinger any update?

i meed to do tests yes. I will reopen if needed.