espressif/openocd-esp32 #300: OpenOCD timed out in Espressif-IDE

Shijomts, 2023-10-18:
### Development Kit

ESP32-S3-Pico

### Module or chip used

ESP32-S3R2

### Debug Adapter

Built in JTAG

### OpenOCD version

v0.12.0-esp32-20230419

### Operating System

Windows 11 Pro

### Using an IDE ?

Espressif-IDE

### OpenOCD command line

-s ${openocd_path}/share/openocd/scripts -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

esp-idf-v5.1.1

### Problem Description

Hi,

I am working on ESP32-S3 board (ESP32-S3-Pico - https://www.waveshare.com/wiki/ESP32-S3-Pico). I am facing issue in OpenOCD debugging. When the application size is smaller, there is no issue in starting the debugging session. But when the application size increased, I am not able to start the debugger. It is giving timeout error.

Error in services launch sequence
Starting OpenOCD timed out.

Working Environment:
 - ESP32-S3R2 Integrated 512KB SRAM, 384KB ROM, 2MB PSRAM, 16MB Flash memory
 - Open On-Chip Debugger v0.12.0-esp32-20230419
 - Total image size: 1300145 bytes

I have installed the latest Espressif-IDE using online installer.

I have increased the  OpenOCD  time also ( set remotetimeout 25), but issue is not resolved.

Regards,
Shijo Thomas

### Expected behavior

OpenOCD should not timeout and work properly.

### Screenshots

[image]

Shijomts, 2023-10-18:
Attaching the OpenOCD logs
[oocd.log](https://github.com/espressif/openocd-esp32/files/13029582/oocd.log)

Shijomts, 2023-10-18:
Yes....There is no error message in the log file. 
But couple seconds after clicking  the debug button, the error message pop-ups and saying " OpenOCD timeout", How to identify the issue

Shijomts, 2023-10-19:
I am attaching the Espressif-IDE Error log for both working and non working debug sessions,

Please help to identify the issue.

[Eclipse_log_debug_ok.log](https://github.com/espressif/openocd-esp32/files/13041544/Eclipse_log_debug_ok.log)
[Eclipse_log_debug_timeout.log](https://github.com/espressif/openocd-esp32/files/13041547/Eclipse_log_debug_timeout.log)

Shijomts, 2023-10-25:
@erhankur Anybody is there to help me

erhankur, 2023-10-25:
@Shijomts Do you try to load different sized application in both logs? What is the main difference? 

@kolipakakondal @alirana01 Do you have any comment?

Shijomts, 2023-10-26:
@erhankur Yes. I have loaded different sized applications in both logs. Actually I have commended some section of the application to make it smaller. In that case the loading was successful.

alirana01, 2023-10-26:
@Shijomts @erhankur 
You can find the option for configuring the timeout in the Preferences->Espressif
You can change the timeout settings according to your needs the default one is 25

[image]

Shijomts, 2023-10-31:
@erhankur 
I have tried with different timeout values( 50, 100, 200 etc.), but same timeout issue.

Shijomts, 2023-11-01:
@alirana01
 **I was able to download successfully using the CLI, and the application started running. No issues. Below is the CLI logs.**

Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
** program_esp input args <0x10000 verify exit> **
Info : esp_usb_jtag: serial (48:27:E2:3B:39:80)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x420179A8, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x4037C1EA, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (1) - (Power on reset)
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c0c0020, 231 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 704 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c0c0020, 231 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 704 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu1.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu1.flash' size 16384 KB
** Programming Started **
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : PROF: Erased 1302528 bytes in 5679.41 ms
Info : PROF: Data transferred in 16433.5 ms @ 77.4028 KB/s
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : PROF: Wrote 1302528 bytes in 16815.7 ms (data transfer time included)
** Programming Finished in 24325 ms **
** Verify Started **
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : PROF: Flash verified in 897.202 ms
** Verify OK **
shutdown command invoked

**But not able to debug using Espressif-IDE. Timeout error**

[image]

alirana01, 2023-11-01:
@Shijomts 
Can you please share the command you are running from CLI and also try to run this from your CLI

`C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin/openocd.exe -c gdb_port 3333 -c telnet_port 4444 -c tcl_port 6666 -s C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin/share/openocd/scripts -f board/esp32s3-builtin.cfg -c program_esp_bins C:/Espressif/frameworks/esp-idf-v5.1.1/workspace_work/spi_lcd_touch/build flasher_args.json verify reset -c echo "Started by GNU MCU Eclipse"`

alirana01, 2023-11-01:
@Shijomts can you also please share the version of IDE you have and please try to take the latest as well.

kolipakakondal, 2023-11-01:
Hi @Shijomts We have fixed a bug related to OpenOCD timeout issue with the IEP release v2.10.0 and the latest version is v2.11.1

Could you please upgrade to the latest one and see if you have the same issue.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.11.1

Here is the update site you can use to update to the latest one
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

Shijomts, 2023-11-01:
@kolipakakondal 
After upgrade I am able to debug. The issue is resolved.

I appreciate your support and assistance. Thank you very much.

Thanks a lot....
@erhankur 
@kolipakakondal
@alirana01

closed 2023-11-01 by erhankur