espressif/openocd-esp32 #302: Error: Failed to get flash maps (-8)!

8300, 2023-11-24:
### Development Kit

openocd

### Module or chip used

esp32c3fn4

### Debug Adapter

usb-jtag bultin

### OpenOCD version

Open On-Chip Debugger v0.12.0-dirty (2023-11-15-16:27)

### Operating System

linux

### Using an IDE ?

no

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

40000 kHz

### ESP-IDF version

v5.2

### Problem Description

1.openocd -f board/esp32c3-builtin.cfg
2.riscv32-unknown-linux-gnu-gdb _local/examples/gpio 
  2.1 target remote localhost:3333
3.openocd error

### Debug Logs

```shell
Open On-Chip Debugger v0.12.0-dirty (2023-11-15-16:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (A0:76:4E:41:3A:0C)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] datacount=2 progbufsize=16
Info : [esp32c3] Examined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32c3] Hart unexpectedly reset!
Info : [esp32c3] Reset cause (3) - (Software core reset)
Info : [esp32c3] Hart unexpectedly reset!
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32c3] Reset cause (3) - (Software core reset)
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Error: Failed to get flash maps (-8)!
Warn : Invalid magic number in app image!
Warn : Failed to get flash mappings (-4)!
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Error: Failed to get flash maps (-8)!
Warn : Invalid magic number in app image!
Warn : Failed to get flash mappings (-4)!
Info : Using flash bank 'esp32c3.irom' size 0 KB
Error: Failed to get flash maps (-8)!
Warn : Invalid magic number in app image!
Warn : Failed to get flash mappings (-4)!
Info : Using flash bank 'esp32c3.drom' size 0 KB
Warn : negative reply, retrying
Info : dropped 'gdb' connection
```

### Expected behavior

no error

gerekon, 2023-11-24:
@8300 Do you have problems with debugging or you are just confused with those warn/err messages?

8300, 2023-11-25:
> @8300 Do you have problems with debugging or you are just confused with those warn/err messages? Looks like they appear because you have no app image in flash. After flashing the binaries they will disappear. We are going to lower debug level for `Failed to get flash mappings` since they are normal when there are no image in flash.

yes，drop gdb connection. I can't load new firmware

gerekon, 2023-11-27:
Could you run openocd with verbose log enabled and attach it here. You need to add the following options `-d3 -l ocd.log`

8300, 2023-11-27:
> Could you run openocd with verbose log enabled and attach it here? You need to add the following options `-d3 -l ocd.log`

[Uploading ocd.log…]()

8300, 2023-11-27:
[ocd.log](https://github.com/espressif/openocd-esp32/files/13471473/ocd.log)

8300, 2023-11-27:
> Could you run openocd with verbose log enabled and attach it here? You need to add the following options `-d3 -l ocd.log`

[ocd.log](https://github.com/espressif/openocd-esp32/files/13471479/ocd.log)

gerekon, 2023-11-27:
Looks like GDB closes connection due to internal timeout (see below). Could you add `set remotetimeout 20` before `target remote localhost:3333`?

And one more question. Why do you use `riscv32-unknown-linux-gnu-gdb` instead of one supplied by Espressif (`ricv32-esp-elf-gdb`)?
```
Info : 33136 70206 server.c:90 add_connection(): accepting 'gdb' connection on tcp/3333
.
.
.
Debug: 42594 73967 gdb_server.c:1056 gdb_new_connection(): New GDB Connection: 1, Target esp32c3, state: halted
Debug: 42595 73967 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;memory-tagging+
Debug: 42596 73967 riscv.c:2404 riscv_get_gdb_reg_list_internal(): [esp32c3] reg_class=0, read=0
Debug: 42597 73967 gdb_server.c:391 gdb_log_outgoing_packet(): [esp32c3] sending packet: $PacketSize=4000;qXfer:memory-map:read+;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+#02
Debug: 42598 73967 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: -
Warn : 42599 73967 gdb_server.c:475 gdb_put_packet_inner(): negative reply, retrying
Debug: 42600 73967 gdb_server.c:388 gdb_log_outgoing_packet(): [esp32c3] sending packet: $<binary-data-116-bytes>#02
Debug: 42601 73967 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: +
Debug: 42602 73967 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: vMustReplyEmpty
Debug: 42603 73967 gdb_server.c:391 gdb_log_outgoing_packet(): [esp32c3] sending packet: $#00
Debug: 42604 73967 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: +
Debug: 42605 73967 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: +
Debug: 42606 73968 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: QStartNoAckMode
Debug: 42607 73968 gdb_server.c:391 gdb_log_outgoing_packet(): [esp32c3] sending packet: $OK#9a
Debug: 42608 73968 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: +
Debug: 42609 73968 gdb_server.c:678 gdb_get_packet_inner(): Received first acknowledgment after entering noack mode. Ignoring it.
Debug: 42610 73968 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: Hg0
Debug: 42611 73968 rtos.c:484 rtos_thread_packet(): RTOS: GDB requested to set current thread to 0x0
Debug: 42612 73968 gdb_server.c:391 gdb_log_outgoing_packet(): [esp32c3] sending packet: $OK#9a
Debug: 42613 73968 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: qXfer:features:read:target.xml:0,1000
Debug: 42614 73968 riscv.c:2404 riscv_get_gdb_reg_list_internal(): [esp32c3] reg_class=0, read=0
Debug: 42615 73968 gdb_server.c:388 gdb_log_outgoing_packet(): [esp32c3] sending packet: $<binary-data-4097-bytes>#7f
Debug: 42616 73968 gdb_server.c:375 gdb_log_incoming_packet(): [esp32c3] received packet: qXfer:features:read:target.xml:1000,1000
Debug: 42617 73968 gdb_server.c:388 gdb_log_outgoing_packet(): [esp32c3] sending packet: $<binary-data-2998-bytes>#93
Debug: 42618 74018 gdb_server.c:232 gdb_get_char_inner(): GDB connection closed by the remote client
```

8300, 2023-11-28:
> Looks like GDB closes connection due to internal timeout (see below). Could you add `set remotetimeout 20` before `target remote localhost:3333`?
>
> And one more question. Why do you use [...]

(gdb) set remotetimeout 20
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
bfd requires flen 8, but target has flen 0
(gdb) load
You can't do that when your target is `exec'
(gdb) set remotetimeout 50
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
bfd requires flen 8, but target has flen 0
(gdb) set remotetimeout 20
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
bfd requires flen 8, but target has flen 0

Setting a timeout does not work，same error
[ocd.log](https://github.com/espressif/openocd-esp32/files/13482575/ocd.log)
ricv32-esp-elf-gdb can't install using apt.

gerekon, 2023-11-28:
Now negative reply error has gone. But GDB still closes connection.

> bfd requires flen 8, but target has flen 0

Could you try with `ricv32-esp-elf-gdb`? It is not available via apt, but you can install it via IDF's `install.sh`. If you do not use IDF you can download it [here](https://github.com/espressif/binutils-gdb/releases/tag/esp-gdb-v12.1_20221002). 
If you use IDF I recommend you to go through setup process described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/get-started/linux-macos-setup.html)

> (gdb) load
> You can't do that when your target is `exec'

`load` command is not supported for Espressif chips. Please, see [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html#upload-application-for-debugging) for how to do this with OpenOCD or you can use the same command in GDB `mon program_esp filename.bin 0x10000 verify`.

If you use IDF you can download all binaries (bootloader, partition and app) with single command `mon prorgam_esp_bins <path_to_flasher_args_json>`

8300, 2023-11-29:
> Now negative reply error has gone. But GDB still closes connection.
>
>
> Could you try with `ricv32-esp-elf-gdb`? It is not available via apt, but you can install it via IDF's `install.sh`. If you do [...]

how to run a program, using ricv32-esp-elf-gdb?
(gdb) r
Don't know how to run.  Try "help target".
(gdb) help target
Connect to a target machine or process.
The first argument is the type or protocol of the target machine.
Remaining arguments are interpreted by the target protocol.  For more
information on the arguments for a particular protocol, type
`help target ' followed by the protocol name.

List of target subcommands:

target core -- Use a core file as a target.
target exec -- Use an executable file as a target.
target extended-remote -- Use a remote computer via a serial line, using a gdb-specific protocol.
target record-btrace -- Collect control-flow trace and provide the execution history.
target record-core -- Log program while executing and replay execution from log.
target record-full -- Log program while executing and replay execution from log.
target remote -- Use a remote computer via a serial line, using a gdb-specific protocol.
target tfile -- Use a trace file as a target.

Type "help target" followed by target subcommand name for full documentation.
Type "apropos word" to search for commands related to "word".
Type "apropos -v word" for full documentation of commands related to "word".
Command name abbreviations are allowed if unambiguous.
(gdb) thb main
No hardware breakpoint support in the target.
(gdb) c
The program is not being run

gerekon, 2023-11-29:
Please see this document https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/using-debugger.html#command-line

8300, 2023-11-29:
> Please see this document https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/using-debugger.html#command-line
> 
> And this one https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/debugging-examples.html#command-line

i only used your header files,  the c file is my own. so no app_main

gerekon, 2023-11-29:
> i only used your header files, the c file is my own. so no app_main

In that case you need to specify the name of the entry point of your app.

How do you build your app? Is it supposed to be executed in RAM only or you build an image for flash? If you want to load ELF into RAM and debug it you can use our test app as a reference (https://github.com/espressif/esp-idf/tree/master/tools/test_apps/system/gdb_loadable_elf). But loading and debugging ELF in RAM is not standard scenario for debugging app on our chips.

gerekon, 2023-11-29:
See this gdbinit file for loading and debugging ELF in RAM https://github.com/espressif/esp-idf/blob/master/tools/test_apps/system/gdb_loadable_elf/gdbinit_esp32c3

erhankur, 2023-12-13:
@8300 any progress on your side?

erhankur, 2023-12-21:
Closed due to inactivity. Feel free to re-open if there is still an issue.

closed 2023-12-21 by erhankur

RomkaE, 2024-02-23:
I encountered a similar issue, but with some interesting observations.

The functionality of JTAG operation varies depending on the manufacturer of the flash memory. Specifically, I am working with ESP32-WROOM-32E modules.

JTAG debugging functions properly only on modules equipped with flash memory from XMC (manufacturer code 20), while it fails on modules with GigaDevice flash memory (manufacturer code c8).

These modules are installed on identical boards, and for debugging purposes, I utilize ESP-PROG. To test my hypothesis, I performed a chip replacement on one of the devices. I replaced the GigaDevice flash memory chip with an XMC chip. Subsequent to this replacement, debugging functionality is restored.

The command used to initiate the openocd server is:

```openocd -f esp32.cfg -c "init; halt; esp appimage_offset 0x30000"```

For XMC flash, the openocd log output indicates everything is functioning as expected:

```Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:41)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32.cpu1] Target halted, PC=0x40095C03, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : Set GDB target to 'esp32.cpu1'
Info : [esp32.cpu0] Target halted, PC=0x4009252E, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x30020 -> 0x3f400020, 254 KB
Info : Flash mapping 1: 0x70020 -> 0x400d0020, 913 KB
Info : Using flash bank 'esp32.cpu0.irom' size 916 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x30020 -> 0x3f400020, 254 KB
Info : Flash mapping 1: 0x70020 -> 0x400d0020, 913 KB
Info : Using flash bank 'esp32.cpu0.drom' size 256 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x30020 -> 0x3f400020, 254 KB
Info : Flash mapping 1: 0x70020 -> 0x400d0020, 913 KB
Info : Using flash bank 'esp32.cpu1.irom' size 916 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x30020 -> 0x3f400020, 254 KB
Info : Flash mapping 1: 0x70020 -> 0x400d0020, 913 KB
Info : Using flash bank 'esp32.cpu1.drom' size 256 KB
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections```

However, for GigaDevice flash, the log output is as follows:

```Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:41)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40095C32, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu1'
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40092532, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Error: Failed to get flash maps (-8)!
Warn : Invalid magic number in app image!
Warn : Failed to get flash mappings (-4)!
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Using flash bank 'esp32.cpu0.irom' size 0 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Error: Failed to get flash maps (-8)!
Warn : Invalid magic number in app image!
Warn : Failed to get flash mappings (-4)!
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Using flash bank 'esp32.cpu0.drom' size 0 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Error: Failed to get flash maps (-8)!
Warn : Invalid magic number in app image!
Warn : Failed to get flash mappings (-4)!
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Using flash bank 'esp32.cpu1.irom' size 0 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Error: Failed to get flash maps (-8)!
Warn : Invalid magic number in app image!
Warn : Failed to get flash mappings (-4)!
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Using flash bank 'esp32.cpu1.drom' size 0 KB
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections```

Of course, I flash both firmware and bootloader before launching the openocd server.

RomkaE, 2024-02-23:
I managed to find a "strange" solution to launch openocd without errors for the ESP32 module with GigaDevice flash memory. Here are the steps:
1. Flash the firmware.
2. Start openocd with the command: `openocd -f esp32.cfg.`
3. Begin debugging in Eclipse, and encounter the gdb server error "Failed to get flash maps (-6)!"
4. Stop the gdb server, and restart it with the following command: `openocd -f esp32.cfg -c "init; halt; esp appimage_offset 0x30000".`
Upon restarting the gdb server, the <Flash mapping> works as intended:
```
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:41)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : Set GDB target to 'esp32.cpu1'
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x30020 -> 0x3f400020, 254 KB
Info : Flash mapping 1: 0x70020 -> 0x400d0020, 913 KB
Info : Using flash bank 'esp32.cpu0.irom' size 916 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x30020 -> 0x3f400020, 254 KB
Info : Flash mapping 1: 0x70020 -> 0x400d0020, 913 KB
Info : Using flash bank 'esp32.cpu0.drom' size 256 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x30020 -> 0x3f400020, 254 KB
Info : Flash mapping 1: 0x70020 -> 0x400d0020, 913 KB
Info : Using flash bank 'esp32.cpu1.irom' size 916 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x30020 -> 0x3f400020, 254 KB
Info : Flash mapping 1: 0x70020 -> 0x400d0020, 913 KB
Info : Using flash bank 'esp32.cpu1.drom' size 256 KB
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

And debugging functions perfectly