espressif/openocd-esp32 #304: Fail to flash by jtag

puppy112358, 2023-12-10:
### Development Kit

ESP32-S3-DevKitC

### Module or chip used

ESP32-S3-WROOM-1

### Debug Adapter

ESP32-S3-JTAG

### OpenOCD version

v0.12.0-esp32-20230921

### Operating System

Windows

### Using an IDE ?

yes, I used CLion.

### OpenOCD command line

openocd -f board/esp32-wrover-kit-3.3v.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v5.3.0

### Problem Description

1.flash my code to esp32-s3
2.run the esp32-s3
3.reflash the code and failed

### Expected behavior

it is expected that esp32-s3 should be able to be flashed by jtag, but now the time runs out to flash the code to it while flashing, which means I failed to flash .The code can be flashed for many times normally before but now it can't and it really annoys me

### Screenshots

[image]
when I try to flash, it takes a long time and then tell me the time runs out

erhankur, 2023-12-10:
As I see, you have esp32s3 board but you selected a board config for esp32. If you have a debug probe with 4 pins jtag connection, you should select esp32s3-ftdi.cfg, if you are connecting from the usb-serial-jtag, you should select esp32s3-builtin.cg from the tcl/board directory.

puppy112358, 2023-12-10:
> As I see, you have esp32s3 board but you selected a board config for esp32. If you have a debug probe with 4 pins jtag connection, you should select esp32s3-ftdi.cfg, if you are connecting from the usb-serial-jtag, you should select esp32s3-builtin.cg from the tcl/board directory.

Maybe my above expression is not clear, but I actually select esp32-s3-builtin.cfg as you say instead of esp32. I can flash my code normally before my stupid operations but I have already changed it back while it still makes nothing

puppy112358, 2023-12-10:
> As I see, you have esp32s3 board but you selected a board config for esp32. If you have a debug probe with 4 pins jtag connection, you should select esp32s3-ftdi.cfg, if you are connecting from the usb-serial-jtag, you should select esp32s3-builtin.cg from the tcl/board directory.

[image]
Actually, what I used is esp32s3-builtin.cfg.  I could flash it before my stupid operations, but I have already changed it back and it didn't work

erhankur, 2023-12-11:
Can you send the detailed OpenOCD log (enabled with -d3) option. 
After you erase the flash completely, do you still have an issue?
What were your operations? Maybe can give us a hint.

puppy112358, 2023-12-12:
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/13648931/openocd_log.txt)
this is my OpenOCD log

puppy112358, 2023-12-12:
> Can you send the detailed OpenOCD log (enabled with -d3) option. After you erase the flash completely, do you still have an issue? What were your operations? Maybe can give us a hint.

My operations are shown as follows: 
Because I suspected there may be stack overflow in the project so I add a function called uxTaskGetStackHighWaterMark(), after  using it, the problem came out

puppy112358, 2023-12-12:
> Why you are insist on using esp32 config instead of esp32s3?

NO, what I used is esp32s3 instead of esp32

puppy112358, 2023-12-12:
> Why you are insist on using esp32 config instead of esp32s3?

what I used is actually esp32s3 config instead of esp32

erhankur, 2023-12-12:
OK, then send me the correct logs please. And I don't get it your problem, yet. Please summarize your problem again with using correct esp32s3 config. You can check your log. It is registered as esp32 target. Not esp32s3.

erhankur, 2023-12-21:
@puppy112358 any progress on your side?

erhankur, 2023-12-28:
Closed due to inactivity. Feel free to reopen if your issue continues.

puppy112358, 2023-12-29:
> OK, then send me the correct logs please. And I don't get it your problem, yet. Please summarize your problem again with using correct esp32s3 config. You can check your log. It is registered as esp32 target. Not esp32s3.

[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/13790590/openocd_log.txt)

puppy112358, 2023-12-29:
> OK, then send me the correct logs please. And I don't get it your problem, yet. Please summarize your problem again with using correct esp32s3 config. You can check your log. It is registered as esp32 target. Not esp32s3.

I'm sorry that I didn't reply to you on time.  This is my logs. 
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/13790703/openocd_log.txt)

puppy112358, 2023-12-29:
> @puppy112358 any progress on your side?

Sorry for my poor English, I can't get your point. What's the meaning of "progress on the side"?

erhankur, 2023-12-29:
Still I see esp32 config files in the log. 

```
Debug: 14 36 configuration.c:88 find_file(): found C:\Users\86180\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/board/esp32-wrover-kit-3.3v.cfg
Debug: 15 43 command.c:151 script_debug(): command - ocd_find interface/ftdi/esp32_devkitj_v1.cfg
Debug: 16 44 configuration.c:88 find_file(): found C:\Users\86180\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/interface/ftdi/esp32_devkitj_v1.cfg
Debug: 17 46 command.c:151 script_debug(): command - adapter driver ftdi
Debug: 18 46 command.c:151 script_debug(): command - ftdi vid_pid 0x0403 0x6010 0x0403 0x6014
Debug: 19 46 command.c:151 script_debug(): command - ftdi channel 0
Debug: 20 46 command.c:151 script_debug(): command - ftdi layout_init 0x0008 0xf00b
Debug: 21 46 command.c:151 script_debug(): command - ftdi layout_signal LED -data 0x1000
Debug: 22 46 command.c:151 script_debug(): command - ftdi layout_signal LED2 -data 0x2000
Debug: 23 46 command.c:151 script_debug(): command - ftdi layout_signal LED3 -data 0x4000
Debug: 24 46 command.c:151 script_debug(): command - ftdi layout_signal LED4 -data 0x8000
```

puppy112358, 2023-12-29:
> Still I see esp32 config files in the log.
>
> ```
> Debug: 14 36 configuration.c:88 find_file(): found [...]
ii
Yeah, I see. I have just checked my file but only to find that the openocd file has only one version of openocd. I guess the openocd actually has only one version because I can use it normally before.

erhankur, 2023-12-29:
I am not sure, how can I help you. You can re-install your idf tools and try to flash with the idf.py openocd command in the application directory. Something like; 
```
idf.py openocd --openocd-commands "-f board/esp32s3-builtin.cfg -c 'program_esp_bins build flasher_args.json verify reset exit' -d3 -l log.txt"
```
If this works, then you can have a look at your IDE settings. If not, share the log.txt

puppy112358, 2023-12-31:
I have re-install my idf tools and then flash again.  
[image]
I checked my microcontroller and it didn't change as the new code so the flashing operation failed again. 

Actually I think the point is that my code can be flashed the first time and failed the next time.  The senior told me that maybe my code has conflictions with the j-tag pin , but he has no ideas either.

puppy112358, 2023-12-31:
[log.txt](https://github.com/espressif/openocd-esp32/files/13800246/log.txt)
This is my new log

erhankur, 2023-12-31:
Can you try the below?
```
idf.py openocd --openocd-commands "-f board/esp32s3-builtin.cfg -c 'program_esp_bins build flasher_args.json verify reset no_clock_boost exit' -d3 -l log.txt"
```

I added the `no_clock_boost` option to the program command.

Also is it possible to re-flash the `hello_world` app from esptool? Then you can try from the JTAG again.

puppy112358, 2024-01-05:
> Can you try the below?
>
> ```
> idf.py openocd --openocd-commands "-f board/esp32s3-builtin.cfg -c 'program_esp_bins build flasher_args.json verify reset no_clock_boost exit' -d3 -l log.txt"
> ```
>
> I [...]

Feeling so sorry that I didn't reply on time

puppy112358, 2024-01-05:
[image]

puppy112358, 2024-01-05:
[log.txt](https://github.com/espressif/openocd-esp32/files/13838679/log.txt)

erhankur, 2024-01-05:
Ok. `hello_world` loaded successfully. What about your application?

puppy112358, 2024-01-05:
> Ok. `hello_world` loaded successfully. What about your application?

[image]

puppy112358, 2024-01-05:
[image]
This is what is shown on CLion

erhankur, 2024-01-05:
Did you add `no_clock_boost` to the CLion OpenOCD config?

puppy112358, 2024-01-05:
oh, I didn't. I'll try again

puppy112358, 2024-01-05:
It seems that there is no such option for CLion to add `no_clock_boost`

puppy112358, 2024-01-05:
[image]

erhankur, 2024-01-05:
OK, please find below lines in the esp_common.cfg and comment out them. 
```
	if {$clock_boost == 1} {
		if {[catch {eval esp flash_stub_clock_boost "on"}] != 0} {
			program_error "** Clock configuration set failed **" $exit
		}
	}
```

Please make sure you are doing this in the correct config file.

puppy112358, 2024-01-05:
OH! How magic！
[image]

puppy112358, 2024-01-05:
Could you tell me the principle?

erhankur, 2024-01-05:
Good. Can you try one more thing? Then I will update the master branch.

Below, I allowed programming to continue even when the clock command failed. (removed `$exit` after the error message.)

```
	if {$clock_boost == 1} {
		if {[catch {eval esp flash_stub_clock_boost "on"}] != 0} {
			program_error "** Clock configuration set failed **"
		}
	}
```

This command was added to speed up flashing time. But sometimes RTC  functions fail, that's why you see an error. I will investigate it.  
Also if it is possible to send me your binaries, I can test it here.

puppy112358, 2024-01-05:
Did you mean it?
[image]

erhankur, 2024-01-05:
Yes. Is this from example directory or did you add something? I would like to reproduce the error here. So, you can send the build directory or the full project.

puppy112358, 2024-01-05:
[image]
The file is too big...

erhankur, 2024-01-05:
Yes, it contains a lot of output files. You can remove the build directory and send the project files only. And please one more try without exit command.

puppy112358, 2024-01-05:
[main.zip](https://github.com/espressif/openocd-esp32/files/13841511/main.zip)

puppy112358, 2024-01-05:
This is the whole project

erhankur, 2024-01-05:
It works for me :( 

Ok. Waiting for your last test.

puppy112358, 2024-01-05:
[image]
It remains showing this.

puppy112358, 2024-01-05:
[log.txt](https://github.com/espressif/openocd-esp32/files/13842548/log.txt)
This is the latest log.

puppy112358, 2024-01-05:
May I ask you a question? Why did this wired question happened? Is it because the clock setting is too high?

puppy112358, 2024-01-07:
[image]
it takes a long time but didn't work ,and the log file turned out to be empty

erhankur, 2024-01-07:
OK, thanks. I am closing this and will investigate the root cause within the next weeks.

closed 2024-01-07 by erhankur