espressif/openocd-esp32 #305: ESP32S3 Espressif sysview with FreeRTOS multicore does not work (IDFGH-11368)

TiboDevC, 2023-12-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## General issue report

I try to use the Espressif sysview to debug a memory leak. Here is the Espressif documentation about sysview:
https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32s3/api-reference/system/heap_debug.html#host-based-mode 

This example works well when FreeRTOS is configured to run on a single core but does not work when it runs on dual core.

## Environment 

- ESP32S3
- Espressif SDK v5.1.1
- openocd `v0.12.0-esp32-20230921 (2023-09-21-13:41)`
- xtensa-esp32s3-elf-gdb `12.1_20221002`

## Demo code

### Working example

The Espressif example works well, here is an example code: 
https://github.com/TiboDevC/esp32_heap_tracing/tree/master

### Non-working example

The only difference with the working one is `Run FreeRTOS only on first core` is unchecked in `menuconfig`
https://github.com/TiboDevC/esp32_heap_tracing/tree/freertos_multi_core

### Steps to reproduce

Flash one of the above example code, then:
```
# Run openocd
openocd -f board/esp32s3-builtin.cfg

# In another terminal, run gdb with the `gdbinit` described as above in github links
xtensa-esp32s3-elf-gdb -x gdbinit ./build/heap-trace-host-example.elf
```
### logs when it works
```
$ xtensa-esp32s3-elf-gdb -x gdbinit ./build/heap-trace-host-example.elf 
GNU gdb (esp-gdb) 12.1_20221002
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/heap-trace-host-example.elf...
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
[Switching to Thread 1070174560]
0x40379b46 in esp_cpu_wait_for_intr () at /home/myUserName/esp/esp-idf/components/esp_hw_support/cpu.c:121
121 }
JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[esp32s3.cpu0] requesting target halt and executing a soft reset
[... 117 lines cut ...]

Thread 4 "main" hit Temporary breakpoint 1, heap_trace_start (mode_param=(HEAP_TRACE_LEAKS | unknown: 0x2)) at /home/myUserName/esp/esp-idf/components/app_trace/heap_trace_tohost.c:36
36	{
Total trace memory: 16384 bytes
Open file /tmp/heap_cpu0.svdat
Open file /tmp/heap_cpu1.svdat
App trace params: from 2 cores, size 4294967295 bytes, stop_tmo -1 s, poll period 0 ms, wait_rst 0, skip 0 bytes
Connect targets...
Targets connected.
```
It is stuck here. When I break with gdb:
```
^C[esp32s3.cpu0] Target halted, PC=0x4037EBE8, debug_reason=00000000
Set GDB target to 'esp32s3.cpu0'
[esp32s3.cpu1] Target halted, PC=0x4037EBFA, debug_reason=00000000

Thread 4 "main" received signal SIGINT, Interrupt.
spinlock_acquire (timeout=-1, lock=0x3fc9409c <xKernelLock>) at /home/myUserName/esp/esp-idf/components/esp_hw_support/include/spinlock.h:112
112	        lock_set = esp_cpu_compare_and_set(&lock->owner, SPINLOCK_FREE, core_id);
```
it seems to be stuck in this function.

I tried also with Espressif v5.2-dev but same result.

Is it a bug or a miss-configuration?

erhankur, 2024-02-13:
@TiboDevC can you try again with the latest idf-master branch and openocd master branch? 

I gave another try now and looks like it works. 

```
I (179) cpu_start: Multicore app
I (189) cpu_start: Pro cpu start user code
I (189) cpu_start: cpu freq: 160000000 Hz
I (189) app_init: Application information:
I (189) app_init: Project name:     sysview_tracing_heap_log
I (194) app_init: App version:      v5.3-dev-2032-g4d90eedb6e-dirty
I (200) app_init: Compile time:     Feb 13 2024 21:48:23
I (205) app_init: ELF file SHA256:  3448a2978...
I (210) app_init: ESP-IDF:          v5.3-dev-2032-g4d90eedb6e-dirty
I (216) app_init: Min chip rev:     v0.0
I (219) app_init: Max chip rev:     v0.99 
I (223) app_init: Chip rev:         v0.1
I (227) heap_init: Initializing. RAM available for dynamic allocation:
````

```
Open On-Chip Debugger v0.12.0-esp32-20230921-271-gc400745e-dirty (2024-02-13-19:18)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
0
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
WARNING!!! OpenOCD binary version (Open On-Chip Debugger v0.12.0-esp32-20230921-271-gc400745e-dirty (2024-02-13-19:18)) doesn't match the .cfg file version (v0.12.0-esp32-20230419)
WARNING: ESP flash support is disabled!
WARNING: ESP flash support is disabled!
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:E1:19:00)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Examination succeed
Info : [esp32s3.cpu1] Examination succeed
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x4037AD52, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (21) - (USB UART reset)
Info : [esp32s3.cpu1] Target halted, PC=0x4037AD52, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (21) - (USB UART reset)
Info : accepting 'telnet' connection on tcp/4444
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Target halted, PC=0x4037AD52, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4037AD52, debug_reason=00000000
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x403CD669, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] Target halted, PC=0x40043A3B, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32s3.cpu0] Target halted, PC=0x4037AD52, debug_reason=00000000
Info : [esp32s3.cpu1] Target halted, PC=0x4037AD52, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : Detected FreeRTOS version: (10.5.1)
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Target halted, PC=0x420080D0, debug_reason=00000001
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4037AD52, debug_reason=00000000
Info : Detected FreeRTOS version: (10.5.1)
Info : Total trace memory: 16384 bytes
Info : Open file /tmp/heap_cpu0.svdat
Info : Open file /tmp/heap_cpu1.svdat
App trace params: from 2 cores, size 4294967295 bytes, stop_tmo -1 s, poll period 0 ms, wait_rst 0, skip 0 bytes
Connect targets...
Info : Targets connected.
Info : [esp32s3.cpu0] Target halted, PC=0x4037D298, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x400559E0, debug_reason=00000000
Info : dropped 'gdb' connection
```

closed 2024-03-05 by erhankur

TiboDevC, 2024-04-15:
Hi @erhankur , I tried again with the tag `v5.3-dev`
```
➜  esp-idf git:(v5.3-dev) ✗ git log --oneline -1
8171b22c40 (HEAD, tag: v5.3-dev) change(version): Update version to 5.3.0
```

I also download the last openocd release available https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20240318 but I still have the same problem:
```
➜  heap_tracing openocd -f board/esp32s3-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:25)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
WARNING!!! OpenOCD binary version (Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:25)) doesn't match the .cfg file version (v0.12.0-esp32-20230921)
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (48:27:E2:F0:01:60)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Examination succeed
Info : [esp32s3.cpu1] Examination succeed
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x4037F73D, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x40001C38, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
Info : Flash mapping 0: 0x40020 -> 0x3c0c0020, 314 KB
Info : Flash mapping 1: 0x90020 -> 0x42000020, 713 KB
Info : [esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : [esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
Info : Flash mapping 0: 0x40020 -> 0x3c0c0020, 314 KB
Info : Flash mapping 1: 0x90020 -> 0x42000020, 713 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 716 KB
Info : [esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
Info : Flash mapping 0: 0x40020 -> 0x3c0c0020, 314 KB
Info : Flash mapping 1: 0x90020 -> 0x42000020, 713 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 316 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Info : Detected FreeRTOS version: (10.5.1)
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Target halted, PC=0x4037BF56, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4200B5DC, debug_reason=00000001
Info : Detected FreeRTOS version: (10.5.1)
[... 65 lines cut ...]

Thread 2 "task0" hit Temporary breakpoint 1, heap_trace_start (mode_param=HEAP_TRACE_ALL) at esp/esp-idf/components/app_trace/heap_trace_tohost.c:36
36	{
Total trace memory: 16384 bytes
Open file /tmp/heap_cpu0.svdat
Open file /tmp/heap_cpu1.svdat
App trace params: from 2 cores, size 4294967295 bytes, stop_tmo -1 s, poll period 0 ms, wait_rst 0, skip 0 bytes
Connect targets...
Targets connected.
```

From the UART (the app running)
```
[2024-04-15 13:32:06.792] D (23637) esp_apptrace: force flush
[2024-04-15 13:32:06.792] D (23637) esp_apptrace: Recvd 1 bytes from host (@ 0x3fce0000) [1 0 1 0 3e d5 3e 90 .. 73 3a 20 46 61
 6c 6c 69]
[2024-04-15 13:32:06.808] D (23640) esp_apptrace: esp_apptrace_trax_down_buffer_write_nolock WRS 0-0-32 1
[2024-04-15 13:32:06.808] D (23647) esp_apptrace: esp_apptrace_trax_down_buffer_write_nolock wr 1
[2024-04-15 13:32:06.808] D (23654) esp_apptrace: esp_apptrace_trax_down_buffer_write_nolock wr 1 to 0x3fc9d2fc from 0x3fce0003
[2024-04-15 13:32:06.824] D (23663) esp_apptrace: esp_apptrace_trax_down_buffer_write_nolock wr 1/1
[2024-04-15 13:32:06.824] D (23670) esp_apptrace: Read 1 bytes from host
```

@erhankur what is your GDB version? Did you try this example? https://github.com/TiboDevC/esp32_heap_tracing/tree/freertos_multi_core.
I'm running ESP32S3 chip revision v0.2

erhankur, 2025-05-22:
@TiboDevC We found the root cause and applied to the fix to IDF release branches except v5.4 . Please check it out.

https://github.com/espressif/esp-idf/commit/aae949a2777b8f36a1b34c210e6b6abee14ebfc5