espressif/openocd-esp32 #307: ESP32-S3 Built-in debug not working on Seeed Studio Xiao ESP32-S3 board

AlbertoLopSie, 2023-12-22:
### Development Kit

Seeed Studio Xiao ESP32-S3

### Module or chip used

ESP32-S3

### Debug Adapter

Built-In

### OpenOCD version

v0.12.0-esp32-20230313

### Operating System

Mac OS 13.3.1 (Apple M2)

### Using an IDE ?

VS Code

### OpenOCD command line

PlatformIO generated

### JTAG Clock Speed

12000

### ESP-IDF version

4.4.6

### Problem Description

Debugging is not working in the Seeed Studio xiao-ESP32-S3 board.  According to documentation, the S3 supports direct (no external probe) debugging through its internal JTAG USB interface however, I cannot make i work.

Also, there's a confusion on the Debugging section of the [PlatformIO Board doc](https://docs.platformio.org/en/latest/boards/espressif32/seeed_xiao_esp32s3.html): it says that this board **doesn't support 1-click debugging** when it does.

``` yaml
; platformio.ini

[env:seeed_xiao_esp32s3]
platform = espressif32
board = seeed_xiao_esp32s3
framework = arduino
build_type = debug 
debug_tool = esp-builtin
debug_init_break = tbreak setup

debug_port = /dev/cu.usbmodem21301     ; on my Mac
monitor_port = /dev/cu.usbmodem21301
upload_port = /dev/cu.usbmodem21301
```
Only causes debugger to start but freeze with the following at the debug console:

```sh
undefinedOpen On-Chip Debugger v0.12.0-esp32-20230313 (2023-03-13-09:08)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 12000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (34:85:18:91:38:88)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 13333 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x4201F15E, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] Target halted, PC=0x4201F15E, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Reading symbols from /Users/Alberto/dev/DMI/HelloS3/.pio/build/seeed_xiao_esp32s3/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
```
and debugging session never starts.

It looks like the JTAG interface connects ok because the CPU is recognized, but after that nothing happens.

Uploading through `esp-builtin` works **only once**

``` sh
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 12000 kHz

** Programming Started **
** Programming Finished in 6079 ms **
** Verify Started **
** Verify OK **
** Programming Started **
** Programming Finished in 1296 ms **
** Verify Started **
** Verify OK **
** Programming Started **
** Programming Finished in 1028 ms **
** Verify Started **
** Verify OK **
** Programming Started **
** Programming Finished in 1151 ms **
** Verify Started **
** Verify OK **
shutdown command invoked
```

To get upload working after a failed debug session, I had to kill rogue `openocd`and `python3.9` processes left by VSCode/PlatformIO

### Debug Logs

```shell
Attached above
```

### Expected behavior

A normal debug session started

### Screenshots

[image]

gerekon, 2023-12-22:
HI @AlbertoLopSie 

I do not see any errors related to OpenOCD here. The only strange thing I see is that for debugging you use `Open On-Chip Debugger v0.12.0-esp32-20230313 (2023-03-13-09:08)`, but for "Uploading through esp-builtin" another version is used `Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49`.

I'd recommend you to redirect this question to PlatformIO guys.

AlbertoLopSie, 2023-12-22:
Hi @gerekon !

May be those logs were created with some time between, in the middle I tried upgrading the openocd package to see if it helps, but it didn't.

I already posted an [issue on PlaformIO](https://github.com/platformio/platform-espressif32/issues/1263) repo, but giving that platformio only runs openocd, and according to the logs it appears as running ok and detecting the CPU but not entering debug session I though this was an openocd related issue.

gerekon, 2023-12-22:
I can have a look at verbose OpenOCD log to see where the process sticks. To get that log you need to add `-d3` option to OpenOCD command line. But I can not tell you how to do this in PlatformIO.

AlbertoLopSie, 2023-12-22:
There is a way… I don’t remember it right know but I recall having to use a
custom openocd invoke command some time ago. I’ll check documentation.

AlbertoLopSie, 2023-12-23:
**Update**:  A guy in PlatformIO project hit in the target with platformio.ini configuration and the debugger is now working although with some quirks related to breakpoints.

Anyone interested can look [here](https://github.com/platformio/platform-espressif32/issues/1263) for details.

Thank you very much @gerekon !!!

erhankur, 2023-12-23:
@AlbertoLopSie Thank you for the information. We can close the issue.

closed 2023-12-23 by erhankur